Bodmer/TFT_eSPI #2998: minor documentation issue for pico poin assignment in Setup62_RP2040_Nano_Connect_ILI9341.h?

AJ6CU, 2023-11-23:
In Setup62_RP2040_Nano_Connect_ILI9341.h

At line 135 - 143  we have:
```
#else // For Earle Philhower's package the Arduino Dx pin designations must be used:
  #define TFT_MISO  D12
  #define TFT_MOSI  D11
  #define TFT_SCLK  D13
  #define TFT_CS    D10  // Chip select control pin
  #define TFT_DC    D8   // Data Command control pin
  #define TFT_RST   D9   // Reset pin (could connect to Arduino RESET pin)
  #define TOUCH_CS  D7
#endif`
```
According to the Pico pinout, D13 (GPIO13) is Chip select and D10 (GPIO10) is Clock for SPI1. 

The other pin assignments are as expected (Dxx->GPIOXX->maps to function for SPI1).

Mark

Bodmer, 2023-12-03:
It looks like Earle has updated to board package to use pxx numbers for compatibility with the mbed package, so instead of Dxx  try using those pxx  pin references.

Bodmer, 2023-12-05:
That setup file is for the RP2040 Nano Connect and D13 (GPIO 6) is SCK. See documentation for that board here:
https://docs.arduino.cc/tutorials/nano-rp2040-connect/rp2040-01-technical-reference

For the PIco use GPIO numbers as the Dxx to GPIO number mapping is not the same as the Nano Connect board. It is important to select the correct board in the IDE if Dxx designations are used.

closed 2023-12-05 by Bodmer