Bodmer/TFT_eSPI #3021: ILI9486 display connected to nodeMCU-32S only flashes white

Tiltolein, 2023-12-05:
I hope someone can help me as I´m by now clueless to the point I don´t even know what infos to provide and how to further investigate and potential errors on my side.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
   When connecting the display and running any program the display will simple stay white or flash between black and white

2. IDE (e.g. Arduino or PlatformIO)
   The issue appears on both Arduino and PlatformIO
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
   I´m using version 2.5.34
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
   I´m using version 2.0.14
5. Procesor, e.g RP2040, ESP32 S3 etc
   The processor is an nodeMCU-32S
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
    The TFT driver is ILI9486 :https://www.berrybase.de/3-5-display-fuer-raspberry-pi-mit-resistivem-touchscreen
7. Interface type (SPI or parallel)
    Tbh I have no idea but I think its SPI. Should be clear regarding the product-page for someone more clever than me :)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
    I have tried liking up the display using multiple different setups, all have led to the same result.

   My current setup is as follows: 
   [image]

2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
    I don´t know what file exactly is meant. At this point I´ve tried at least 10 different combinations of code but none of them seem to 
    work for me

3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.

4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

DO NOT paste code directly into the issue. To correctly format code put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies. [See here:](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

Example output:

```
  Serial.begin(115200);
  tft.init();
```

Tiltolein, 2023-12-06:
Thanks for the fast reply! This is my output

[code]
TFT_eSPI ver = 2.5.34
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9486
Display width  = 320
Display height = 480

MOSI    = GPIO 12
MISO    = GPIO 2
SCK     = GPIO 14
TFT_CS   = GPIO 5
TFT_DC   = GPIO 15
TFT_RST  = GPIO 13
TOUCH_CS = GPIO 18

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 26.00
Touch SPI frequency   = 2.50

Bodmer, 2023-12-07:
You need to add this line to the setup file:

#define RPI_DISPLAY_TYPE

and change the SPI  clock rate to 20MHz:

#define SPI_FREQUENCY 20000000

Tiltolein, 2023-12-07:
I´ve changed the mentioned values.

Now I´m getting the following errors while compiling any code.

In file included from c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:
c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data4_io_num' [-Wmissing-field-initializers]
   };
   ^
c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data5_io_num' [-Wmissing-field-initializers]
c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data6_io_num' [-Wmissing-field-initializers]
c:\Users\benj\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:805:3: warning: missing initializer for member 'spi_bus_config_t::data7_io_num' [-Wmissing-field-initializers]

Bodmer, 2023-12-10:
I have looked into this further and it is board package version related. Try changing you library code to disable this line thus by adding the "x":
https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.c#L796

I have temporarily updated the master library with this bix. I am not sure if this will cause problems with older board pckages.

Tiltolein, 2023-12-10:
Hey again thanks a lot for the effort. However I was always using version 2.0.14. The changed code including the "x" sadly didn´t help either. I appreciate the effort a lot

Tiltolein, 2023-12-10:
I´ve just got it to work, thank you so much for your patience and time

Bodmer, 2023-12-11:
Great!

closed 2023-12-11 by Bodmer