Bodmer/TFT_eSPI #3036: ESP32 crash/corruption when OFR calls back to _drawPixel()

chconnor, 2023-12-13:
Hi!

I'm using an ESP32-DevKitC-VIE which uses a ESP32-WROVER-IE module. VScode + PlatformIO in linux. I'm using OpenFontRender in conjunction with TFT_eSPI (2.5.41) to draw to an ILI9488 screen ([the DFR0669](https://wiki.dfrobot.com/3.5inches_480_320_TFT_LCD_Capacitive_Touchscreen_SKU_DFR0669)).

I'm getting a crash that seems to be coming from corruption of some kind. The reason I'm reporting this here instead of with OFR is that it crashes at the _drawPixel() command (see below) which, if I understand correctly, is the callback to TFT_eSPI to actually draw the pixel. (Let me know if I should report this at OFR instead.)

I spent the last couple days getting the smallest reproducible example of this that I can, here it is (see below for font .h file and other config info):

<details>
<summary>main.cpp</summary>

```
#include <Arduino.h>
#include "SPI.h"
#include <TFT_eSPI.h>
#include "URWGothic_Book.h"
#include "OpenFontRender.h"
static OpenFontRender ofr;
static TFT_eSPI tft = TFT_eSPI();

void setup()
{
  pinMode(4, OUTPUT); // screen blanking pin
  digitalWrite(4, HIGH); // HIGH=screen on
  tft.init();
  tft.fillScreen(TFT_BLACK);
	ofr.setDrawer(tft);
	ofr.loadFont(URWGothic_Book, sizeof(URWGothic_Book)); // must be this font
  ofr.setFontSize(12); // 12 crashes, 13 crashes, 40 does not, 5 does not
  ofr.setAlignment(Align::TopCenter); // must be *Center to make it crash
  ofr.drawString("-7.0d", 40, 294, 0xec91, TFT_BLACK); // this color not important, x coord important, string matters vis a vis font size
  ofr.drawString("-6.0d", 70, 294, 0xec91, TFT_BLACK); // color important, x coord important
}

void loop() { }
```
</details>

When I run this code, maybe 1 in 3 times I get a "crash" -- screen goes crazy, etc. in the debugger seems that threads keep running, etc, but something goes haywire. In the real project that I am seeing this in, sometimes the screen goes white (i.e. reset), sometimes it draws crazy things, flickers, etc. It's obvious that the instruction code has just crashed and things are randomly happening.

[Here is a video of this demo code running that demonstrates the problem](https://www.youtube.com/shorts/yYAlnW8pruI). The video shows the intermittent nature. It does not show how sometimes the screen just goes to black, or does other things.

Note that changing almost anything will prevent the crash. E.g. if I use a different font, no crash; if I use an Align that is not *Center, it will not crash. If I set font size to 5 or 40, no crash. 12 or 13: crash. If I change the content of either string being drawn, sometimes it crashes, sometimes it does not. If I change the foreground color (0xec91) of the second drawString() to TFT_WHITE, TFT_DARKGREY, TFT_BLUE, or TFT_RED, no crash. 0xec90, 0xec91, 0xec92: crash. FG color of the first drawString() doesn't seem to matter. If I change the x coordinate passed to drawString(), no crash.

I tried to debug it (with ESP-PROG); since this appears to be some kind of corruption, I'm not sure how useful this information is, but just in case I will report: when the drawString() is drawing the "0" character from the "-6.0d" string, and when it gets to [line 1418 of OpenFontRender.cpp](https://github.com/takkaO/OpenFontRender/blob/f163cc6a74d6a98752ff896d880f2bdbea269084/src/OpenFontRender.cpp#L1418), when _y is 10 and _x is 4 and x is 70, stepping over _drawPixel() is what causes things to crash. (I guess it could be alphaBlend() but that function is so simple I don't imagine it could be related.)

Again, I know that may not be "the cause", that's just when the crash visibly happens for me. I set a breakpoint at [line 1384](https://github.com/takkaO/OpenFontRender/blob/f163cc6a74d6a98752ff896d880f2bdbea269084/src/OpenFontRender.cpp#L1384), resume execution over and over until I'm about to draw the "0" digit in the -6.0d string,  then I enable the conditional breakpoint at line 1418 which has condition "_y == 10 && _x == 4 && x == 70" and resume execution. Then it stops at the conditional breakpoint, and if I step over _drawPixel, that's when a crash happens. Well, maybe 1 out of 5 times, anyway.

Thanks in advance for any time you have to look into this!

<details>
<summary>TFT_eSPI_Setup.h</summary>
// for ILI9488 DFRobot screen
// https://github.com/Bodmer/User_Manual_TFT_eSPI/blob/master/TFT_eSPI.cpp.All.pdf

// See SetupX_Template.h for all options available
#define USER_SETUP_ID 21
#define DISABLE_ALL_LIBRARY_WARNINGS

#define ILI9488_DRIVER

//#define TFT_INVERSION_OFF

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    5  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1 // using EN (aka RST) so do this

// #define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
// #define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
// #define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
// #define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
// #define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
// #define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
// #define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
// #define SMOOTH_FONT

// #define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// this seems to work as well:
//#define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY  2500000
</details>

<details>
<summary>platformio.ini</summary>
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_port = /dev/ttyUSB0
monitor_filters = esp32_exception_decoder
build_flags = -DBOARD_HAS_PSRAM
              -mfix-esp32-psram-cache-issue
lib_deps = 
	bodmer/TFT_eSPI@^2.5.33
	SPI
debug_tool = esp-prog
debug_init_break = tbreak setup
</details>

The font used (reduced character set exported from fontforge): [URWGothic_Book.h
](https://caseyconnor.org/pub/font/ofrissue/URWGothic_Book.h)

Bodmer, 2023-12-15:
I ran the example code provided on an ESP32 Dev Module 10 times and it did not crash. I then modified the example to the following and it runs without crashing:

```
#include <Arduino.h>
#include "SPI.h"
#include <TFT_eSPI.h>
#include "URWGothic_Book.h"
#include "OpenFontRender.h"
static OpenFontRender ofr;
static TFT_eSPI tft = TFT_eSPI();

void setup()
{
  pinMode(4, OUTPUT); // screen blanking pin
  digitalWrite(4, HIGH); // HIGH=screen on
  tft.init();
  tft.fillScreen(TFT_BLACK);
  ofr.setDrawer(tft);
  ofr.loadFont(URWGothic_Book, sizeof(URWGothic_Book)); // must be this font
  ofr.setFontSize(12); // 12 crashes, 13 crashes, 40 does not, 5 does not
  ofr.setAlignment(Align::TopCenter); // must be *Center to make it crash
  ofr.drawString("-7.0d", 40, 294, 0xec91, TFT_BLACK); // this color not important, x coord important, string matters vis a vis font size
  ofr.drawString("-6.0d", 70, 294, 0xec91, TFT_BLACK); // color important, x coord important
  ofr.unloadFont();
}

void loop() {
  tft.fillScreen(TFT_BLACK);
  ofr.loadFont(URWGothic_Book, sizeof(URWGothic_Book)); // must be this font
  ofr.setFontSize(12); // 12 crashes, 13 crashes, 40 does not, 5 does not
  ofr.setAlignment(Align::TopCenter); // must be *Center to make it crash
  ofr.drawString("-7.0d", 40, 294, 0xec91, TFT_BLACK); // this color not important, x coord important, string matters vis a vis font size
  ofr.drawString("-6.0d", 70, 294, 0xec91, TFT_BLACK); // color important, x coord important
  ofr.unloadFont();
  delay(200);
}

```
The only way I could get it to crash was by loading the font again without first using unload(). I think that is an issue with OFR not the TFT_eSPI library, it probably does not check if a font is already loaded and does not release the memory used from the previous load. However that seems to be a separate issue to the one you have.

Since I cannot get your sketch to crash as written I am not able to investigate further.

Bodmer, 2023-12-15:
I notice your red power LED changes brightness significantly in th evideo, this suggest the power supply is not able to cope at boot time with the processor load and display current demands, this may be the reason the processor reboots.

chconnor, 2023-12-15:
Thanks for looking at it!

May I ask how closely you were able to match the hardware for testing? ESP32/WROVER/ILI9488?

On further inspection, it seems like the screen is "crashing", not the ESP32. When this "crash" happens, all the threads seem to keep running, I can turn the backlight on/off, the touch screen still works, etc. The ESP32 program is functioning normally, as far as I can tell. (Embarrassed that I didn't realize that sooner.)

Tracing the issue further with the debugger, it always happens at [line 3772 in TFT_eSPI](https://github.com/Bodmer/TFT_eSPI/blob/9ab133f810c45c0b2f10c0089d5fd4aeceb23160/TFT_eSPI.cpp#L3772), specifically the tft_Write_8(TFT_RAMWR) call (I split the line to find out which of those was the trigger) -- that call makes the screen crash maybe 1 out of 8 times for me (when the line 1418 conditional breakpoint is as described in the first post, and only with the very specific demo code.) Does this maybe point to an issue with the screen hardware/firmware?

**Here's a weird finding**... on a whim, on line 3772 I added a delay(20) between the DC_C; and tft_Write_8() calls to see if it would change anything. It did not help, but check this video out; the "point of crashing" is always when it finishes drawing the 0 in "-6.0d". That's always where it will crash, if indeed it crashes. Here is a run where right when it gets to the bottom of the 0, _instead_ of crashing, you will notice that the overall brightness of the screen jumps up slightly (!?):

https://www.youtube.com/shorts/6ZwonP0g_NE

...i.e. in lieu of crashing, this suspicious brightness shift happens... I don't know what to make of that except that either the screen module itself is buggy and that's just a side effect of corrupted internal state, or if somehow the code driving it was confusing it. Any thoughts on that?

I don't think it's a PSU issue -- the red light you mentioned doesn't actually change, that's just an artifact of the auto-iris on the cell camera. It's connected via USB in the video, but when I try with a standalone 5V/3A PSU the same issue happens. Also note how sensitive the issue is to the particulars; e.g. the second drawString happens at x,y = 70,294. If I change that to 71, 294, it never crashes. (Also, my normal application runs for days without crashing or issue, using sensors, wifi, drawing lots of stuff, touch screen, etc, so if it was just a power thing I would think it would show up at other times.)

Bodmer, 2023-12-16:
My test setup is stable and has no problems, ESP32 DevkitC SPI ILI9488 display. The only differences I can see are:

Arduino IDE
ESP32 board package 2.0.14
TFT CS is on pin 15 not 5 (PCB in use)
No debugging switched on

Setup:
```
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO 

#define TFT_BL   32            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  16000000
#define SPI_TOUCH_FREQUENCY  2500000
```
Since I am not able to reproduce the problem I can only advise that some hardware is swapped out and wiring checked (signal, power and GND etc) for short connections to maintain signal integrity. Make sure there is not unusual compiler optimisations running and it may be a good idea to actually turn off debugging in case the RTOS is doing something in the background that corrupts memory in particular circumstances.

chconnor, 2023-12-16:
Well, I think you nailed it; long story short, I wrapped MISO, MOSI, and SCLK in grounded copper foil and it seems to be stable now. If I get other wires near the display-related wires (e.g. SDA of the I2C bus) there is still the occasional crash, but it does seem to be a wire routing issue.

Ironically, using short wires might have contributed: many of my connections are so short that it causes various unrelated wiring to be excessively close together :-).

Sorry to take the time. If I discover that this isn't a real solution I'll post with any more relevant info. I appreciate the time, thanks.

I'll post your crash demo code (loading fonts multiple times) over at OFR unless you were planning to?

commit cbf06d7a (2023-12-22): Fix #3036

Remove SPI  LL HAL functions that seem to be broken in latest STM32duino board package.

Replace with low level register access writes and reads.

This change also has significant performance benefits for SPI display update speed!
closed 2023-12-16 by chconnor