Bodmer/TFT_eSPI #3037: old calibrating code

BananaGrapeJuice, 2023-12-14:
old calibrate does not work anymore for some reason i have done everything correctly. when i use this the calibration just automatically done it for me after that i cant do anything

void touch_calibrate()
{
  uint16_t calData[5];
  uint8_t calDataOK = 0;

  // check file system exists
  if (!LittleFS.begin()) {
    Serial.println("Formating file system");
    LittleFS.format();
    LittleFS.begin();
  }

  // check if calibration file exists and size is correct
  if (LittleFS.exists(CALIBRATION_FILE)) {
    if (REPEAT_CAL)
    {
      // Delete if we want to re-calibrate
      LittleFS.remove(CALIBRATION_FILE);
    }
    else
    {
      File f = LittleFS.open(CALIBRATION_FILE, "r");
      if (f) {
        if (f.readBytes((char *)calData, 14) == 14)
          calDataOK = 1;
        f.close();
      }
    }
  }

  if (calDataOK && !REPEAT_CAL) {
    // calibration data valid
    tft.setTouch(calData);
  } else {
    // data not valid so recalibrate
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(20, 0);
    tft.setTextFont(2);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);

    tft.println("Touch corners as indicated");

    tft.setTextFont(1);
    tft.println();

    if (REPEAT_CAL) {
      tft.setTextColor(TFT_RED, TFT_BLACK);
      tft.println("Set REPEAT_CAL to false to stop this running again!");
    }

    tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.println("Calibration complete!");

    // store data
    File f = LittleFS.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calData, 14);
      f.close();
    }
  }
}

Bodmer, 2023-12-14:
This may be due to a high Z value. Can you run the Test_Touch_Controller sketch and post the serial output here, with and without the screen being touched.

BananaGrapeJuice, 2023-12-15:
> This may be due to a high Z value. Can you run the Test_Touch_Controller sketch and post the serial output here, with and without the screen being touched.

i got this

14:38:56.412 -> // Use this calibration code in setup():
14:38:56.412 ->   uint16_t calData[5] = { 1, 1, 4095, 1, 0 };
14:38:56.412 ->   tft.setTouch(calData);

Bodmer, 2023-12-15:
Can you run the [Test_Touch_Controller](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller) sketch and post the serial output here.

BananaGrapeJuice, 2023-12-15:
> Can you run the [Test_Touch_Controller](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller) sketch and post the serial output here.

```
7:05:18.781 -> x: 0     y: 4095     z: 359 
17:05:19.006 -> x: 0     y: 4095     z: 361 
17:05:19.270 -> x: 0     y: 4095     z: 360 
```

Bodmer, 2023-12-15:
What are the readings when the screen is touched? Example here, not touched, then touched. This verifies the controller is being read. Your values will be different:
```
x: 0     y: 3519     z: 14 
x: 0     y: 3538     z: 19 
x: 2355     y: 1679     z: 1986 
x: 2359     y: 1676     z: 2034 
```

Since it used to work, what has been changed to make it fail?

Also what processor? Which version of TFT_eSPI, which version of board pacakge.

BananaGrapeJuice, 2023-12-16:
> which version of board pacakge.

im using esp32 and the version are all latest ones im encountering is the calibration function after that i cant seem to do anything but if i remove this function void touch_calibrate() all the hitbox does not match with the image drawn into the screen i can press them at a different location 

```
void TFT_eSPI::calibrateTouch(uint16_t *parameters, uint32_t color_fg, uint32_t color_bg, uint8_t size){
  int16_t values[] = {0,0,0,0,0,0,0,0};
  uint16_t x_tmp, y_tmp;

  for(uint8_t i = 0; i<4; i++){
    fillRect(0, 0, size+1, size+1, color_bg);
    fillRect(0, _height-size-1, size+1, size+1, color_bg);
    fillRect(_width-size-1, 0, size+1, size+1, color_bg);
    fillRect(_width-size-1, _height-size-1, size+1, size+1, color_bg);

    if (i == 5) break; // used to clear the arrows
    
    switch (i) {
      case 0: // up left
        drawLine(0, 0, 0, size, color_fg);
        drawLine(0, 0, size, 0, color_fg);
        drawLine(0, 0, size , size, color_fg);
        break;
      case 1: // bot left
        drawLine(0, _height-size-1, 0, _height-1, color_fg);
        drawLine(0, _height-1, size, _height-1, color_fg);
        drawLine(size, _height-size-1, 0, _height-1 , color_fg);
        break;
      case 2: // up right
        drawLine(_width-size-1, 0, _width-1, 0, color_fg);
        drawLine(_width-size-1, size, _width-1, 0, color_fg);
        drawLine(_width-1, size, _width-1, 0, color_fg);
        break;
      case 3: // bot right
        drawLine(_width-size-1, _height-size-1, _width-1, _height-1, color_fg);
        drawLine(_width-1, _height-1-size, _width-1, _height-1, color_fg);
        drawLine(_width-1-size, _height-1, _width-1, _height-1, color_fg);
        break;
      }

    // user has to get the chance to release
    if(i>0) delay(1000);

    for(uint8_t j= 0; j<8; j++){
      // Use a lower detect threshold as corners tend to be less sensitive
      while(!validTouch(&x_tmp, &y_tmp, Z_THRESHOLD/2));
      values[i*2  ] += x_tmp;
      values[i*2+1] += y_tmp;
      }
    values[i*2  ] /= 8;
    values[i*2+1] /= 8;
  }

  // from case 0 to case 1, the y value changed. 
  // If the measured delta of the touch x axis is bigger than the delta of the y axis, the touch and TFT axes are switched.
  touchCalibration_rotate = false;
  if(abs(values[0]-values[2]) > abs(values[1]-values[3])){
    touchCalibration_rotate = true;
    touchCalibration_x0 = (values[1] + values[3])/2; // calc min x
    touchCalibration_x1 = (values[5] + values[7])/2; // calc max x
    touchCalibration_y0 = (values[0] + values[4])/2; // calc min y
    touchCalibration_y1 = (values[2] + values[6])/2; // calc max y
  } else {
    touchCalibration_x0 = (values[0] + values[2])/2; // calc min x
    touchCalibration_x1 = (values[4] + values[6])/2; // calc max x
    touchCalibration_y0 = (values[1] + values[5])/2; // calc min y
    touchCalibration_y1 = (values[3] + values[7])/2; // calc max y
  }

  // in addition, the touch screen axis could be in the opposite direction of the TFT axis
[... 398 lines cut ...]
    tft.println("Touch corners as indicated");

    tft.setTextFont(1);
    tft.println();

    if (REPEAT_CAL) {
      tft.setTextColor(TFT_RED, TFT_BLACK);
      tft.println("Set REPEAT_CAL to false to stop this running again!");
    }

    tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.println("Calibration complete!");

    // store data
    File f = LittleFS.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calData, 14);
      f.close();
    }
  }
}

```

BananaGrapeJuice, 2023-12-16:
I just completely remove the touch_calibrate() and just add this now its working 

```
#define REPEAT_CAL true
#define TS_MINX 200
#define TS_MINY 200
#define TS_MAXX 3800
#define TS_MAXY 3900

#define TFT_WIDTH  320
#define TFT_HEIGHT 480

void loop() {
  static uint32_t scanTime = millis();
  uint16_t t_x, t_y;  // To store the touch coordinates

  // Scan keys every 50ms at most
  if (millis() - scanTime >= 20) {
    bool pressed = tft.getTouchRaw(&t_x, &t_y);

    // Convert raw touch coordinates to TFT's native orientation
    uint16_t displayX = map(t_x, TS_MINX, TS_MAXX, 0, TFT_WIDTH);
    uint16_t displayY = TFT_HEIGHT - map(t_y, TS_MINY, TS_MAXY, 0, TFT_HEIGHT);

    scanTime = millis();
    for (uint8_t b = 0; b < buttonCount; b++) {
      if (pressed) {
        if (btn[b]->contains(displayX, displayY)) {
          btn[b]->press(true);
          btn[b]->pressAction();
        } else {
          btn[b]->press(false);
          btn[b]->releaseAction();
        }
      } else {
        btn[b]->press(false);
        btn[b]->releaseAction();
      }
    }
```

closed 2023-12-16 by BananaGrapeJuice