Bodmer/TFT_eSPI #3038: ST7789T3 ESP32

bbuster2, 2023-12-14:
I have bought a 1.83" ST7789T3 display but cannot get it working.

Display is bought here:
https://nl.aliexpress.com/item/1005005565821377.html?spm=a2g0o.order_list.order_list_main.11.205779d2rzxJEm&gatewayAdapt=glo2nld

Schematic:
[image]

In platformio, i use these biuld_flags:

```
build_flags = 
	-D USER_SETUP_LOADED=1
	-D ST7789_DRIVER=1
	-D SMOOTH_FONT=1
	-D TFT_MISO=-1
	-D TFT_MOSI=23
	-D TFT_SCLK=18
	-D TFT_CS=-1
	-D TFT_DC=2
	-D TFT_RST=4
	-D SPI_FREQUENCY=40000000
	-D SPI_READ_FREQUENCY=20000000
```

The display stays "black". I can control the backlight but i cannot get text or an image on the display.
I have the reset pin high (but also tried low), and the CS port connected to ground.

Is this display incompatible with the ESP32 or am i doing something wrong?

Bodmer, 2023-12-14:
Try using the CS line on a GPIO pin. The reason is that the ST7789 is easy to lock up with garbage commands during power up.

Also run the Read_User_Setup sketch and check the output is correct, this shows whether the correct settings are being picked up.

bbuster2, 2023-12-14:
Thank you for your reply. Adding the CS pin didn't make a difference (also connected to the ESP32).

Here is my output:
[code]
TFT_eSPI ver = 2.5.41
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 320

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 27
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4

Font GLCD   loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]

bbuster2, 2023-12-18:
Any idea?

Bodmer, 2023-12-18:
The Read_User_Setup output looks good. Check the connections are sound and correct. Check the supply voltage to the display is correct.  Software options to try:

1. Use the alternative replacement driver, viz.  -D ST7789_2_DRIVER=1
2. Add option -D TFT_SPI_MODE SPI_MODE0=1
3. Reduce the SPI frequency to 27MHz

You also need to define the width and height to get the coordinate offsets correct, but that would not stop the display working.
-D TFT_WIDTH=240
-D TFT_HEIGHT=280

bbuster2, 2023-12-19:
I tried but the same result.
I couldn't get -D TFT_SPI_MODE SPI_MODE0=1 to work
When i added that, i got the following error:

> *** [.pio/build/esp32dev/firmware.elf] Implicit dependency `/Users/kenzo/.platformio/platforms/espressif32/builder/SPI_MODE0=1' not found, needed by target `.pio/build/esp32dev/firmware.elf'.

```
-D USER_SETUP_LOADED=1
	-D ST7789_2_DRIVER=1
	; -D TFT_SPI_MODE SPI_MODE0=1
	-D TFT_SPI_MODE=1
	-D SPI_MODE0=1

	-D DISPLAY_WIDTH=280
	-D DISPLAY_HEIGHT=240
	-D TFT_WIDTH=280
	-D TFT_HEIGHT=240
	-D SMOOTH_FONT=1
	-D TFT_MISO=-1
	-D TFT_MOSI=23
	-D TFT_SCLK=18
	-D TFT_CS=27
	-D TFT_DC=2
	-D TFT_RST=4
	-D TFT_WIDTH=240
	-D TFT_HEIGHT=280
	-D SPI_FREQUENCY=27000000
	-D SPI_READ_FREQUENCY=20000000
```

But i am not sure if that can work. The display is still black

bbuster2, 2023-12-19:
I tried but the same result.
I couldn't get -D TFT_SPI_MODE SPI_MODE0=1 to work
When i added that, i got the following error:

> *** [.pio/build/esp32dev/firmware.elf] Implicit dependency `/Users/{username}/.platformio/platforms/espressif32/builder/SPI_MODE0=1' not found, needed by target `.pio/build/esp32dev/firmware.elf'.

```
-D USER_SETUP_LOADED=1
	-D ST7789_2_DRIVER=1
	; -D TFT_SPI_MODE SPI_MODE0=1
	-D TFT_SPI_MODE=1
	-D SPI_MODE0=1

	-D DISPLAY_WIDTH=280
	-D DISPLAY_HEIGHT=240
	-D TFT_WIDTH=280
	-D TFT_HEIGHT=240
	-D SMOOTH_FONT=1
	-D TFT_MISO=-1
	-D TFT_MOSI=23
	-D TFT_SCLK=18
	-D TFT_CS=27
	-D TFT_DC=2
	-D TFT_RST=4
	-D TFT_WIDTH=240
	-D TFT_HEIGHT=280
	-D SPI_FREQUENCY=27000000
	-D SPI_READ_FREQUENCY=20000000
```

But i am not sure if that can work. The display is still black

bbuster2, 2023-12-28:
My display looks really similar to this: https://www.buydisplay.com/1-83-inch-ips-tft-lcd-display-240x280-round-circle-screen-for-smart-watch

I see here the specifications show it is a **NV3030B** IC.
Can this work?

It is also 4 pin SPI

Bodmer, 2023-12-28:
That display may look similar but that does not mean yours has that NV3030B controller. You have to assume the vendors information is correct.

I made a mistake above, the option should read:

-D TFT_SPI_MODE=SPI_MODE0

bbuster2, 2023-12-29:
Thank you for your reply. Unfortunately i still cannot get anything on the display. It stays black

Maybe a really stupid question, but i don't see any chip on the display itself. It is just the screen with the 13 pin connector. Am i not missing something? (The driver chip?)

bbuster2, 2024-01-10:
Any idea?