Bodmer/TFT_eSPI #3040: first char lost when using drawstring after using SD function

sjorsvr, 2023-12-15:
description of the problem

I'm using a ESP32c3 module and a ST7789 driven tft display using SPI 
I use the Arduino platform 2.1.1
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
borads package esp32 by espressif 2.0.14
in my sketch I use the TFT-eSPI (2.5.34)  library and SD library in the esp32 package
the display and sd card work properly exept for :
the problem is that when I use the drawString function after a SD function the first character of the printed string is blank instead of the first character
for example (part of the whole sketch, see below)

........
    listDir(SD, "/", 2); //calls a SD function
  tft.drawString("before writeFile", 0, 150, UseFont);  //"b" GETS LOST / RESULTS IN A BLANK "e" is printed at correct position
  tft.drawString("before writeFile2", 0, 180, UseFont);  //"b" is NOT lost
    writeFile(SD, "/hello.txt", "Hello ");
  tft.drawString("writeFile done",00, 210, UseFont);  //"w" GETS LOST
  tft.drawString("writeFile done",00, 240, UseFont);  //"w" is NOT lost

output on the display :
 efore write
before write
 riteFile done
writeFile done

3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

my sketch 

```
/*
 * dit is een kopie van esp32c3 example SD_Test
 * SD_Test werkt ok met cs SD aan 7
 * cs sd aan 7 en cs tft aan 1 (in mysetupesp32c3)
 */

#include "FS.h"
#include "SD.h"
#include "SPI.h"

#include <TFT_eSPI.h>
//#include <SPI.h>
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

const byte pin1 = 1;
byte UseFont;
byte cardok;

void setup(){
  pinMode(pin1, OUTPUT);
  digitalWrite(pin1,HIGH);

  Serial.begin(115200);
  delay(5000);
  Serial.println("Started");

  if(!SD.begin()){
      Serial.println("Card Mount Failed");
      cardok = 0;
      return;
  } else {
      Serial.println("Card Mount OK");
      cardok = 1;
  }

  // display ST7735
  tft.init(); // tft.begin(); zijn hetzelfde
  tft.setRotation(0);
  tft.fillScreen(TFT_DARKBLUE);
  tft.setTextColor(TFT_YELLOW, TFT_DARKBLUE);
  tft.setTextSize(1);
  //  tft.setTextWrap(false, true); //werkt niet met drawstring
  UseFont = 4;
  tft.drawString("display started", 0, 30, UseFont);  

  if (cardok == 0) {
    tft.drawString("card failed", 0, 0, UseFont);  
  } else{
    tft.drawString("card ok", 0, 0, UseFont);  
  }

    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

//    Serial.print("SD Card Type ");
    tft.drawString("sd card type", 0, 60, UseFont);  
    if(cardType == CARD_MMC){
//        Serial.println("MMC");
  tft.drawString("mmc",120, 60, UseFont);  
    } else if(cardType == CARD_SD){
//        Serial.println("SDSC");
  tft.drawString("sdsc", 120, 60, UseFont);  
    } else if(cardType == CARD_SDHC){
//        Serial.println("SDHC");
  tft.drawString("sdhc", 120, 60, UseFont);  
    } else {
//        Serial.println("UNKNOWN");
  tft.drawString("unknown", 120, 60, UseFont);  
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);
  tft.drawString("cardSize done", 0, 90, UseFont);  

    listDir(SD, "/", 0);
  tft.drawString("listdir done", 0, 120, UseFont);  //"l" gets lost

    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
//  tft.setCursor(0,120);
//  tft.println("before File");
  tft.drawString("before writeFile", 0, 150, UseFont);  //"b" gets lost
//  tft.setCursor(0,150);
//  tft.println("before File2");
  tft.drawString("before writeFile2", 0, 180, UseFont);  //"b" is NOT lost
    writeFile(SD, "/hello.txt", "Hello ");
//  tft.setCursor(0,180);
//  tft.println("writefile done");
  tft.drawString("writeFile done",20, 210, UseFont);  //"w" gets lost
  tft.drawString("writeFile done",20, 220, UseFont);  //"w" is NOT lost
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
  tft.drawString("before read file", 0, 230, UseFont);  

    readFile(SD, "/foo.txt");

  tft.drawString("after read file", 0, 250, UseFont);  
    testFileIO(SD, "/test.txt");
  tft.drawString("before no bytes", 0, 270, UseFont);  
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
  tft.drawString("done", 0, 290, UseFont);  

}

void loop(){

}

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
    } else {
        Serial.println("rmdir failed");
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
    file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("File renamed");
    } else {
        Serial.println("Rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    File file = fs.open(path);
    static uint8_t buf[512];
    size_t len = 0;
    uint32_t start = millis();
    uint32_t end = start;
    if(file){
        len = file.size();
        size_t flen = len;
        start = millis();
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            len -= toRead;
        }
        end = millis() - start;
        Serial.printf("%u bytes read for %u ms\n", flen, end);
        file.close();
    } else {
        Serial.println("Failed to open file for reading");
    }

    file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }

    size_t i;
    start = millis();
    for(i=0; i<2048; i++){
        file.write(buf, 512);
    }
    end = millis() - start;
    Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
    file.close();
}
```

Bodmer, 2023-12-15:
I don't have the exact same setup but I tested on an ESP32 with an ILI9488 and the sketch runs fine.

One peculiarity of some ST7789 displays is that they only work in SPI mode 3, so that may be a problem when the SPI bus switches from mode 0 (used by SD) and mode 3.  Can you try changing [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L133) to set to mode 0 (as in [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L135))

Another possibility is a clash between SPI instances. Try initialising the tft first then giving the SD library the SPI instance that TFT_eSPI is using:
```
  // display ST7735
  tft.init(); // tft.begin(); zijn hetzelfde
  tft.setRotation(0);
  tft.fillScreen(TFT_BLUE);
  tft.setTextColor(TFT_YELLOW, TFT_BLUE);
  tft.setTextSize(1);
  //  tft.setTextWrap(false, true); //werkt niet met drawstring
  UseFont = 4;
  tft.drawString("display started", 0, 30, UseFont);  

  if(!SD.begin(sd_cs_pin, tft.getSPIinstance())){
      Serial.println("Card Mount Failed");
      cardok = 0;
      return;
  } else {
      Serial.println("Card Mount OK");
      cardok = 1;
  }

```

sjorsvr, 2023-12-15:
Hi
thanks for quick reply
trying different order of sd and tft init resultet earlier an a error
so order is ok
I will try spi advises
thanks

sjorsvr, 2023-12-16:
Hi
thanks for the quick reply
I'm going to test these possibilities in the evening.

I already found that init tft first and then init sd did not work but I will try your suggestion to reuse the spi instance

thanks again
I will let you know the results

regards
Sjors

sjorsvr, 2023-12-17:
Hi Bodmer,

changing the mode to 0 worked ok : display contents are now ok.
I still need to init SD first and later ini TFT.

The other suggestion : init TFT first then init SD using the tft spi instance did not work : card init failed.

The solution to change mode to 0 is ok for me.
Thank you for the help.

regards
Sjors

closed 2023-12-17 by Bodmer