Bodmer/TFT_eSPI #3044: not get woriking esp32 and st7789V2

sithco952, 2023-12-16:
I just get the back light on and it is black.

1.**The Read user setup:**
TFT_eSPI ver = 2.5.34
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 170
Display height = 320

R0 x offset = 35
MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 5
TFT_DC   = GPIO 16
TFT_RST  = GPIO 17

TFT_BL           = GPIO 4
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]

2.**my settings:**

// See SetupX_Template.h for all options available
#define USER_SETUP_ID 18

#define ST7789_DRIVER

#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 display only

#define TFT_WIDTH  170
#define TFT_HEIGHT 320

// If colours are inverted (white shows as black) then uncomment one of the next
// 2 lines try both options, one of the options should correct the inversion.
//#define TFT_INVERSION_ON
 #define TFT_INVERSION_OFF

// For ST7789 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display
// #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// My ST7789 display has TCT_CS wired permananently low so the pin is not defined here

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
//#define TFT_DC   PIN_D3  // Data Command control pin
//#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

#define CGRAM_OFFSET     

#define TFT_MOSI            23
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST            17

#define TFT_BL          4  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

//#define SPI_FREQUENCY  10000000
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

//#define SPI_TOUCH_FREQUENCY  2500000

//#define SUPPORT_TRANSACTIONS

3. **What i get from test the lcd read and write**

Pixel value written = 4
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 8
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 10
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 20
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 40
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 80
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 100
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 200
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 400
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 800
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 1000
 Pixel value read    = 0
 ERROR                 ^^^^
 Pixel value written = 2000

Bodmer, 2023-12-16:
Check your connections from each GPIO to the display, they do not look right, for example SDA (MOSI) should connect to GPIO 23.

sithco952, 2023-12-17:
> Check your connections from each GPIO to the display, they do not look right, for example SDA (MOSI) should connect to GPIO 23.

I have worked on this project 3 day and night.  Untill, I realised the quality of the wire is not good. now I get this little screen working pretty good. thank you Bodmer and your effort of this cool library!

closed 2023-12-28 by Bodmer