Bodmer/TFT_eSPI #3046: Not work if STM32 board version > 2.6.0

CodeNameHawk, 2023-12-17:
After update STM32 board version to > 2.6.0 library stop working.
(Working on 2.50, 2.60. Not work on 2.7.0, 2.7.1. I only see the backlight. I can’t see the screen initializing)
IDE arduino
STM32F103C8
ILI9488 SPI
```
TFT_eSPI ver = 2.5.34
Processor    = STM32
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO -57
MISO    = GPIO -58
SCK     = GPIO -59

>>>>> Note: STM32 pin references above D15 may not reflect board markings <<<<<
TFT_CS   = D16
TFT_DC   = D31
TFT_RST  = D30

TFT_BL           = D10
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
```

Bodmer, 2023-12-18:
It looks like some breaking changes have been made to the board package, so stick with 2.6.0 until I have time to investigate.

CodeNameHawk, 2023-12-18:
Ok. Thx.

JFBow, 2023-12-21:
Hello,
I have a similar problem. I upgraded from stm32duino version 2.6.0 to version 2.7.1 because version 2.7.0 generated compilation errors for SdFat. 
With 2.7.1 no more errors for SdFat, it works, but it no longer works with TFT-eSPI (version 2.5.34), the screen remains dark.
If I switch back to stm32duino 2.6.0 the screen works again (without changing the wiring).

I work with a BlackPill F411CE, a TFT ST7789 240x240 screen on SPI 1.

Bodmer, 2023-12-22:
I have updated the Github master to fix this.

I have adopted low level register writes, this has been teted on an STM32F401 board with board package 2.7.1 but it should work for other boards and core versions. This change is backwards compatible with core 2.6.0.

A key advantage of this change is a significant performance boost as the low level HAL drivers in the core package were surprisingly boated and slow.

Please check on your hardware and report back and raise a new issue if there are problems.

closed 2023-12-22 by Bodmer

CodeNameHawk, 2023-12-22:
I checked it on the sketch UTFT_Demo_480x320 - it works. Thank you.