Bodmer/TFT_eSPI #3054: ILI9486 with Blackpill(stm32f411CE)  display turns to grey and flicker

Kps131200, 2023-12-21:
#define USER_SETUP_ID 32

#define STM32
#define RPI_DISPLAY_TYPE
// #define ILI9341_DRIVER
#define ILI9486_DRIVER

#define TFT_CS   PA4
#define TFT_DC   PA3
#define TFT_RST  PA2

#define TFT_MOSI PA7
#define TFT_MISO PA6
#define TFT_SCLK PA5

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  36000000   // 36MHz SPI clock
//#define SPI_FREQUENCY  18000000   // 18MHz SPI clock

#define SPI_READ_FREQUENCY  12000000

#define SPI_TOUCH_FREQUENCY  2500000
Here is my user setup i have checked wiring and power supply still every time i upload code it tures grey and then flicker

Bodmer, 2023-12-21:
Post a link to the display you have, it may not be supported by the library.
Some RPi interface displays require a lower SPI frequency, reduce the SPI frequency to 20MHz.
Run the Read_User_Setup sketch and check the output.

Bodmer, 2023-12-22:
Alsom which board package version are you using?

Kps131200, 2023-12-22:
Sorry for the trouble but there was a problem with the blackpill board. But now i am having issues with the touch can you tell me the steps to start touch control with stm32f411(blackpill)

Kps131200, 2023-12-22:
https://www.waveshare.com/3.5inch-rpi-lcd-a.htm. Display link
Board package number:-2.6.0

Bodmer, 2023-12-27:
Glad to hear it is working now. To add touch capability, connect the touch controller CS pin to a GPIO and then add that to the setup, for example:
#define TOUCH_CS PA1

closed 2023-12-27 by Bodmer