Bodmer/TFT_eSPI #3059: Identify type of display and runtime selection of driver

andreash-esp, 2023-12-24:
**Issue**
I have two similar 2.8" SPI ILI9341 (that's what they were sold as) displays that work both with ILI9341_DRIVER, ILI9341_2_DRIVER and ST7789_2_DRIVER, but not with the ST7789_DRIVER. 
Is there a way to find out for sure which is the correct driver, and possible select it during runtime?

The displays look like this: http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807
The older one is 50x86mm with a "2.8 TFT SPI 240x320 V1.2" label
The newer one is 50x82mm with a "2.8" TFT 240xRGBx320 v1.1" label and ROHS marking
I could not find a driver IC on any of them. The passive components are the same on both boards

I'm quite confused as to why the ST7789_2_DRIVER is working but the ST7789_DRIVER is not.

**IDE (e.g. Arduino or PlatformIO)**
Arduino IDE

**TFT_eSPI library version**
TFT_eSPI@2.5.34

**Board package version (e.g. 2.0.3) available from the Boards Manager... menu**
2.0.4

**Procesor, e.g RP2040, ESP32 S3 etc**
ESP32 DevKit C v4

**TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.**
ILI9341, ILI9341_2 and ST7789_2

**Interface type (SPI or parallel)**
SPI

andreash-esp, 2023-12-24:
This is the newer display: https://www.aliexpress.com/item/4001282467099.html
also referenced here as ST7789V type: https://github.com/Bodmer/TFT_eSPI/discussions/2177

Bodmer, 2023-12-25:
The library does not support runtime selection of the display driver.

Select the driver that is written for that display type.

closed 2023-12-25 by Bodmer