Bodmer/TFT_eSPI #3072: STM32F405 SPI does nothing.

ToRoCu, 2023-12-31:
Hello Bodmer,
I'm trying to set up an ILI9341 display with STM32F405 with SPI2.
The problem is, no matter what settings I made in the `User_Setup.h` the SPI does nothing.
It does not matter if i try SPI1 or SPI2.
The DC and CS pins are working correctly.
I use the STM32duino Core from ST. I measured all pins from the uC with an oscilloscope while the `TFT_Starfield` test was running and only DC & CS do anything.
I tested with the Adafruit ILI9341 Lib and it worked fine.

[User_Setup.txt](https://github.com/Bodmer/TFT_eSPI/files/13799550/User_Setup.txt)

What am I doing wrong?

Here my Serial Print from `read_user_setup.ino`
01:08:17.520 -> [code]
01:08:17.520 -> TFT_eSPI ver = 2.5.34
01:08:17.520 -> Processor    = STM32
01:08:17.552 -> Transactions = Yes
01:08:17.586 -> Interface    = SPI
01:08:17.586 -> Display driver = 9341
01:08:17.619 -> Display width  = 240
01:08:17.652 -> Display height = 320
01:08:17.686 -> 
01:08:17.686 -> MOSI    = GPIO 31
01:08:17.686 -> MISO    = GPIO 30
01:08:17.719 -> SCK     = GPIO 29
01:08:17.719 -> 
01:08:17.719 -> >>>>> Note: STM32 pin references above D15 may not reflect board markings <<<<<
01:08:17.820 -> TFT_CS   = D39
01:08:17.820 -> TFT_DC   = D38
01:08:17.853 -> 
01:08:17.853 -> Font GLCD   loaded
01:08:17.886 -> Font 2      loaded
01:08:17.886 -> Font 4      loaded
01:08:17.919 -> Font 6      loaded
01:08:17.919 -> Font 7      loaded
01:08:17.953 -> Font 8      loaded
01:08:17.986 -> Smooth font enabled
01:08:17.986 -> 
01:08:17.986 -> Display SPI frequency = 27.00
01:08:18.032 -> [/code]

ToRoCu, 2024-01-12:
I found this "https://forum.arduino.cc/t/tft_espi-w-stm32f405-feather-and-ili9341-display/659387/29", but it has never been confirmed.
The SPI port still does not output anything...

Bodmer, 2024-01-16:
There is an issue with the latest STM32 board package (2.7.0 and 2.7.1), this has been corrected in the github master. Alternatively downgrade to STMduino board package 2.6.0

closed 2024-01-16 by Bodmer