Bodmer/TFT_eSPI #3074: Inverted (mirror) text on TFT shield for Arduino ili9486 parallel

visionSensor, 2024-01-02:
Hello, I have an issue with a TFT shield for Arduino.
Screen controller: ili9486
Microcontroller: ESP32
IDE: Arduino

Library version TFT_espi: 2.5.43
Data bus: 8-bit parallel
Problem: I have a 3.5-inch TFT shield (480X320) that uses an 8-bit parallel bus, ili9486 driver. The TFT works well with the ESP32, with good examples and colors. However, the problem arises when I display text with a certain screen rotation (setRotation(0);). The text appears inverted or mirrored.
I'm not sure if it's visible, but inverted text is appearing at the top of the screen:

[image]

Here is the result of a sketch that I wrote, in which I think the problem is more noticeable:

[image]

"Back of the TFT shield:

[image]

Help me, please, Bodmer.

Bodmer, 2024-01-05:
Provide a link to the display vendors page. The driver may not be an ILI9486.

Alternatively if may just be another configuration variant, in which case you may need to adjust the flip/mirror settings in the init code here:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Rotation.h

You may find that setRotation with values 4 to 7 corrects the problem.

visionSensor, 2024-01-05:
Hello Bodmer, thank you for the suggestion to try with the rotation (setRotation()). I tested it with this code in the Arduino IDE:

visionSensor, 2024-01-05:
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();  
void setup() {
  // put your setup code here, to run once:
tft.init();
tft.fillScreen(TFT_BLACK);
tft.setRotation(0);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(1);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(2);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(3);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(4);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(5);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(6);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
tft.fillScreen(TFT_BLACK);
tft.setRotation(7);
tft.setCursor(100,100);
tft.setTextColor(TFT_WHITE);
tft.setTextSize(1);
tft.println("Hello World!");
delay(1000);
}

void loop() {
  // put your main code here, to run repeatedly:

}

visionSensor, 2024-01-05:
The result, the text rotates, but it seems that only in the horizontal position, the text is displayed correctly.

[image]

Bodmer, 2024-01-16:
It seems you have a variant of the displays I have.

The TFT_MAD_MX/V/Y bits change the way the pixels are written to the display.  I have not seen correctly rendered characters written in reverse before so do check with the vendor that you have an ILI9486 display.

Unfortunately the only option is to try different settings in the rotation code.