Bodmer/TFT_eSPI #3076: First Compile Error

mghaheri, 2024-01-04:
Hi Bodmer,
I'm so interested in your library and downloaded it from Github, then added it to my Arduino IDE libraries and I had already added STM32duino library. When I tried to compile one of the examples (All-Free-Fonts-Demo) I received this error:
collect2.exe: error: ld returned 1 exit status
Then I edited setup file in its original location and recompiled. Nothing changed. I copied the setup file to my project path. Nothing changed. Added an include line for it to my code. And again nothing change. I couldn't find a good step by step procedure to use your library. Do I do anything wrong? How can I use the library? :(
Pleaaase help me. Is platform IO more suitable for your library or Arduino IDE?
Best regards and thanks in advance

xpresas, 2024-01-04:
Please edit this issue to follow the issue template. Its impossible to help you without more information.

[image]

Bodmer, 2024-01-05:
The error message implies that you have an issue with the build environment, this is something you will need to investigate as there error message could have so many causes.

closed 2024-01-05 by Bodmer

mghaheri, 2024-01-06:
Thank you so much for the answer
But as I didn't find a complete tutorial for using your library, I still have doubts about installing all needed libraries. Sorry for not obeying the template. I'll be thankful if you give me links of discussions and issues like mine.
Best regards