Bodmer/TFT_eSPI #3078: M5stickc Plus2 User_Setup.h

kazz2020, 2024-01-04:
Hi
I have a problem with setup User_Setup.h. I uncommented proper driver ST7789v2 changed pins but screen is black. Anyone have some expirience with this M5 device with Tft_espi?

xpresas, 2024-01-04:
I could have helped you if you would have used the proper issue template.

kazz2020, 2024-01-04:
> I could have helped you if you would have used the proper issue template.

So no help becouse of that?

xpresas, 2024-01-04:
> > I could have helped you if you would have used the proper issue template.
> 
> So no help because use of that?

Not because of that. Because if you would have used the template me and others included would have had basic knowledge on how to debug your problem. Now its impossible to know what, or how to fix the problem.

kazz2020, 2024-01-05:
> Not because of that. Because if you would have used the template me and others included would have had basic knowledge on how to debug your problem. Now its impossible to know what, or how to fix [...]

i hope my change will help

xpresas, 2024-01-05:
> i hope my change will help

I had quite a similar experience using a different ESP32. For unknown reasons my TFT screen did not work using the default GPIO23. I search the internet to no end. And I couldnt figure out why this exact board of mine (ESP32-WROOM-32U ch9102x) was not able to run the screen via VSPI pin. 
I had an older ESP32 dev kit board which worked fine with the default configuration.

By looking at the pinout to my specific board I saw that pin 13 on my board was HSPI pin. So I just changed the config and tested it out and it started showing stuff without a problem.

[image]

kazz2020, 2024-01-05:
> I had quite a similar experience using a different ESP32. For unknown reasons my TFT screen did not work using the default GPIO23. I search the internet to no end. And I couldnt figure out why this [...]

I will check it but what is strange I've changed pin numbers from plus to plus2 because it's different (docs from m5stack) pins and not working.

kazz2020, 2024-01-05:
For ili it was siple as f

Bodmer, 2024-01-05:
I do not provide support for the ESP32 Marauder software.

Compile and run test code provided by M% to make sure the module works.

Since the display is black, this implies the LED backlight is not working or switched on so you will need to investigate this yourself.

closed 2024-01-05 by Bodmer

kazz2020, 2024-01-05:
> I do not provide support for the ESP32 Marauder software.
> 
> Compile and run test code provided by M% to make sure the module works.
> 
> Since the display is black, this implies the LED backlight is not working or switched on so you will need to investigate this yourself.

No problem... buuuuut you can provide ST7789v2 support 😉

kazz2020, 2024-01-05:
> I had quite a similar experience using a different ESP32. For unknown reasons my TFT screen did not work using the default GPIO23. I search the internet to no end. And I couldnt figure out why this [...]

You are the expert 🙄 no you aren't I asked q you wrote piece of sh..... And you talk about template? Very shame

xpresas, 2024-01-05:
> You are the expert 🙄 no you aren't I asked q you wrote piece of sh..... And you talk about template? Very shame

You seem a bit mental. 
I just told you I went about to try to fix the similar problem. If you are not happy with that. Contact bodmer directly. Pay him a couple of hundred buck and stop crying like a little kid. Do you really expect people to work for you for free? Some people...

kazz2020, 2024-01-05:
> > > > > > > I could have helped you if you would have used the proper issue template.
> > > > > > 
> > > > > > 
> > > > > > So no help because use of that?
> > > > > 
> > > > > 
> > > > > Not because of that. Because if you would have used the template me and others included would have had basic knowledge on how to debug your problem. Now its impossible to know what, or how to fix the problem.
> > > > 
> > > > 
> > > > i hope my change will help
> > > 
> > > 
> > > I had quite a similar experience using a different ESP32. For unknown reasons my TFT screen did not work using the default GPIO23. I search the internet to no end. And I couldnt figure out why this exact board of mine (ESP32-WROOM-32U ch9102x) was not able to run the screen via VSPI pin.
> > > I had an older ESP32 dev kit board which worked fine with the default configuration.
> > > By looking at the pinout to my specific board I saw that pin 13 on my board was HSPI pin. So I just changed the config and tested it out and it started showing stuff without a problem.
> > > ![image](https://private-user-images.githubusercontent.com/22633775/294597059-81baaee7-86e5-4559-9e36-425bb04e13db.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0OTIxNTcsIm5iZiI6MTcwNDQ5MTg1NywicGF0aCI6Ii8yMjYzMzc3NS8yOTQ1OTcwNTktODFiYWFlZTctODZlNS00NTU5LTllMzYtNDI1YmIwNGUxM2RiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA1VDIxNTczN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmYjgyZjFjYjViNGRhNmMwNTFhMzJmYjU3Y2U1OGQyY2ZkMjAzYmFlZDNjOGIzMDdmODZiOGI5MDRlNWFlOGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jDALNovzaPHRDuHo1ckWdgjr8uvT9zHE_tB925cq7nw)
> > 
> > You are the expert 🙄 no you aren't I asked q you wrote piece of sh..... And you talk about template? Very shame
> 
> You seem a bit mental. 
> I just told you how I went about to try to fix the similar problem. If you are not happy with that. Contact bodmer directly. Pay him a couple of hundred bucks and stop crying like a little kid. Do you really expect people to work for you for free? Some people...

In the future, don't act like you know something, because throwing random Google results proves that you're a loser. Do you iven know what is plus2? You teach others and you don't know shit yourself. Cheers mtf

xpresas, 2024-01-05:
I guess you not only unable to write in proper grammar, but at the same time you also have problems with reading too.
Cheers. 
Hope you burn your MCU!