Bodmer/TFT_eSPI #3081: readRect() somehow converts RGB 565 to RGB 332

Chineko-Koizumi, 2024-01-05:
**Issue description**
Downloading part of displayed screen from display memory to ESP32 memory, using readRect() somehow converts RBG range.

**Hardware** 

CPU: ESP-WROOM-32
LCD screen driver: ILI9341

**Library version**
 bodmer/TFT_eSPI 2.5.43

**example code to trigger behaviour** 

```
uint16_t tab[10000];
TFT_eSprite tempSprite(&tft);

//(...)

//Fill screen with BG image
  tft.setSwapBytes(true); 
  tft.pushImage(0, 0, MAX_IMAGE_WIDTH, MAX_IMAGE_HIGHT, (uint16_t *)Horo_image.pixel_data);
  delay(100);

  tft.readRect(0, 0, 100, 100, tab);
  
  tempSprite.createSprite(100, 100);
  tempSprite.pushImage(0,0,100,100, tab, 16U);
  tempSprite.pushSprite(5,170);
```

**Example code result on photo**
[image]

Bodmer, 2024-01-16:
readRect() stores colours as 16 bit byte swapped values in a 16 bit array. This allows pushrect() to be used without the overhead of swapping bytes.

closed 2024-01-16 by Bodmer

Chineko-Koizumi, 2024-01-17:
> readRect() stores colours as 16 bit byte swapped values in a 16 bit array. This allows pushrect() to be used without the overhead of swapping bytes.

So if I swap High byte and Low Byte in a "tab" array (from below code fragment) after data download from screen, it should be compatible with TFT_eSprite. Am I correct?
`uint16_t tab[10000];
TFT_eSprite tempSprite(&tft);

//(...)

//Fill screen with BG image
  tft.setSwapBytes(true); 
  tft.pushImage(0, 0, MAX_IMAGE_WIDTH, MAX_IMAGE_HIGHT, (uint16_t *)Horo_image.pixel_data);
  delay(100);

  tft.readRect(0, 0, 100, 100, tab);
  
  tempSprite.createSprite(100, 100);
  tempSprite.pushImage(0,0,100,100, tab, 16U);
  tempSprite.pushSprite(5,170);`

Chineko-Koizumi, 2024-01-20:
@Bodmer I really do not want to be insistent, but I did nothing inside code and somehow after changing cpu from esp32 to esp32S3 screen data fixed itself. I'm not sure why. 
[image]