Bodmer/TFT_eSPI #3086: Color Missing STM32F401CCU6 with 320QDT_9341

mannapijush1coretech, 2024-01-08:
### **Color Missing** and graphics drawing problem when using this library
the main problem is when i used RGB Color Mode then Blue is missing and if i use BGR then Red Is missing 

Arduino IDE using Latest Library Version 2.5.43
STM32F401CCU6 Black Pill
320QDT_9341 (i assumed this IL9341)
Running on Parallel Mode

Connection as defined in _Setup35_ILI9341_STM32_Port_Bus.h_
`
#define TFT_CS  PB7 // Chip select control pin
#define TFT_DC  PB8 // Data Command control pin
#define TFT_RST PB0 // Reset pin
#define TFT_WR  PB9 // Write strobe control pin
#define TFT_RD  PB1 // Read pin

#define TFT_D0 PA0 // 8-bit Port A parallel bus to TFT
#define TFT_D1 PA1
#define TFT_D2 PA2
#define TFT_D3 PA3
#define TFT_D4 PA4
#define TFT_D5 PA5
#define TFT_D6 PA6
#define TFT_D7 PA7
`
i tried all type of sketches provided in the example
_screenshot i am attaching for Arc_Meter_Demo
here is the display --
[image]

when using BGR Mode--
[image]

and when using RGB Mode --
[image]

i think last color is missing in the specified color order
kindly help for fixing this issue

Chineko-Koizumi, 2024-01-08:
Not sure if it is you case, but when I did try parallel mode on display that supported both 8 bit mode and 16 bit mode. I stumble across same color issue. I fact that was not TFT_eSPI library bug but screen hardware configuration. My screen out of the box supported only 16 bit and after some searching and asking provider I got reply that by default display supports only 16 bit so screen get only half of the information thus one color is visible (to change mode I had to solder resistor to other pad in the PCB). Check documentation if this display supports 8 bit out of the box, but my guess is that shorting this pads would change mode from 16 bits to 8 bits(Hi or Lo word)
[image]

mannapijush1coretech, 2024-01-09:
ok there is 2 jumper oe is **jp1** and **jp3**. i both soldered one by one and checked but the problem remains same!

Bodmer, 2024-01-16:
The library only supports 8 bit parallel with the STM32F processors.  You will need an 8 bit parallel display or modify your board to work in 8 bit mode. I am not able to advise how to do this as that is not a board I am familiar with. The vendor may be able to help.

closed 2024-01-16 by Bodmer