Bodmer/TFT_eSPI #3088: SPI_X not defined in latest version (2.5.34) in Arduino IDE.

vishwaa-atomic, 2024-01-08:
Hello,
I am following [this](https://wiki.seeedstudio.com/get_start_round_display/) for a xiao nrf52840 and a round display. When I run the hardwaretest example from the round display I get an error related to this library. I am using the latest version and using arduino ide. I have commented out the default and uncommented the setup 66 for xiao. Error (from the red highlight and below):
[Untitled document.pdf](https://github.com/Bodmer/TFT_eSPI/files/13861106/Untitled.document.pdf)
I have not included additional information, since it should be from the wiki, besides the version #s

Bodmer, 2024-01-16:
The nrf52840  is supported as a generic processor.

Make sure you have selected the correct processor in the IDE because SPI_X is specific to the RP2040 processor which is not used on your board.

closed 2024-01-16 by Bodmer

andriandreo, 2024-06-18:
I know the issue is closed, but I wanted to make a point:

> I get an error related to this library. I am using the latest version and using arduino ide. I have commented out the default and uncommented the setup 66 for xiao. Error (from the red highlight and [...]

I'm not sure why, when the selected board is the `mbed-enabled` version, this library is somehow trying to load stuff from the RP2040 implementation and you get the kind of errors reported above in this issue. This is not happening when using the NON-mbed version of the boards (just a few warnings, but it compiles and uploads to the board)...

Could we check some how a workaround for this weird behaviour in the case we wanted to use the `mbed-enabled` version of the boards?

Thanks!

andriandreo, 2024-06-25:
> Could we check some how a workaround for this weird behaviour in the case we wanted to use the `mbed-enabled` version of the boards?

Fork with workaround for `mbed-enabled` board version (i.e. `ARDUINO_MBED_ARCH` defined): https://github.com/andriandreo/TFT_eSPI