Bodmer/TFT_eSPI #3093: Bluepill STM32F103C8T6 and ST7789 only works by removing "#define STM32"

0tazul0, 2024-01-09:
HI there,
I'm facing a problem with "blupill (STM32F103C8T6)" to run "240x240 ST7789" Display. As recommended I used "Setup32_ILI9341_STM32F103" and modified this by removing
```
#define ILI9341_DRIVER
```
and adding
```
#define ST7789_DRIVER
#define TFT_RGB_ORDER TFT_BGR   //Because the Color wasn't right
```
But for two days I can't seems to get it work. I tried using the recommended connection and also by using other PINs for CS, DC, RST
Today I found out that if Comment out
```
//#define STM32
```
only then it works.

I'm not sure if it's a problem from the library or the problem is somewhere else. I'd appreciate it if you kindly look into this matter.
Thank you.

Arduino IDE version 1.8.19
TFT_eSPI version 2.5.34
Board package STMicroelectronics version 2.7.1 (Official Core)
Procesor blupill (STM32F103C8T6)
TFT driver ST7789
Interface type SPI

Setup File
[Setup32_ILI9341_STM32F103.zip](https://github.com/Bodmer/TFT_eSPI/files/13875907/Setup32_ILI9341_STM32F103.zip)

Example Code
[Test.zip](https://github.com/Bodmer/TFT_eSPI/files/13875912/Test.zip)

Bodmer, 2024-01-09:
This is due to breaking updates to the STM32 board package. This has been fixed as noted in #3046.

Your options are:

1. Commented out the #define STM32, but this will have a screen update performance impact.
2. Downgrade the board package to 2.6.0
3. Upgrade TFT_eSPI from the Github master and then use board package 2.7.1

In the longer term, option 3 is probably the better option.

closed 2024-01-09 by Bodmer

0tazul0, 2024-01-10:
Thank you very much for your reply. I went with the option 3 and the display worked and it's blazing fast.
But unfortunately after doing that "SD_Jpg" example from your "TJpg_Decoder" library didn't worked as expected. It was fine before, but now the color seems yollowish and screen tearing occurs and sometime the image can't be displayed in its entirety.
"Flash_Jpg" works fine.
Another thing is that "Flash_Jpg_DMA"  doesn't work, so I tried DMA test "boing_ball" the background was there but the ball never showed up.

Test Video:
https://github.com/Bodmer/TFT_eSPI/assets/59217560/890f47b9-fb7d-4ffa-8e25-4f7679aab52f

I'm using same SPI port for SD and TFT

Bodmer, 2024-01-12:
OK, looks like the new board package broke the DMA too.

The problem with jpegs might be down to the increased performance. I will investigate further.

In the meantime downgrade the board package to 2.6.0, the latest TFT_eSPI should still work with that board package.

Bodmer, 2024-01-13:
I think the incorrect colours for the jpeg should work, the master library has been updated.

DMA for SPI is still not working for board packages above 2.6.0