Bodmer/TFT_eSPI #3097: color conversion of readRect() different from readPixel() for ILI9488

chconnor, 2024-01-10:
Hello -- vscode + PlatformIO driving an ILI9488 ([DFR0669](https://wiki.dfrobot.com/3.5inches_480_320_TFT_LCD_Capacitive_Touchscreen_SKU_DFR0669)) on an ESP32-DevKitC-VIE, TFT_eSPI v2.5.33

When I use readPixel() it works fine, but when I use readRect() the colors seem wrong.

Read frequency doesn't seem to be the issue (10MHz works for readPixel, even 5MHz doesn't work for readRect).

I notice that in the code (judging by vscode highlighting based on #define variables), readPixel() for me comes to:

```
      uint8_t r = tft_Read_8();
      uint8_t g = tft_Read_8();
      uint8_t b = tft_Read_8();
      color = color565(r, g, b);
```

but readRect() comes to:

```
      // The 6 colour bits are in MS 6 bits of each byte but we do not include the extra clock pulse
      // so we use a trick and mask the middle 6 bits of the byte, then only shift 1 place left
      uint8_t r = (tft_Read_8()&0x7E)<<1;
      uint8_t g = (tft_Read_8()&0x7E)<<1;
      uint8_t b = (tft_Read_8()&0x7E)<<1;
      color = color565(r, g, b);
```

If I change the code to be like the readPixel version (just tft_Read_8() with no mask/shift) -- that fixes the problem and everything works as expected.

Chineko-Koizumi, 2024-01-11:
seems like my issue https://github.com/Bodmer/TFT_eSPI/issues/3081 I'll check later if your workaround works.

Chineko-Koizumi, 2024-01-12:
> seems like my issue #3081 I'll check later if your workaround works.

unfortunately that wasn't my case

Bodmer, 2024-01-16:
readRect creates a bitmap with swapped bytes in the 16-bit colour value. It is intended to be used with pushRect, see here: 

https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L600-L604

closed 2024-01-16 by Bodmer