Bodmer/TFT_eSPI #3117: TTGO T-Display no longer works

MartinNohr, 2024-01-18:
I just built up a new system with arduino IDE. I installed the latest boards and the latest TFT_Espi libraries. I took some code that worked perfectly before and now it does not write to the display correctly. All I get is some random dots. I tried just tft.fillscreen(); and it doesn't work either. I tried the same code with the adafruit libraries and they work correctly. I switched back to the same TFT_Espi version that used to work, and it doesn't work now either. The only difference I am aware is that the board versions changed, it was version 1.? something before and now it is 2.??. Is it possible that something in there broke TFT_Espi? Any help appreciated. I can't go back to the adafruit lib! Thanks.

MartinNohr, 2024-01-19:
I compiled the tft_espi test code and it does write to the display, but it has an odd offset, the text is written offset to the left about 4 characters and the right side has random data.
My other app probably has an interaction with one of the other libraries. I'll try and narrow it down.

dart1280, 2024-01-19:
I have a ttgo T-display device that used to work but when re-flashing the other day it stopped displaying anything. The problem I think was the TFT_eSPI library got updated and the User_Setup_Select.h changed back to setup1 (IL19341) from setup25 (TTGO T display). I also deleted all the C:\Users\XXXXXXXX\AppData\Local\Temp entries before re-compiling and it worked, the display is ok again.

MartinNohr, 2024-01-19:
I'll try that. I did find something else funny, I uncommented this line in the user_setup
#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT
But for some reason the width and height are now set to 320x240, they should 135x240 if I remember correctly.
The compiler also gives an error about redefining those values.

dart1280, 2024-01-19:
I didn't touch anything in User_Setup.h. I edited User_Setup_Select.h to
un-comment
#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32
and TTGO T-Display ST7789V SPI bus TFT

which looks like this -

// Setup for the TTGO T4 ("Bitcoin Tracker") ESP32 board with 2.2" ILI9341
display

// See SetupX_Template.h for all options available

#define ST7789_DRIVER

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required

//#define TFT_MISO -1

#define TFT_MOSI            19
#define TFT_SCLK            18
#define TFT_CS              5
#define TFT_DC              16
#define TFT_RST             23

#define TFT_BL          4  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

//#define SPI_FREQUENCY  27000000
  #define SPI_FREQUENCY  40000000   // Maximum for ILI9341

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed
for the ST7789V

MartinNohr, 2024-01-19:
Sorry,, I meant the user_setup_select.h file. That is the only one I changed. I just uncommented the one TTGO T-display line. It is still setting the wrong display size and I get a redefinition error for the sizes. I'll track it down later today, I have to take kids to school now.

MartinNohr, 2024-01-19:
Found the problem. My bad, I forgot to comment the "user_setup.h" include line. The instructions are clear "only one line should be uncommented". Oops. Working now.

MartinNohr, 2024-01-19:
Working perfectly again. Thanks for the help.

closed 2024-01-19 by MartinNohr