Bodmer/TFT_eSPI #3128: STM32F405/407 DMA is not working (SPI ili9341)

ToRoCu, 2024-01-22:
Hello Bodmer,

I have now tested a few things, without DMA all is working fine but slow.
Tests with the `DMA Example --> boing_ball` were not successful.
The grid is showing but the ball not.
When i commend out the "initDMA" and rename "pushPixelsDMA" to "pushPixels" so it will show a ball.

I think there are more issues in the board package HAL, i have tryed to replace the HAL with LL Driver Registers but that is more difficult than I thought.

Tested with `ili9341` on `SPI 1&2`
Tested board package `2.6.0` & `2.7.1`

Continued to #3072

Bodmer, 2024-01-24:
DMA for the STM32F processors used to work but it appears that a board package update at some point in the past has broken this. Have you tried earlier board packages?

ToRoCu, 2024-01-24:
I have now tested all board packages down to 2.0.0, with the same result that only the grid is displayed in `DMA Example --> boing_ball` but not the ball.

Is it possible that there is another problem in the lib?

zenmetsu, 2024-05-31:
possibly related, but I have a nucleo-144 STM32F767ZI and it also fails to display anything when DMA is enabled.  I can use `#define STM32` just fine and get the optimizations (this doesn't work with my STM32H743ZI2 though).  DMA attempts however result in just a black display.  The board is still responsive, and serial output shows some questionable claims...

`04:34:28.965 -> 1010 fps
04:34:29.059 -> 961 fps
04:34:29.154 -> 1030 fps
04:34:29.247 -> 943 fps
04:34:29.342 -> 1000 fps
`