Bodmer/TFT_eSPI #3152: textWidth sometimes gives wrong width

PE5PVB, 2024-02-02:
Hi,

I use smooth font using command loadFont. I made a ticker where the length of the text may vary. To make an endless ticker I use the command textWidth to get the width of the text to show. But sometimes it just gives wrong values back. This happens when I call it very often. Any ideas?

PE5PVB, 2024-02-02:
This is a sample of my code:

```
void showRadioText() {
  if (radio.rds.hasRT && radio.rds.stationText.length() > 0) {
    if (String(radio.rds.stationText + radio.rds.stationText32).length() != Radiotextlengthold) {
      RadiotextWidth = tft.textWidth(radio.rds.stationText + " " + radio.rds.stationText32) + TickerSpace;
      Radiotextlengthold = String(radio.rds.stationText + radio.rds.stationText32).length();
    }
    if (advancedRDS && radio.rds.stationText.length() < 20) {
      xPos = 0;
      AdvRadiotextSprite.fillSprite(BackgroundColor);
      if (RDSstatus) AdvRadiotextSprite.setTextColor(RDSColor, RDSColorSmooth, false); else AdvRadiotextSprite.setTextColor(RDSDropoutColor, RDSDropoutColorSmooth, false);
      AdvRadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos, 2);
      AdvRadiotextSprite.pushSprite(35, 220);
    } else if (!advancedRDS && radio.rds.stationText.length() < 29) {
      xPos = 0;
      RadiotextSprite.fillSprite(BackgroundColor);
      if (RDSstatus) RadiotextSprite.setTextColor(RDSColor, RDSColorSmooth, false); else RadiotextSprite.setTextColor(RDSDropoutColor, RDSDropoutColorSmooth, false);
      RadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos, 2);
      RadiotextSprite.pushSprite(38, 220);
    } else {
      if (millis() - rtticker >= 15) {
        if (xPos == 0) {
          if (millis() - rttickerhold >= 1000) {
            xPos --;
            rttickerhold = millis();
          }
        } else {
          xPos --;
          rttickerhold = millis();
        }

        if (xPos < -(RadiotextWidth + TickerSpace)) xPos = 0;
        if (advancedRDS) {
          AdvRadiotextSprite.fillSprite(BackgroundColor);
          if (RDSstatus) AdvRadiotextSprite.setTextColor(RDSColor, RDSColorSmooth, false); else AdvRadiotextSprite.setTextColor(RDSDropoutColor, RDSDropoutColorSmooth, false);
          AdvRadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos, 2);
          AdvRadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos + RadiotextWidth + TickerSpace, 2);
          AdvRadiotextSprite.pushSprite(35, 220);
        } else {
          RadiotextSprite.fillSprite(BackgroundColor);
          if (RDSstatus) RadiotextSprite.setTextColor(RDSColor, RDSColorSmooth, false); else RadiotextSprite.setTextColor(RDSDropoutColor, RDSDropoutColorSmooth, false);
          RadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos, 2);
          RadiotextSprite.drawString(radio.rds.stationText + " " + radio.rds.stationText32, xPos + RadiotextWidth + TickerSpace, 2);
          RadiotextSprite.pushSprite(38, 220);
        }
        rtticker = millis();
      }
    }
  }
  if (radio.rds.hasRT) {
    if (!advancedRDS) {
      if (radio.rds.rtAB) {
        tft.fillCircle(314, 223, 2, GreyoutColor);
        tft.fillCircle(314, 234, 2, InsignificantColor);
      } else {
        tft.fillCircle(314, 223, 2, InsignificantColor);
        tft.fillCircle(314, 234, 2, GreyoutColor);
      }
    } else {
      if (radio.rds.rtAB) {
        tft.fillCircle(203, 223, 2, GreyoutColor);
        tft.fillCircle(203, 234, 2, InsignificantColor);
      } else {
        tft.fillCircle(203, 223, 2, InsignificantColor);
        tft.fillCircle(203, 234, 2, GreyoutColor);
      }
    }
  }

  if (wifi) {
    if (RTold != (radio.rds.stationText + " " + radio.rds.stationText32)) {
      Udp.beginPacket(remoteip, 9030);
      Udp.print("from=TEF_tuner " + String(stationlistid, DEC) + ";RT1=");
      char RThex[65];
      radio.rds.stationText.toCharArray(RThex, 65);
      for (int i = 0; i < 64; i++)
      {
        if (RThex[i] < 0x10) Udp.print("0");
        if (RThex[i] == ' ') RThex[i] =  '_';
        Udp.print(String(RThex[i], HEX));
      }
      Udp.endPacket();
    }
  }
  if (RTold != (radio.rds.stationText + " " + radio.rds.stationText32)) xPos = 0;
  RTold = radio.rds.stationText + " " + radio.rds.stationText32;
}
```

Bodmer, 2024-02-05:
If you can provide a simpel and complete sketch that demonstrates the problem then I will investigate. I suspect that there are special circumstances that hve to be invoked.

Bodmer, 2024-03-19:
Unable to reproduce.

PE5PVB, 2024-05-24:
Found the problem. tft.textWidth must be AdvRadiotextSprite.textWidth or RadiotextSprite.textWidth