Bodmer/TFT_eSPI #3154: Misleading directions in user_setup_select.h to use of USER_SETUP_LOADED

WombatHollow, 2024-02-03:
In file user_setup_select.h there is this statement with comments
```
#ifndef USER_SETUP_LOADED //  Lets PlatformIO users define settings in
                          //  platformio.ini, see notes in "Tools" folder.
```
The comments suggest there is an alternate means to define user_setup in the platformio.ini file which would be great as it removes the need to edit this file, and then loose those edits when tft_eSPI gets updates .

But there are no notes in the "Tools" folder nor have I found out how to do this in platformio.ini (Yes I could in Main.ccp add a #include to my specific setup file, and a #define USER_SETUP_LOADED  but I thought there is an issue with libraries seeing #defines made in main.ccp)

Development Configuration is

- platformio running on VSCode and compiling to Arduino framework.  All at latest version for 3 Feb 2024.

Target Hardware 

- TTGO T-Display (not that it is relevant to this problem)

wassfila, 2024-02-04:
I suggested the alternate means (which would be great) here https://github.com/Bodmer/TFT_eSPI/discussions/3161

Bodmer, 2024-02-05:
The info has moved to the docs folder:
https://github.com/Bodmer/TFT_eSPI/tree/master/docs/PlatformIO

I will update the comments at some point.