Bodmer/TFT_eSPI #3163: Esp32 with ILI9488 touch not working

luke2023, 2024-02-05:
I have bought 5 screen, one runs touch correctly (normal z raw is around 90) and the others all have 1890 normal z raw value. and they don't work as well(I guess the program thinks I am pressing them but it is the untouched)

luke2023, 2024-02-05:
[image]

this is my circuit diagram
, and i use touch_calibrate example

luke2023, 2024-02-05:
[image]

This is the one with same circuit and calibration code(may be only this screen is OK?

luke2023, 2024-02-05:
This is the raw and calibration on strange LCD

https://youtu.be/7PVe3VPsW-4?si=7eZyBYNCKUGGZN29[](https://youtu.be/7PVe3VPsW-4?si=7eZyBYNCKUGGZN29)

Bodmer, 2024-02-06:
Try using the optional touch threshold parameter, by default it is 600, see here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.h#L11-L18

Maybe a threshold of 2000 would work.

luke2023, 2024-02-06:
@Bodmer thank you for replying,so I should simply modify the 600 to 2000?

Bodmer, 2024-02-08:
Just add the extra parameter to get touch to define the threshold, e.g.:
tft.getTouch(&x, &y, 2000);

luke2023, 2024-02-09:
@Bodmer thabk you for the suggestion

closed 2024-03-05 by Bodmer