Bodmer/TFT_eSPI #3169: ESP32's touch not working for ILI9486

hemangjoshi37a, 2024-02-08:
getting this error : 
```
In file included from /home/hemang/.var/app/cc.arduino.IDE2/cache/.arduinoIDE-unsaved202419-70-f5p4k3.xj8hg/Keypad_480x320/Keypad_480x320.ino:3:
/home/hemang/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:966:8: error: #error >>>>------>> Touch functions not supported in 8/16 bit parallel mode or with RP2040 PIO.
       #error >>>>------>> Touch functions not supported in 8/16 bit parallel mode or with RP2040 PIO.
        ^~~~~
exit status 1

Compilation error: exit status 1
```

my code : 
```
#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

#define TFT_GREY 0x5AEB // New colour

#define TFT_DC   18  // Data Command control pin - must use a pin in the range 0-31  (LCD_RS on my display)
#define TFT_CS   33  // Chip select control pin
#define TFT_RST  32  // Reset pin
#define TFT_RD    2
#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   21
#define TFT_D5   5
#define TFT_D6   27
#define TFT_D7   14
#define TOUCH_CS   19

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

void setup() {
  tft.init();
  tft.setRotation(1);
}

void loop() {
  // Fill screen with random colour so we can see the effect of printing with and without 
  // a background colour defined
  tft.fillScreen(random(0xFFFF));
  
  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
  tft.setCursor(0, 0, 2);
  // Set the font colour to be white with a black background, set text size multiplier to 1
  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
  // We can now plot text on screen using the "print" class
  tft.println("Hello World!");
  
  // Set the font colour to be yellow with no background, set to font 7
  tft.setTextColor(TFT_YELLOW); tft.setTextFont(7);
  tft.println(1234.56);
  
[... 353 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```

closed 2024-02-12 by Bodmer

hemangjoshi37a, 2024-02-12:
@Bodmer can you please suggest any touch library for example that can work with this?

Bodmer, 2024-02-14:
The library selection depends on the touch controller fitted to your display and the interface. There are quite a few variants. A common controller is the XPT2046 via SPI. In that case this library should work:
https://github.com/achillhasler/TFT_eTouch