Bodmer/TFT_eSPI #3171: Color brightness/displaying off with ILI9486 when fading colors

Linzm99, 2024-02-10:
Hi,

I have the following setup:
- ILI9486 Driver (3,5'' TFT 480x320)
- ESP32C3
- Arduino IDE 2
- latest TFT_eSPI
- 4-Wire SPI 

I noticed something strange when implementing a color fader for fading out stuff. For example:

Draw any bitmap with just the color blue (0b11111 or 31) then start decreasing one by one and redraw every time. The effect that I'm seeing is that it doesnt get just darker and darker but there is another local brightness maxima at around 20. So counting down it gets darker until around 25 then again brighter down to 20, again brighter to around 12 and then finally completely dark at 0.

This applies to every color channel and combined colors as well.

Simplest approach for achieving this is when inputing a counter from 0 to 31 into the tft.fillScreen function like:
tft.fillScreen((uint16_t)counter);

Dont know if this is a conversion problem somewhere or a misconfiguration by me.

Thx in advance.

Bodmer, 2024-02-12:
It sounds like the controller is not an ILI9486 so the gamma settings for the display are not correct.

The alpahBlend test should show if there is a problem.:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Generic/alphaBlend_Test

Linzm99, 2024-02-13:
You are right, on further googleing I found out that it was indeed a ILI9488.

Never trust your dodgy tft screen dealer

closed 2024-02-14 by Bodmer