Bodmer/TFT_eSPI #3172: Esp32 S3 devkit c-1 n16r8, waveshare 1.69inch lcd module (240x280) not working with multiple configurations in user setup

varunpuppala, 2024-02-12:
I am trying to use this library with Waveshare 1.69 inch lcd module (240x280).https://www.waveshare.com/1.69inch-lcd-module.htm
Unlike the previous display modules i have used, this has different pins as below image.
[image]

Can someone pls help me regarding this. I had no luck.
I'm using latest version of the library and testing with Colour_Test code.

The below is my User_Setup.h file, i have tried different combinations but could not get it to work. 

`//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
//#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Only define one driver, the other ones must be commented out
// #define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
// #define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1351_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER
//#define GC9A01_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

 #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For M5Stack ESP32 module with integrated ILI9341 display ONLY, remove // in line below

// #define M5STACK

[... 125 lines cut ...]
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS
`

varunpuppala, 2024-02-14:
Hi @Bodmer , i have identified another issue. I have defined different pins in setup file as below.
// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

----------------------------------------------------------------------------------------------------------
// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

// #define TFT_MISO 19
#define TFT_MOSI 4
#define TFT_SCLK 5
#define TFT_CS   1  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   15  // Reset pin (could connect to RST pin)
// #define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
----------------------------------------------------------------------------------------------------------
but when i upload Read_User_Setup.ino and open serial monitor, i see it as below
----------------------------------------------------------------------------------------------------------
[code]
TFT_eSPI ver = 2.5.34
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 280

R0 y offset = 20
MOSI    = GPIO 4
SCK     = GPIO 5
TFT_DC   = GPIO 5

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]
----------------------------------------------------------------------------------------------------------

Bodmer, 2024-02-15:
It looks like you have more than 1 setup file being called up and that is also defining pins. Check you User_Setup_Select.h file and make sure only one setup file is being loaded.

Alama32, 2026-08-20:
Hello everyone,
I'm trying to connect a 1.28" GC9A01 TFT screen to an ESP32-S3 (N16R8, 44 pins). I've been stuck on this for 3 days and can't get it to work.
The backlight (BL) is working fine since it's hardwired to 3.3V on the module, and some lines briefly appear on the screen, but nothing else.
Here is my current wiring:
VCC>3V3
GND>GND
RST>GPIO11
DC >GPIO9
CS >GPIO10
SCL>GPIO15
SDA>GPIO16
BL >3V3
*************************************************
the actual code is:

```
#include <Arduino.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_GC9A01A.h>

// Câblage réel du GC9A01
#define TFT_SCK   15
#define TFT_MOSI  16
#define TFT_DC     9
#define TFT_CS    10
#define TFT_RST   11

// Deuxième CS du dashboard, maintenu désactivé
#define TFT_CS2   12

SPIClass displaySPI(FSPI);

Adafruit_GC9A01A tft(
    &displaySPI,
    TFT_DC,
    TFT_CS,
    TFT_RST
);

void setup()
{
    Serial.begin(115200);
    delay(1000);

    Serial.println("Demarrage du test GC9A01");

    // Désactive le deuxième écran
    pinMode(TFT_CS2, OUTPUT);
    digitalWrite(TFT_CS2, HIGH);

    // SCK, MISO inutilisé, MOSI, CS
    displaySPI.begin(TFT_SCK, -1, TFT_MOSI, TFT_CS);

    // Test prudent à 10 MHz
    tft.begin(10000000);
    tft.setRotation(0);

    Serial.println("Initialisation terminee");
}

void loop()
{
    Serial.println("Rouge");
    tft.fillScreen(GC9A01A_RED);
    delay(2000);

    Serial.println("Vert");
    tft.fillScreen(GC9A01A_GREEN);
    delay(2000);

    Serial.println("Bleu");
    tft.fillScreen(GC9A01A_BLUE);
    delay(2000);

    Serial.println("Blanc");
    tft.fillScreen(GC9A01A_WHITE);
    delay(2000);

    Serial.println("Noir");
    tft.fillScreen(GC9A01A_BLACK);
    delay(2000);
}
```