Bodmer/TFT_eSPI #3185: What is the image size limit for the Flash_PNG example?

rtek1000, 2024-02-21:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
- The ESP32 is rebooting with an image of just 347x45 pixels.
- Small images load normally.
- Code ref.: https://github.com/Bodmer/TFT_eSPI/tree/master/examples/PNG%20Images/Flash_PNG
- Usage ref.: https://www.youtube.com/watch?v=ZjPZcyIQFhw

2. IDE (e.g. Arduino or PlatformIO)
- Arduino IDE 1.8.19 (Version 2 of the IDE does not compile)

3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
- Downloaded from GITHUB on Feb 24, 2024.

4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu

5. Procesor, e.g RP2040, ESP32 S3 etc
- ESP32 DEV KIT V1

6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
- SSD1963 (800x480, 7 inch)

7. Interface type (SPI or parallel)
- Parallel

Plus further information as appropriate to the problem:
1. TFT to processor connections used
- LED_A: GPIO23 (PWM)

2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

DO NOT paste code directly into the issue. To correctly format code put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies. [See here:](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

Example output:

```
16:44:11.148 -> Initialisation done.
16:44:11.182 -> Successfully opened png file
16:44:11.215 -> image specs: (347 x 45), 8 bpp, pixel type: 6
16:44:12.077 -> *Fd⸮⸮⸮⸮⸮O⸮⸮
16:44:12.243 -> 
16:44:12.243 ->  Using the PNGdec library
16:44:12.674 -> 
16:44:12.674 -> Initialisation done.
16:44:12.707 -> Successfully opened png file
16:44:12.707 -> image specs: (347 x 45), 8 bpp, pixel type: 6
16:44:13.569 -> 	1⸮U⸮⸮⸮b⸮q⸮ac3 ⸮
16:44:13.735 -> 
16:44:13.735 ->  Using the PNGdec library
16:44:14.167 -> 
16:44:14.167 -> Initialisation done.
16:44:14.200 -> Successfully opened png file
16:44:14.233 -> image specs: (347 x 45), 8 bpp, pixel type: 6
16:44:15.062 -> ⸮⸮
16:44:15.095 -> / b[)⸮]⸮⸮
```

Bodmer, 2024-02-22:
There is no limit as far as I am aware.

It there is a watchdog timeout then look for places in your code that takes a long time and insert a yield(). 

If that is not the fix then post a complete example sketch that demonstrates the problem.

rtek1000, 2024-02-22:
> There is no limit as far as I am aware.
> 
> It there is a watchdog timeout then look for places in your code that takes a long time and insert a yield().
> 
> If that is not the fix then post a complete example sketch that demonstrates the problem.

I did some more tests and perhaps the problem is when converting the PNG image to the *.h binary file, an image with the same dimensions (240x320) is generating a smaller file (almost half of the panda file).

https://notisrac.github.io/FileToCArray/
- Treat as binary [x]
- static [x]
- const [x]
- unsigned [x]
- Data type: char
- PROGMEM [x]

panda.h:
> // array size is 182750
> static const unsigned char panda[] PROGMEM  = {

fish_240x320.h:
// array size is 90270
static const unsigned char fish_240x320[] PROGMEM  = {

Can you provide the original image of the Panda file to compare, please?

[Fish.zip](https://github.com/Bodmer/TFT_eSPI/files/14380389/Fish.zip)

rtek1000, 2024-02-22:
> There is no limit as far as I am aware.
> 
> It there is a watchdog timeout then look for places in your code that takes a long time and insert a yield().
> 
> If that is not the fix then post a complete example sketch that demonstrates the problem.

- The code is the same as the example

rtek1000, 2024-02-25:
I did more testing and found a width limit of exactly 640 pixels. For example, a 640x480 image is displayed, but a 641x480 image is not displayed.

But a 480x800 image is displayed.

Regarding the error with smaller images, it is due to the MAX_IMAGE_WIDTH parameter, which I was only able to see now.

`#define MAX_IMAGE_WIDTH 900 // Adjust for your images`

rtek1000, 2024-02-25:
There is a limitation in the PNG decompression library (PNGdec lib). But if the value of MAX_IMAGE_WIDTH (TFT_eSPI lib) is smaller than the image size, the CPU reset continues to occur.

```C++
/* Defines and variables */
#define PNG_FILE_BUF_SIZE 2048
// Number of bytes to reserve for current and previous lines
// Defaults to 640 32-bit pixels max width
#define PNG_MAX_BUFFERED_PIXELS (640*4 + 1)
```

https://github.com/bitbank2/PNGdec/blob/master/src/PNGdec.h

https://github.com/bitbank2/PNGdec/issues/23#issuecomment-1962858503

rtek1000, 2024-02-25:
There are at least 2 limitations, for example, if it is 800x480:
- So to be able to operate with images larger than 240 pixels wide, you need to be careful with the **MAX_IMAGE_WIDTH** parameter (**_which causes the CPU to reset_**).

- And you need to change the value of the **PNG_MAX_BUFFERED_PIXELS** parameter in the PNGdec library, to be able to operate with images larger than 640 pixels wide. If the image is larger than PNG_MAX_BUFFERED_PIXELS, the png.openFLASH() function does not return PNG_SUCCESS.

/libraries/PNGdec/src/PNGdec.h
```C++
/* Defines and variables */
#define PNG_FILE_BUF_SIZE 2048
// Number of bytes to reserve for current and previous lines
// Defaults to 640 32-bit pixels max width
#define PNG_MAX_BUFFERED_PIXELS (800*4 + 1) // <-------------------------------- need change from 640 to 800
```

Arduino IDE sketch:
```C++
#include <PNGdec.h>
#include "fish.h" // Image is stored here in an 8-bit array

PNG png; // PNG decoder instance

#define MAX_IMAGE_WIDTH 240 // Adjust for your images  // <-------------------------------- need change from 320 to 800
```

closed 2024-02-25 by rtek1000

rtek1000, 2024-02-25:
> There are at least 2 limitations, for example, if it is 800x480:
>
>     * So to be able to operate with images larger than 240 pixels wide, you need to be careful with the **MAX_IMAGE_WIDTH** [...]

From https://github.com/bitbank2/PNGdec/issues/23#issuecomment-1963017246 :

> the max buffered pixels needs to be large enough to hold 2 lines (current and previous). If you're decoding 32-bit pixels, then that's 800 x 2 x 4

> MAX_BUFFERED_PIXELS is for the pixels, FILE_BUF is for uncompressed data. A small file buffer size will slow it down due to more calls to read more data.

rtek1000, 2024-02-25:
> Use my image_to_c tool to convert files for Arduino. No options needed, the data comes out the way it should.

https://github.com/bitbank2/image_to_c