Bodmer/TFT_eSPI #3202: Problem related to pushToSprite when deleting the sprite edges.

goqeq, 2024-03-02:
Hello everyone, 

I recently wanted to draw a rectangle in a sprite named "battery" and put it in the background sprite using the pushToSprite function. However, when I use the "transparent" argument with TFT_BLACK, the rectangle appears in the correct color and yes, the background is properly removed, but a bug occurs on the rectangle where it seems like black pixel lines are filling the rectangle. I have tried different approaches before asking this question and the only way for it to work is to not use this argument. Which is really unfortunate... I have tried other things before asking this question and the only solution I found is to not include this line "battery.setColorDepth(8);". So, I would like to know if this is the only possible solution and if I have written my test code incorrectly? I am new to this library, I apologize if I asked a silly question. 

I am working on an esp32 wroom 32 with a 240x240 screen with GC9A01.

Merci d'avance !

[image]

goqeq, 2024-03-03:
Finally I found using TFT_TRANSPARENT or at least a different color from the background sprite. I didn't really understand I couldn't use TFT_BLACK but it's not very important, as long as it works.

Bodmer, 2024-03-04:
Thanks for reporting this, it has been fixed in the Github master.

commit 9ee4bf68 (2024-03-04): Fix #3202
closed 2024-03-04 by Bodmer