Bodmer/TFT_eSPI #3247: ESP32+ ST7796 320x480 display white screen

Apprisco, 2024-03-22:
Hi all, I'm using an esp32s2 Saola DevKit R, and it seems to be working properly.
I have connected a 4 inch 320x480 SPI ST7796 320x480 to it, and configured the user_setup and wrote it properly.

#define ST7796_DRIVER

#define TFT_MISO    10
#define TFT_MOSI    7
#define TFT_SCLK    9
#define TFT_CS      12      // Chip select control pin
#define TFT_DC      11      // Data Command control pin
#define TFT_RST     13
#define TFT_BL      8       // TFT backlight  pin
#define TFT_BACKLIGHT_ON HIGH

#define SPI_FREQUENCY  20000000
#define SPI_READ_FREQUENCY  20000000

user setup above

I have confirmed that changing the backlight on high/low does affect the screen brightness, which is perfect. However, I cannot get ANY image to properly display on this display. Do you have any ideas?

FengKuangDeJieKeSi, 2024-03-24:
Mine is the same, I use 9431 driver, modify the register, can display, but the TFT function sometimes stuck

FengKuangDeJieKeSi, 2024-03-24:
The 7796 register address in the TFT library is similar to that provided by the manufacturer, but the register modified into that provided by the manufacturer cannot be displayed normally

//The following is the pin definition
//#define TFT_MISO 19
#define TFT_MOSI       37   //23
#define TFT_SCLK       36   //18
#define TFT_CS         38   //14  // Chip select control pin
#define TFT_DC         35   //27  // Data Command control pin
#define TFT_RST        47   //33  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL         48   //32  // LED back-li

OlliHenri, 2025-02-03:
Hello,
I use this display: http://www.lcdwiki.com/4.0inch_Capacitive_SPI_Module_ST7796#Product_Picture 
MSP4031. When I use tft_eSPI I get only the white screen but the program is working (visible on serial output) and does not stop.
When I use the simple_test.ino that is provided at LCD wiki the display is working normal.
When I compare the init files I realise they are quite different.
I tried both SPI bus HSPI and VSPI. I use VScode.
Here is what I get from TFT_eSPI as output, but white screen.
[code]
TFT_eSPI ver = 2.5.43
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7796
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
Benchmark                Time (microseconds)
HaD pushColor            73301
Screen fill              31803
Text                     12158
Pixels                   366309
Lines                    395305
Horiz/Vert Lines         15319
Rectangles (outline)     8338
Rectangles (filled)      388002
Circles (filled)         59799
Circles (outline)        39547
Triangles (outline)      21913
Triangles (filled)       136613
Rounded rects (outline)  19783
Rounded rects (filled)   392105
Done!

Is this a driver issue?

kind regards Oliver

barman75, 2025-06-28:
I had the same problem with this 3.5 inch 9-pin ST7796 display, the examples from the library don't work, and only a white screen is displayed. And I was about to throw this bastard in the trash. But at the last moment I found a mention from some user [Mark Ahn from Korea](https://blog.naver.com/ysahn2k/223679581814), and it turned out that with a simple code the library itself [still works](https://youtu.be/isvf_AuPh1o). I have checked myself and can confirm, that library somehow works with this 9pin display, but examples from this library are not. 
So I want to find out why is it happening? Could it be an initialization problem or something? My sketch uses  `tft.init();` instead of  `tft.begin();` and has `#include <SPI.h>` string, but just changing it on example files has no success to me. I even try to use a simplest example `Gradient_Fill.ino` from the library, and it does not worked, but the code from it somehow works fine in my sketch, look at `testGradientFill`. What is wrong? (I have added `User_Setup.h` file in my sketch folder as well, but I guess it is not the case).

[image]

My User_Setup:
```
#define USER_SETUP_INFO "User_Setup"
#define ST7796_DRIVER
#define TFT_WIDTH  480 // ST7796 320 x 480
#define TFT_HEIGHT 320 // ST7796 320 x 480
#define TFT_MOSI 23 // MOSI = SDA on ST7796 = GPIO23
#define TFT_SCLK 18  // SPI Clock = SCL on ST7796 = GPIO18
#define TFT_CS    5  // Chip select = CS on ST7796 = GPIO5
#define TFT_DC    2  // Data Command control = DC on ST7796 = GPIO2
#define TFT_RST   4  // Reset pin = RST pin on ST7796 = GPIO4. Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-. 
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT
#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
```

This sketch works fine to me

```
#include <TFT_eSPI.h>
#include <SPI.h>
#include "Free_Fonts.h"

TFT_eSPI tft = TFT_eSPI();
unsigned long drawTime = 0;
TFT_eSprite sprite = TFT_eSprite(&tft);

void setup() {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  delay(500);
}

void loop() {
  testGradientFill();
  fontstest();
  testFill();
  testLines();
  testRectangles();
  testCircles();
  testTriangles();
  testArcs();
  testText();
  testPixels();
  testFPS();
  testAnimatedGraph();
  testSpriteDemo();
  testProgressBar();
  testColorPalette();
  testHistogram();
  testOscilloscope();
  testAnalogClock();
  //testScrollingText(); // -- works bad
  //testRainEffect();
  testAntiAliasingArc();
[... 358 lines cut ...]
  uint8_t thickness    = random(1, radius / 4);     // Thickness
  uint8_t inner_radius = radius - thickness;        // Calculate inner radius (can be 0 for circle segment)

  // 0 degrees is at 6 o'clock position
  // Arcs are drawn clockwise from start_angle to end_angle
  uint16_t start_angle = random(361); // Start angle must be in range 0 to 360
  uint16_t end_angle   = random(361); // End angle must be in range 0 to 360
  bool arc_end = random(2);           // true = round ends, false = square ends (arc_end parameter can be omitted, ends will then be square)
  tft.drawSmoothArc(x, y, radius, inner_radius, start_angle, end_angle, fg_color, bg_color, arc_end);
  count++;
  if (count < 30) delay(500); // After 15s draw as fast as possible!
  delay(1000);
}

void testGradientFill() {
  tft.fillRectHGradient(0, 0, 160, 50, TFT_MAGENTA, TFT_BLUE);
  tft.setCursor(10,10);
  tft.print("Horizontal gradient");
  tft.fillRectVGradient(0, 60, 160, 50, TFT_ORANGE, TFT_RED);
  tft.setCursor(10,70);
  tft.print("Vertical gradient");
  delay(2000);
}

```