Bodmer/TFT_eSPI #3253: ILI9488 working, ILI9341 NOT

bbuster2, 2024-03-25:
I need 2 different screen for 2 devices. Both bought here:
https://nl.aliexpress.com/item/32837085500.html?spm=a2g0o.order_list.order_list_main.5.17b579d2X3nRKb&gatewayAdapt=glo2nld
I have the 2.4" ILI9341 and the 3.5" ILI9488

De documentation shows both screens use exactly the same wiring but for some reason, the ILI9341 stays white and the ILI9488 works perfect.

I changed the driver in code corresponding the used display but the ILI9341 stays white.

The only difference i made is the speeds.

ILI9488:
 SPI_FREQUENCY=27000000
 SPI_READ_FREQUENCY=10000000

ILI9431:
 SPI_FREQUENCY=40000000
 SPI_READ_FREQUENCY=20000000

What can be the problem?
I have tested 3 different ILI9431 displays and all 3 have the same issue

vikraj-ai, 2024-03-29:
Hi bbuster 2, 
 May you provide me with you wiring diagrams and user setup files you have used for your ILI9488

bbuster2, 2024-03-29:
I got it working. The documentation switched 2 pins (SDA and SDO). After using the correct pins the display started to work

vikraj-ai, 2024-04-01:
Thank you for efforts my display is now working but I don't know how to use the touch functionality of my touch screen can you give wiring diagrams and user setup file for that it would be a great help

fungreenfox, 2024-07-15:
Same here. Having both displays, and the 9488 works ok.
The ili9341 gos white background with black text on with tft.setTextColor(WHITE, BLACK);

define LTBLUE 0xB6DF gives a pastel yellow color. 
The black color lies around a pixcolor 254.
The driver choise is #include <User_Setups/Setup42_ILI9341_ESP32.h>           // Setup file for ESP32 and SPI ILI9341 240x320

Do you have a clue why this happens after a library upgrade for TFT_Espi?
How to fix this?