Bodmer/TFT_eSPI #3254: ESP32S3 DevKit C-1 Invalid pin selected error

ctaskin, 2024-03-25:
I am flawlessly using your library with ESP32 in PlatformIO but when I try to upgrade to ESP32S3 DevKit C-1 board, I am getting Invalid pin selected error. It tried several pin configurations but could not manage it to work. I am sending some of the configs I tried. Can you guide me to selve the problem.

Note: enabling "-DUSE_HSPI_PORT=1" gives "HSPI Does not have default pins on ESP32S3!" error.
Thanks

**Board:** 
[image]

**CONFIG 1:**
[env:esp32-s3-devkitc-1-n16r8v]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
#board_build.partitions = huge_app.csv
build_unflags = -DARDUINO_USB_CDC_ON_BOOT=1
build_flags = 
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DTFT_SDA_READ=1
	-DTFT_RGB_ORDER=0
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=240
	-DTFT_MOSI=11
	-DTFT_MISO=13
	-DTFT_SCLK=12
	-DTFT_CS=10
	-DTFT_DC=7
	-DTFT_RST=6
	-DTFT_BL=3
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=27000000
	#-DUSE_HSPI_PORT=1
	-DARDUINO_USB_CDC_ON_BOOT=0
	-DARDUINO_USB_MODE=1
lib_deps = 
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	bblanchon/ArduinoJson@^7.0.3
	knolleary/PubSubClient@^2.8
	h2zero/NimBLE-Arduino@^1.4.1
	bodmer/TFT_eSPI@^2.5.43
	bodmer/TFT_eWidget@^0.0.6
	lvgl/lvgl@8.3.6

**ERROR:** 
[   309][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (26) gpio: gpio_set_level(227): GPIO output gpio_num error
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403793e7
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

**CONFIG 2:**
[env:esp32-s3-devkitc-1-n16r8v]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
#board_build.partitions = huge_app.csv
build_unflags = -DARDUINO_USB_CDC_ON_BOOT=1
build_flags = 
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DTFT_SDA_READ=1
	-DTFT_RGB_ORDER=0
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=240
	-DTFT_MOSI=41
	-DTFT_MISO=-1
	-DTFT_SCLK=42
	-DTFT_CS=38
	-DTFT_DC=39
	-DTFT_RST=40
	-DTFT_BL=3
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=27000000
	#-DUSE_HSPI_PORT=1
	-DARDUINO_USB_CDC_ON_BOOT=0
	-DARDUINO_USB_MODE=1
lib_deps = 
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	bblanchon/ArduinoJson@^7.0.3
	knolleary/PubSubClient@^2.8
	h2zero/NimBLE-Arduino@^1.4.1
	bodmer/TFT_eSPI@^2.5.43
	bodmer/TFT_eWidget@^0.0.6
	lvgl/lvgl@8.3.6

**ERROR 2:**
[   309][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (26) gpio: gpio_set_level(227): GPIO output gpio_num error
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403793e7
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

**CONFIG 3:**
[env:esp32-s3-devkitc-1-n16r8v]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
#board_build.partitions = huge_app.csv
build_unflags = -DARDUINO_USB_CDC_ON_BOOT=1
build_flags = 
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DTFT_SDA_READ=1
	-DTFT_RGB_ORDER=0
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=240
	-DTFT_MOSI=35
	-DTFT_MISO=37
	-DTFT_SCLK=36
	-DTFT_CS=34
	-DTFT_DC=7
	-DTFT_RST=6
	-DTFT_BL=3
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=27000000
	#-DUSE_HSPI_PORT=1
	-DARDUINO_USB_CDC_ON_BOOT=0
	-DARDUINO_USB_MODE=1
lib_deps = 
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	bblanchon/ArduinoJson@^7.0.3
	knolleary/PubSubClient@^2.8
	h2zero/NimBLE-Arduino@^1.4.1
	bodmer/TFT_eSPI@^2.5.43
	bodmer/TFT_eWidget@^0.0.6
	lvgl/lvgl@8.3.6

**ERROR 2:**
[   309][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (26) gpio: gpio_set_level(227): GPIO output gpio_num error
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403793e7
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0

ctaskin, 2024-04-16:
Any suggestions?

freiregtr, 2024-04-26:
I have the same configuration, but with an ST7796s driver, however, I can configure the screen but not the touch functions. Do you want me to share what I have configured? since I was able to display the information on the screen

ctaskin, 2024-04-29:
Hi @freiregtr thank you, that would be great, I do not need the touch either.

freiregtr, 2024-04-29:
ok @ctaskin I share with you the schematic of the ESP32-S3-DevKitC-1 and the screen with the ST7796s controller, here I mention the pins that, as I said before, I cannot connect, but since this is not your case, you can ignore them.

`ST7796S LCD Display    ESP32-S3 DevKitC-1
+------------------+   +------------------+
|                  |   |                  |
|       VCC -------|---| 3V3              |
|       GND -------|---| GND              |
|       CS  -------|---| GPIO15           |
|       RESET -----|---| GPIO4            |
|       DC/RS -----|---| GPIO2            |
|       MOSI ------|---| GPIO7            |
|       SCK -------|---| GPIO6            |
|       LED -------|---| 3V3 (directo)    |
|                  |   |                  |
|    Touch Screen  |   |   ESP32-S3 Pins  |
|       T_CS ------|---| GPIO10           |
|       T_IRQ -----|---| GPIO3 (optional) |
|       T_DIN -----|---| GPIO8            |
|       T_CLK -----|---| GPIO5            |
|       T_DO ------|---| GPIO9            |
|                  |   |                  |
+------------------+   +------------------+`

If you are working on platformio you can configure it using flags, I will share my platformio.ini configuration with you

`build_flags = 
    -DBOARD_HAS_PSRAM 
    -mfix-esp32-psram-cache-issue
    -D USER_SETUP_LOADED
    -D ST7796_DRIVER
    -D TFT_MISO=-1
    -D TFT_MOSI=7
    -D TFT_SCLK=6
    -D TFT_CS=15
    -D TFT_DC=2
    -D TFT_RST=4
    -D TOUCH_CS=10  ; Asegúrate de agregar esta línea
    -D LOAD_GLCD=1
    -D LOAD_FONT2
    -D LOAD_FONT4
    -D LOAD_FONT6
    -D LOAD_FONT7
    -D LOAD_FONT8
    -D LOAD_GFXFF
    -D SMOOTH_FONT
    -D SPI_FREQUENCY=27000000
`

If you are working from arduino ide you only have to change the parameters in user_setup.h and replace the pins with the ones I mention here

`// Define los pines SPI para el ESP32-S3
#define TFT_MISO -1       // Si no usas el pin MISO, déjalo como -1
#define TFT_MOSI 7        // MOSI
#define TFT_SCLK 6        // SCK (Reloj)

// Define los pines para el chip select, DC y reset del TFT
#define TFT_CS   15       // Chip select
#define TFT_DC   2        // Data Command
#define TFT_RST  4        // Reset

// Define el pin para el chip select del touch
#define TOUCH_CS 10       // Chip select del touch
`

anything, if you need help you can ask, happy to help

ctaskin, 2024-04-30:
Finally solved the problem, it was not related the config or pin layout, some how I left out the code below for TFT from a sample project 🤦‍♂️, which works fine with ESP32 but causing problem with S3. Commenting out it fixed the issue. 
@freiregtr Thank you.

`  // background light pin
   pinMode(27, OUTPUT);
   digitalWrite(27, HIGH);`

closed 2024-04-30 by ctaskin