Bodmer/TFT_eSPI #3255: Support for ESP32C6?

HonestQiao, 2024-03-26:
datasheet of ESP32C6:
https://www.espressif.com.cn/sites/default/files/documentation/esp32-c6_datasheet_en.pdf

Love4yzp, 2024-06-18:
Have you faced any problem?

OscarTofa, 2024-06-18:
Hello

I am trying to compile for c6 but I get the several errors. 

"
oc/esp32c6/include/soc/soc.h:112:54: error: invalid operands of types 'void' and 'long unsigned int' to binary 'operator|'
  112 |             WRITE_PERI_REG((reg), (READ_PERI_REG(reg)|(mask)));                                                        \

esp32c6/include/soc/soc.h:112:54: error: invalid operands of types 'void' and 'long unsigned int' to binary 'operator|'
  112 |             WRITE_PERI_REG((reg), (READ_PERI_REG(reg)|(mask)));                                                        \

TFT_eSPI_ESP32.h:75:22: error: 'VSPI' was not declared in this scope; did you mean 'SPI'?
   75 |     #define SPI_PORT VSPI
TFT_eSPI_ESP32.c:409:48: error: invalid operands of types 'void' and 'long unsigned int' to binary 'operator&'
  409 |     while (READ_PERI_REG(SPI_CMD_REG(SPI_PORT))&SPI_USR);

TFT_eSPI_ESP32.h:189:50: error: no match for 'operator=' (operand types are 'volatile gpio_out_w1tc_reg_t' and 'int')
  189 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
  
TFT_eSPI_ESP32.h:235:50: error: no match for 'operator=' (operand types are 'volatile gpio_out_w1ts_reg_t' and 'int')
  235 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
  
If I compile for c2 or s3 I have no error

OscarTofa, 2024-06-25:
Has anyone else experienced the same problem?

limengdu, 2024-06-25:
I'm having the same problem, it looks like the TFT library is no longer supported since esp-arduino updated to 3.0.x.

Cincinnatu, 2024-07-08:
ESP32C6 is not supported. Can someone submit a PR (pull request) for it

Cincinnatu, 2024-07-09:
I finished the PR about ESP32C6 and TFT_eSPI :https://github.com/Bodmer/TFT_eSPI/pull/3399

yuangezhizao, 2024-08-06:
> Has anyone else experienced the same problem?

Same issue, waiting for the [Cincinnatu](https://github.com/Cincinnatu)'s PR https://github.com/Bodmer/TFT_eSPI/pull/3399.
``` C
/Users/yuangezhizao/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:190:50: error: no match for 'operator=' (operand types are 'volatile gpio_out_w1ts_reg_t' and 'int')
  190 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                                                  ^
/Users/yuangezhizao/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:3779:5: note: in expansion of macro 'DC_D'
 3779 |     DC_D; tft_Write_16N(color);
      |     ^~~~
/Users/yuangezhizao/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/include/soc/esp32c6/include/soc/gpio_struct.h:51:3: note: candidate: 'constexpr gpio_out_w1ts_reg_t& gpio_out_w1ts_reg_t::operator=(gpio_out_w1ts_reg_t&&)'
   51 | } gpio_out_w1ts_reg_t;
      |   ^~~~~~~~~~~~~~~~~~~
/Users/yuangezhizao/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/include/soc/esp32c6/include/soc/gpio_struct.h:51:3: note:   no known conversion for argument 1 from 'int' to 'gpio_out_w1ts_reg_t&&'
/Users/yuangezhizao/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/include/soc/esp32c6/include/soc/gpio_struct.h:51:3: note: candidate: 'constexpr gpio_out_w1ts_reg_t& gpio_out_w1ts_reg_t::operator=(const gpio_out_w1ts_reg_t&)'
/Users/yuangezhizao/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-bd2b9390ef/esp32c6/include/soc/esp32c6/include/soc/gpio_struct.h:51:3: note:   no known conversion for argument 1 from 'int' to 'const gpio_out_w1ts_reg_t&'
```

pdppdp, 2024-12-18:
Will these fixes address the similar issue I am having with ESP32-H2's?

mattfox27, 2025-01-07:
same error

ShocKwav3, 2025-01-23:
I have been having a lot of compile errors as well. Came across this issue and found [Cincinnatu](https://github.com/Cincinnatu)'s PR https://github.com/Bodmer/TFT_eSPI/pull/3399.

Tried his branch and worked like a charm!

I don't prefer it but I suppose for now I'll need to keep using his branch locally without having it in package manager.

Hoping the PR will be merged soon!