Bodmer/TFT_eSPI #3260: Hi Bodmer

vikraj-ai, 2024-03-29:
I need your help to connect my ILI9488 TFT SPI  3.5  480*320 resolution display with ESP32 . I need wiring diagram and user setup file for my project. Also I am not confirm
about my display driver chup name at first the vendor told me it ILI9486 then he confirmed from his vendor and told it is ILI9488 . I don't know what to do. If you need any information please tell me this project is very important for me please help

Your small effort will be a great help for me 

Thank you,

vikraj-ai, 2024-03-29:
If you need any pictures then tell me

AgainPsychoX, 2024-03-29:
Is `User_Setups/Setup21_ILI9488.h` not good for you?

AgainPsychoX, 2024-03-29:
For ESP32 you might want use HSPI pins:
```cpp
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
```
or VSPI pins, usually:
```cpp
#define TFT_MISO 19 // (leave TFT SDO disconnected if other SPI devices share MISO)
#define TFT_MOSI 23
#define TFT_SCLK 18
```

Other pins can use whatever GPIO you want I guess:
```cpp
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
```

vikraj-ai, 2024-04-01:
No it worked finally but I don't know how to use the touch function of the screen can you guide me to do that it would be a great help for me

AgainPsychoX, 2024-04-02:
The library supports XPT2046 touch chipset. You didn't provide picture or link to datasheet of your display board so it's hard to say how to wire it up exactly, but the outline is: 

Set one of pins as chip select in the user setup:
```cpp
#define TOUCH_CS 8
```
and share the SPI bus (MISO/MOSI/CLK pins) with the touch controller.