Bodmer/TFT_eSPI #3261: Display can show glitches/artefacts if touch used called immediately after

AgainPsychoX, 2024-03-29:
### Problem

Display can show glitches/artefacts if touch used called immediately after. I noticed it the most while drawing strings with font 2. 

### My setup

#### Hardware 

Microcontroller: [Raspberry Pi Pico](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html)
Display: [4'' TFT SPI ST7796 480x320](http://www.lcdwiki.com/4.0inch_SPI_Module_ST7796)
Display driver: [ST7796(S)](https://www.displayfuture.com/Display/datasheet/controller/ST7796s.pdf)
Touch chipset: [XPT2046](https://grobotronics.com/images/datasheets/xpt2046-datasheet.pdf)

| Pin | GPIO | Notes                                                           |
|-----|------|-----------------------------------------------------------------|
|    1 |    0 | Reserved for UART0 TX                                          |
|    2 |    1 | Reserved for UART0 RX                                          |
|    3 |  GND | Ground                                                         |
|    4 |    2 | SPI0 SCK, used for display & touch (shared)                    |
|    5 |    3 | SPI0 TX (MOSI), used for display & touch (shared)              |
|    6 |    4 | SPI0 RX (MISO), used for display & touch (shared)              |
|    7 |    5 | Chip select for the display                                    |
|    8 |  GND | Ground                                                         |
|    9 |    6 | Reset for the display & touch (shared?)                        |
|   10 |    7 | Register select (Data/Command) for the display                 |
|   11 |    8 | Chip select for the touch                                      |
|      |      |                                                                |

#### Software

TFT_eSPI library version: 2.5.43

I use PlatformIO with `earlephilhower` Arduino Core:
```ini
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
```

User setup done also via PlatformIO build flags:
```ini
	; TFT driver configuration
	-DUSER_SETUP_LOADED=1
	-DST7796_DRIVER=1
	-DTFT_WIDTH=320
	-DTFT_HEIGHT=480
	-DTFT_MISO=4
	-DTFT_MOSI=3
	-DTFT_SCLK=2
	-DTFT_CS=5
	-DTFT_DC=7
	-DTFT_RST=6
	;-DTFT_BL=-1 (hardware)
	-DTOUCH_CS=8
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=80000000
	; -DSPI_READ_FREQUENCY=20000000 ; same as default
	; -DSPI_TOUCH_FREQUENCY=2500000 ; same as default
```

Minimalistic code would be (approximately):

```cpp
// ... #include's and tft object

void setup() {
	tft.init();
	tft.setRotation(1);
	// ... calibration (optional)
	tft.fillScreen(TFT_BLACK);
	tft.setTextFont(2); // most noticeable on this font
}

void loop() {
	tft.drawString("Hello! 123", 0, 0);
	
	uint16_t x, y;
	tft.getTouchRaw(&x, &y); // not even used, but it matters
}
```

On the photo below I used different code, but it all comes to the drawing. If necessary I might confirm the problem with the minimalistic code itself. 

### Result

String, the characters are drawn, but last character has weird, random blue/red/black dots artefacts. 

#### Photo

[image]

### Expected result

Normal operation without any glitches/artefacts.

### Workarounds

1) Use `tft.setTextColor(TFT_WHITE);` - it makes rendering (font 2) characters using slower method, which cuts the glitch visibility.
2) OR, Use different font which uses flower drawing, again, cuts the glitch visibility. 
3) OR, add some `delay(2)` before using touch related function (2 ms just from my testing on my setup).
4) ~~OR, add spaces to the end of the text to make last character readable, glitch persists.~~.
5) OR, maybe lowering display SPI speed could help (but it makes it slower to draw/update, so it's sad).

### Other notes

I guess it's due to (shared) SPI speed changes: graphics operations are put to SPI buffer to be pushed out, but before it finishes the speed changes, which makes mess. I tried putting `SPI_BUSY_CHECK` in the beginning of `TFT_eSPI::begin_touch_read_write` but it didn't help. 

Bug is most noticeable with font 2 due to `Faster drawing of characters and background using block write` in `TFT_eSPI::drawChar`. Making it slower by setting `textcolor == textbgcolor` by `setTextColor(...)` (which makes it use a lot of `drawPixel` calls), or using different font (which doesn't have fast drawing) makes it far less noticeable.

Bodmer, 2024-04-07:
See this post and check if your display needs a modification.
https://github.com/Bodmer/TFT_eSPI/discussions/898

closed 2024-04-10 by Bodmer

AgainPsychoX, 2024-04-13:
Yes, it seems it's working without glitches now. Thank you.