Bodmer/TFT_eSPI #3270: Vertical offset text

PE5PVB, 2024-04-04:
I use drawString with ttf converted smooth fonts. I made a ticker in my script using a sprite. Unfortunatly the space between te text rows is little too small. So when I have characters like Ö in it, I can see the dots of the Ö at the bottom of the sprite when they are out of the sprite. I want to add more space between the text lines to avoid this. Is there a command for this?

PE5PVB, 2024-04-05:
Here a video of the effect:
https://www.youtube.com/watch?v=-WQ4mnL89J8

Bodmer, 2024-04-10:
Increase the spacing here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Smooth_font.cpp#L138

closed 2024-04-10 by Bodmer

PE5PVB, 2024-04-10:
Thanks. I added 4 to that formula. Problem solved.