Bodmer/TFT_eSPI #3273: It is impossible to get multiple displays to work on the CT7789 chip

Dimitry-Muzi, 2024-04-08:
In my ESP32 S3 project I need to use multiple 2.0" GMT 020-02 displays on ST7789 chip. Individually and in parallel using a common CS pin everything works fine, but there is absolutely no way to get the displays to work by using multiple GPIOs to the CS pins on each display and looping through them in the sketch to select the display. Even in the eyes example, only one display works.

Bodmer, 2024-04-10:
See here:
https://www.xtronical.com/multiplescreens/

The CS GPIO must be defined in the dual display eyes example.

closed 2024-04-10 by Bodmer

Dimitry-Muzi, 2024-04-11:
I wouldn't write if I hadn't tried everything. It's not even possible to force write to two screens at once.

Dimitry-Muzi, 2024-04-11:
> See here: https://www.xtronical.com/multiplescreens/
> 
> The CS GPIO must be defined in the dual display eyes example.
I wouldn't write if I hadn't tried everything. It's not even possible to force write to two screens at once. The displays simply do not initialize.

Bodmer, 2024-04-12:
It sounds like a electrical signal integrity issue then caused by the bus loading. Keep wires short and use extra GND wires to reduces ground bounce. The other possibility is the choice of GPIO and the start up bias level. Some GPIO must be in a certain state, does the processor boot and run with 2 displays connected? (ie. Print to serial etc.

Bodmer, 2024-04-12:
OK, did you put both CS lines low for the first tft.init() ?

Dimitry-Muzi, 2024-04-14:
Certainly. I even checked it with a multimeter - the signals are sent correctly, but the displays do not work, except for the one that is registered in the User Setup.

Bodmer, 2024-04-14:
The chip selects should not be defined in the user setup but managed by the sketch.
See here:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Animated_Eyes_2/wiring.ino

Dimitry-Muzi, 2024-04-15:
I tried and this option - it still doesn't work.