Bodmer/TFT_eSPI #3281: ESP32-2432S022 2.2" tft with ST7789 parallel driver not working with TFT_eSPI library

dart1280, 2024-04-15:
Have you got any further with a parallel interface TFT? [https://www.aliexpress.com/item/1005006284154750.html](https://github.com/Bodmer/TFT_eSPI/issues/url) and [https://www.aliexpress.com/item/1005006622769773.html](https://github.com/Bodmer/TFT_eSPI/issues/url)

1/ display just displays the backlight - no other activity from any sketch

2/Arduino IDE

3/ TFT_eSPI library version 2.5.43

4/ ESP32 by Espressif version 2.0.15

5/ ESP32 dev

6/ TFT Driver tried both ST7789 and ST7789_2

7/ Parallel interface

8/ Read_User_Setup result

19:19:55.307 -> TFT_eSPI ver = 2.5.43
19:19:55.307 -> Processor = ESP32
19:19:55.307 -> Frequency = 240MHz
19:19:55.307 -> Transactions = Yes
19:19:55.307 -> Interface = Parallel
19:19:55.307 -> Display driver = 778B
19:19:55.307 -> Display width = 240
19:19:55.341 -> Display height = 320
19:19:55.341 ->
19:19:55.341 -> TFT_CS = GPIO 17
19:19:55.341 -> TFT_DC = GPIO 16
19:19:55.341 -> TFT_WR = GPIO 4
19:19:55.341 -> TFT_RD = GPIO 2
19:19:55.341 ->
19:19:55.341 -> TFT_D0 = GPIO 15
19:19:55.341 -> TFT_D1 = GPIO 13
19:19:55.341 -> TFT_D2 = GPIO 12
19:19:55.341 -> TFT_D3 = GPIO 14
19:19:55.341 -> TFT_D4 = GPIO 27
19:19:55.341 -> TFT_D5 = GPIO 25
19:19:55.341 -> TFT_D6 = GPIO 33
19:19:55.341 -> TFT_D7 = GPIO 32
19:19:55.341 ->
19:19:55.341 -> TFT_BL = GPIO 0
19:19:55.341 -> TFT_BACKLIGHT_ON = HIGH
19:19:55.341 ->
19:19:55.341 -> Font GLCD loaded
19:19:55.341 -> Font 2 loaded
19:19:55.341 -> Font 4 loaded
19:19:55.341 -> Font 6 loaded
19:19:55.381 -> Font 7 loaded
19:19:55.381 -> Font 8 loaded
19:19:55.381 -> Smooth font enabled

[Setup4_ESP322432S022.zip](https://github.com/Bodmer/TFT_eSPI/files/14971547/Setup4_ESP322432S022.zip)

[TFT_Meters.zip](https://github.com/Bodmer/TFT_eSPI/files/14971550/TFT_Meters.zip)

Specifications from supplier - 
[ESP32-2432S022 Specifications-EN.zip](https://github.com/Bodmer/TFT_eSPI/files/14976975/ESP32-2432S022.Specifications-EN.zip)

Display works fine using the supplied parameters with LVGL Library.

lsdlsd88, 2024-04-17:
pretty sure the issue is GPIO over 31 for data..

https://github.com/Bodmer/TFT_eSPI/issues/940

dart1280, 2024-04-19:
Hi Isdlsd88,
Thanks for your reply. If it is an issue with selecting data pins over 31 then I wonder how it is that this display works fine when using the LGFX library with the same data pin designations - 
```
LGFX_Sprite canvas;

class LGFX : public lgfx::LGFX_Device {

  lgfx::Panel_ST7789 _panel_instance;  // ST7789UI
  lgfx::Bus_Parallel8 _bus_instance;   // MCU8080 8B

public:
  LGFX(void) {
    {
      auto cfg = _bus_instance.config();
      cfg.freq_write = 25000000;
      cfg.pin_wr = 4;
      cfg.pin_rd = 2;
      cfg.pin_rs = 16;
      cfg.pin_d0 = 15;
      cfg.pin_d1 = 13;
      cfg.pin_d2 = 12;
      cfg.pin_d3 = 14;
      cfg.pin_d4 = 27;
      cfg.pin_d5 = 25;
      cfg.pin_d6 = 33;
      cfg.pin_d7 = 32;

      _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }
```
Is it possible that this over 31 issue is not the cause of the TFT_eSPI library not working with this configuration?

I do not have access to changing the pin designations as the display and uP are soldered on a pcb from the shop. True, they do 'push' the LGFX library in their manual supplied with the display but I think the TFT_eSPI library is nicer..... If It Works!

Anyhow... thanks.

lsdlsd88, 2024-04-19:
yeah im afraid is due to how this library manages writes on data pins. it says right in the user setup that you cant use them over 31. 

// Must use pins in the range 0-31 for the data bus
// so a single register write sets/clears all bits

it would be nice if a patch can be added optionally for supporting these boards!

dart1280, 2024-04-20:
Ah..I see. OK many thanks.  

It looks on your photo that you have patched gpio33 to gpio4, not 5, but seeing as 4 is already used on this board for TFT_WR it must just be a trick of the light!  I also think perhaps if I decide to do it I would swap them around to avoid the jumpers crossing - 33 to 26 and 32 to 5.  

I'm not 100% sure it would be worthwhile pursuing this too much more because the 2.2" variant seems a lot less popular than the 2.8" one that uses an IL9341 driver and doesn't suffer from this problem.

For anybody also following along I think this is the fix.

[image]

 Then change 

```
 TFT_CS = GPIO 17
 TFT_DC = GPIO 16
 TFT_WR = GPIO 4
 TFT_RD = GPIO 2

 TFT_D0 = GPIO 15
 TFT_D1 = GPIO 13
 TFT_D2 = GPIO 12
 TFT_D3 = GPIO 14
 TFT_D4 = GPIO 27
 TFT_D5 = GPIO 25
 TFT_D6 = GPIO 26
 TFT_D7 = GPIO 5
```

Thanks again for your time.

Mike

lsdlsd88, 2024-04-20:
yeah sorry for the confusion the red wire on gpio4 was just to take advantage of that touch pin to wake up from deep sleep.

yeah you can steal pins from the tft touch too. swapping them instead of bridging would involve cutting the small track that originally goes to the tft. doable but tricky

systematak, 2024-12-29:
I am trying to install Internet Radio, there is no screen information, when I install it, a white screen appears.
[image]
[image]

bascimehmet, 2025-01-14:
> I am trying to install Internet Radio, there is no screen information, when I install it, a white screen appears. ![20241221_201250](https://private-user-images.githubusercontent.com/22760432/399192164-ae4dab30-3c01-4d6c-bada-71d286402d0d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4Mzk1NjIsIm5iZiI6MTczNjgzOTI2MiwicGF0aCI6Ii8yMjc2MDQzMi8zOTkxOTIxNjQtYWU0ZGFiMzAtM2MwMS00ZDZjLWJhZGEtNzFkMjg2NDAyZDBkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDA3MjEwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM5N2VhMTVmNjBmMTA2ZDFhOWZkMzAyNmM0YTM3MGU1ZDJiZmJjMmY5MTUzODRmOTZlYTFhNTM2MGJkYWU2MDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.5id2NukH7AqD7fdS-fsfDptzt7Fa3iLDwhgMz2-34xw) ![20241221_201113](https://private-user-images.githubusercontent.com/22760432/399192166-db756d35-4150-49ab-8f29-e8e8f6a7b061.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4Mzk1NjIsIm5iZiI6MTczNjgzOTI2MiwicGF0aCI6Ii8yMjc2MDQzMi8zOTkxOTIxNjYtZGI3NTZkMzUtNDE1MC00OWFiLThmMjktZThlOGY2YTdiMDYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDA3MjEwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY2NDcwZWE2MTg5YTJjOTM5ODJjOGE0Y2JlMTZjZDI0MDI5NjYwMjNjZjAxODY1ZTdkYWY0MjliMjQwYTg3ZTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nfza5Yv-LM-unHVY8-wKU6sqcsvSfSAJSCl_0kRE1tQ)

how is going your project ??