Bodmer/TFT_eSPI #3284: esp32 2.0.15 and tft_espi keeps restarting esp32c3 mini

sjorsvr, 2024-04-17:
Hi,
I use arduino ide 2.3.0,  tft_espi 2.5.43 library, and esp32 by espressif 2.0.15 package , and the esp32c3 super mini module.
The following code lights the led 1 time.
But when I uncomment // tft.init() the led keeps blinking, so the module keeps restarting.

It does not happen when I use esp32 package 2.0.14

Any idea what the problem is ?
Regards
Sjors

code :
/*

 */

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

const byte led8 = 8;
byte   UseFont;
char str[80];

void setup(){
//  Serial.begin(115200);
//  delay(2000);

  pinMode(8, OUTPUT);
  digitalWrite(8, HIGH);
  delay(1000);
  digitalWrite(8, LOW);
  delay(1000);
  digitalWrite(8, HIGH);

//  tft.init(); // tft.begin(); zijn hetzelfde
}

void loop(){
}

elgerg, 2024-04-17:
I think you will need to post your user setup so people know what pins you are trying to use.

sjorsvr, 2024-04-17:
my user setup :
// ST7789 240 x 280 display with no chip select line
#define USER_SETUP_ID 203

#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 320

//#define CGRAM_OFFSET      // Library will add offsets required

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

//#define TFT_INVERSION_ON
//#define TFT_INVERSION_OFF

// DSTIKE stepup
//#define TFT_DC    23
//#define TFT_RST   32
//#define TFT_MOSI  26
//#define TFT_SCLK  27

// Generic ESP32 setup
//#define TFT_MISO 19
//#define TFT_MOSI 23
//#define TFT_SCLK 18
//#define TFT_CS    -1 // Not connected
//#define TFT_DC    2
//#define TFT_RST   4  // Connect reset to ensure display initialises

//ESP32 C3 generic default
//TFT_CS 7
//TFT_MOSI 6
//TFT_MISO 5
//TFT_SCLK 4

#define TFT_CS   1
#define TFT_MOSI 6
#define TFT_MISO -1 //5
#define TFT_SCLK 4

#define TFT_DC    10
//#define TFT_RST   10
#define TFT_RST   -1

//#define TOUCH_CS  1 // Optional for touch screen
#define TOUCH_CS  -1 // Optional for touch screen

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
//#define TFT_CS   7      // Define as not used
//#define TFT_DC   8      // Data Command control pin
//#define TFT_RST  PIN_D4  // TFT reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1      // TFT reset pin connect to NodeMCU RST, must also then add 10K pull down to TFT SCK

//#define TOUCH_CS -1

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

 #define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS

cgrobi, 2024-04-23:
The same happened to me. I use an ESP32-S3 and the TFT_eSPI library and an ST7796 display. With version 2.0.14 everything works fine. Under 2.0.15 the controller reboots everytime I call tft.init(). Even some old projects that worked for months or even years are not unsable under 2.0.15.

HSBallina, 2024-04-23:
#3289 Linking in this issue as well, since it's related.

sjorsvr, 2024-05-04:
esp32 2.0.16 same behaviour as 2.0.15.
Is this a esp32 or a tft_espi issue ?

HSBallina, 2024-05-05:
Not sure TBH. Seems like there were some 3.0 features that snuck into the 2.0.15 release that ended up being breaking changes. I just realized there's a new release out since two days back: https://github.com/espressif/arduino-esp32/releases/tag/2.0.16. Too late for me to test it out now, but I'll give it a go tomorrow.

SergePD, 2024-05-05:
ESP32-S3 2.0.14 was working with TFT_eSPI, 2.015 and 2.0.16 - Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

dbtronics, 2024-05-06:
Plese refer to this issue: https://github.com/Bodmer/TFT_eSPI/issues/3304#issuecomment-2095080442

Revert your esp32 board package to 2.0.14 and that should solve the reset loop. The latest version are causing issues with memory and is not compatible with the TFT_eSPI library. This is tested for ESP32-S3 and I can confirm for this.

HSBallina, 2024-05-06:
@dbtronics it has already been established that there's an incompatibility with both 2.0.15 and 2.0.16.

moeburn, 2024-05-06:
Thanks for these comments guys, I was pulling my hair out trying to figure out why my code no longer worked on my ESP32-C3, just reboots forever.  

It was a Store Access Fault at MTVAL 0x00000010 if that helps.

JimDrewGH, 2024-05-15:
I can also confirm the same issues exists with a custom board using a ESP32-C3-Mini module.  Ironically, PlatformIO worked perfectly with the same Arduino setup that did not - until I upgraded the ESP32 core from v6.6.0 to v6.7.0 and the same thing happens (constant panic/reset) with PlatformIO.

HSBallina, 2024-05-15:
> I can also confirm the same issues exists with a custom board using a ESP32-C3-Mini module. Ironically, PlatformIO worked perfectly with the same Arduino setup that did not - until I upgraded the ESP32 core from v6.6.0 to v6.7.0 and the same thing happens (constant panic/reset) with PlatformIO.

IIRC PlatformIO v. 6.6.0 uses ESP32 v. 2.0.14. Probably got updated to either 2.0.15 or 2.0.16 in v. 6.7.0.

dbtronics, 2024-05-15:
Yes I can confirm for ESP32-S3-devkitc-1 for platformio. Have to go with V6.6.0 for screens to work.

Jason2866, 2024-05-16:
The bug is here in the library https://github.com/Bodmer/TFT_eSPI/blob/fae22f785f8b4f7970195487826a348d08d082ae/Processors/TFT_eSPI_ESP32_C3.h#L30-L35
same for the S3.
See open issue in Arduino repo https://github.com/espressif/arduino-esp32/issues/9618#issuecomment-2113344119

toddbu, 2024-06-02:
Ugh, this is really painful. I'm trying to use LoRa with a TFT display on a C3-zero so I need both the FSPI and HSPI buses. Thankfully, LoRa allows me to attach my own SPI bus (setSPI) so I just created a new FSPI bus. But it broke under `2.0.14` so I upgraded to `2.0.15` and LoRa worked ok on the FSPI bus. But then this bug blew up my TFT that had been working on HSPI. I  tested against .15, .17, and the newly released `3.0.0` and all appear to fail in the same way

illigtr, 2024-07-25:
I can confirm I have exactly the same issue using the Waveshare ESP32-S3-PICO board with the Waveshare Touch-LCD-3.5 Display HAT. I have tested all versions of esp32 board drivers from v2.0.9 to v3.0.3. The problem starts with v2.0.15. I have also tests with TFT_eSPI versions 2.5.31 through 2.5.43 and all versions start failing with esp32 v2.0.15.

I did notice a warning in the compiler with v2.0.14 concerning UART.

```
C:\Users\tony_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c: In function 'uartSetPins':
C:\Users\tony_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
C:\Users\tony_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\cores\esp32\esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
      ^~~~~~~~~~~

```

This is my core debug message when using v2.0.15

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202704e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   326][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42002826  PS      : 0x00060a30  A0      : 0x820028fc  A1      : 0x3fcebc00  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000024  
A6      : 0x000000ff  A7      : 0x80000001  A8      : 0x08000000  A9      : 0x3fcebbd0  
A10     : 0x3fc95618  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x00000100  A15     : 0x3fc91fe8  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x42006490  LEND    : 0x420064f4  LCOUNT  : 0x00000002  

Backtrace: 0x42002823:0x3fcebc00 0x420028f9:0x3fcebc30 0x42002904:0x3fcebc50 0x4200194a:0x3fcebc70 0x4200700e:0x3fcebc90
```

Also, a slightly more detailed error message from core debug using esp32 v3.0.3 (latest).

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379686
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
[   103][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x420081c0
[   114][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200818c
[   125][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42008158
[   137][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42008124
[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x420081c0
[   159][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200818c
[   171][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42008158
[   182][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42008124
[   193][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x420081c0
[   205][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4200818c
[   216][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42008158
[   227][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42008124
[... 68 lines cut ...]
[   717][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SS (37) successfully set to 0x420071dc
[   729][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4202b804
[   740][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 35 successfully set to type GPIO (1) with bus 0x24
[   750][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 35 successfully set to type SPI_MASTER_SCK (34) with bus 0x1
[   761][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4202b804
[   772][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 37 successfully set to type GPIO (1) with bus 0x26
[   782][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 37 successfully set to type SPI_MASTER_MISO (35) with bus 0x1
[   793][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x4202b804
[   804][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 36 successfully set to type GPIO (1) with bus 0x25
[   814][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 36 successfully set to type SPI_MASTER_MOSI (36) with bus 0x1
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
[... cut ...]
A10     : 0x3fc953f0  A11     : 0x00000001  A12     : 0x01312d00  A13     : 0xffffffff  
A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x42007864  LEND    : 0x420078ab  LCOUNT  : 0x00000002  

Backtrace: 0x42002cf3:0x3fcebc10 0x42002dc9:0x3fcebc40 0x42002dd4:0x3fcebc60 0x42001e7a:0x3fcebc80 0x42008703:0x3fcebca0 0x4037d85a:0x3fcebcc0

```

Does anyone have a workaround or update on this issue? I'll keep using v2.0.14 of esp32 drivers in the meantime.

sjorsvr, 2024-09-27:
@Bodmer, any chance that you can work on this issue ? Would be appreciated a lot. Then we can upgrade to esp32 version 3.0.5. We are now stuck at 2.0.14. Thanks

elgerg, 2024-09-27:
Just a quick one to try and link these issues together.

There has been some success by adding in extra defines based on this:
https://github.com/Bodmer/TFT_eSPI/issues/3329

I'd be looking out for feedback from:
https://github.com/Bodmer/TFT_eSPI/issues/3329#issuecomment-2370475614

I know it might not be a solution but hopefully it will help us. @Bodmer seems to be quite absent of late and it kind of worries me.

sjorsvr, 2024-09-27:
added the 3 lines mentioned in #3329 to my user_setup.h and just tested on arduino 2.3.0, esp32 3.0.5 : it worked : the esp32s3 zero did not reboot anymore and I ran 2 different sketches successful. thanks elgerg.
I will not close those this issue : final solution may be provided by Bodmer

sjorsvr, 2024-10-19:
it later turned out that this was not a solution, at first glance sketches seem to run, but they crashed later on. Reverted back to esp 2.0.14

mtwombley, 2024-12-01:
Does anyone know how to fix this on Platform IO? I got the fix working with Arduino but the same fix didn't work in Platform IO.

senkisemsem, 2025-02-17:
Some idea, I had the same problem, but then I found a tip on Redit, enable the following line in your User_Setup.h
#define USE_HSPI_PORT
Now for me it works like a charm!

HebertWP, 2025-02-18:
@senkisemsem, thanks, just complementing,  if using idf menuconfig use the HSPI option on the TFT component. 
Note: I was going to code the UI yesterday but procrastinated, that was an excellent decision :D as your comment is 18 hours old.

RobinWung, 2025-05-02:
I was inspired by this comment: [https://github.com/Bodmer/TFT\_eSPI/issues/3329#issuecomment-2585954379](https://github.com/Bodmer/TFT_eSPI/issues/3329#issuecomment-2585954379).
It turns out the issue was likely caused by an incorrect `SPI_HOST` setting. After changing it, everything worked fine.

I'm using PlatformIO Core 6.1.18 · Home 3.4.4 with an ESP32-C3.

To fix the issue, I had to modify line 71 of
`.pio/libdeps/airm2m_core_esp32c3/TFT_eSPI/Processors/TFT_eSPI_ESP32_C3.h`
from:

```cpp
#define SPI_PORT SPI2_HOST
```

to:

```cpp
#define SPI_PORT SPI3_HOST
```

Hope this helps others facing the same problem!

infrafast, 2025-05-16:
> #define USE_HSPI_PORT

This also fixed the issue on my ESP32-S3 xiao seeed with platform = espressif32@6.10.0 ;

HonestQiao, 2025-08-09:
I use it [](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-lcdkit/user_guide.html)
and I also encountered the same situation.

If using version 2.0.14, everything is normal.
If using a later version, even if defined in TFT_eSPI_SSP32_C3. h:
```
#define SPI_PORT SPI2_HOST
```
It's also useless.

niu541412, 2025-08-22:
@HonestQiao, see my reply and the attachment https://github.com/Bodmer/TFT_eSPI/issues/3743#issuecomment-3145633679 . It should work in v3.0.0+.