Bodmer/TFT_eSPI #3289: Guru Meditation Error on tft.begin() (ESP32-S3 dev module)

HSBallina, 2024-04-18:
I'm having  issues with my ESP32-S3 module crashing and rebooting on `tft.begin()` (or on `tft.init()`). The following output repeats over and over in the monitor:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (99) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='db82bb1c'
Image checksum='ffffffff'
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4200289b  PS      : 0x00060a30  A0      : 0x82002928  A1      : 0x3fcebd80  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000015  
A6      : 0x000000ff  A7      : 0x00000001  A8      : 0x08000000  A9      : 0x3fcebd50  
A10     : 0x3fc952b0  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000010  
A14     : 0x00000000  A15     : 0x3fc91d98  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x42004dec  LEND    : 0x42004e4f  LCOUNT  : 0x00000003  

Backtrace: 0x42002898:0x3fcebd80 0x42002925:0x3fcebdb0 0x42002930:0x3fcebdd0 0x4200184d:0x3fcebdf0 0x420058da:0x3fcebe20
```
Using the [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to decode the dump (only works in Arduino 1.8.19) with the [ELF file](https://github.com/Bodmer/TFT_eSPI/files/15031350/minimal_tft.ino.elf.zip) from the build gives this:
```
PC: 0x4200289b: TFT_eSPI::writecommand(unsigned char) at C:\Users\stompp\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 81
EXCVADDR: 0x00000010

Decoding stack results
0x42002898: TFT_eSPI::writecommand(unsigned char) at C:\Users\stompp\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 81
0x42002925: TFT_eSPI::init(unsigned char) at C:\Users\stompp\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 692
0x42002930: TFT_eSPI::begin(unsigned char) at C:\Users\stompp\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp line 603
0x4200184d: setup() at C:\DEV\ESP32\minimal_tft/minimal_tft.ino line 7
0x420058da: loopTask(void*) at C:\Users\stompp\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\cores\esp32\main.cpp line 42
```
I've minimized the code to the following:
```C
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();
void setup() {
  Serial.begin(115200);
  tft.begin();
}

void loop() {
}
```
My setup is the following:
IDE: Arduino 1.8.19 & 2.3.2
TFT_eSPI version: 2.5.43
Board package version:  [2.0.15](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)
Processor: [ESP32-S3](https://github.com/Freenove/Freenove_ESP32_S3_WROOM_Board)
TFT driver: [GC9A01](https://www.waveshare.com/wiki/1.28inch_LCD_Module?amazon)
Interface: SPI

The [setup file](https://github.com/Bodmer/TFT_eSPI/files/15031361/Setup406_GC9A01_ESP32.zip) is a copy of 46 with a different MOSI pin. It's added to the `User_Setup_Select.h` file and the `User_Setup.h` is set to use the GC9A01 driver.

dbtronics, 2024-05-06:
Please refer to this link: https://github.com/Bodmer/TFT_eSPI/issues/3304#issuecomment-2095080442

Revert your ESP32 board package to 2.0.14 will solve this issue.

blackCmd, 2025-02-25:
@AlexBestoso 
This issue is not related to the configuration of the SPI pins. This is because there is no error in version 2.0.14, but there is an infinite reboot error in version 2.0.15. This problem has not been fixed as of February 25, 2025. There is an infinite reboot bug present in all ESP32 and ESP32-C3 devices.

AlexBestoso, 2025-02-26:
If I recall my interaction with this issue, it was for me related to the configuration of the pins. 

I believe HSPI ended up working for me reliably,

And that the issue was related to a change in how the code handled FSPI, HSPI, and VSPI.

I'm currently running the most recent version of TfT_espi on an esp32-s3 using dma and have no issues.

Again returning back to your spi configuration (as frustrating as that sounds) i believe is the issue.

The code the boot loop is a result in an invalid memory address being invoked from an improperly configured register.

AlexBestoso, 2025-02-27:
I agree with that.

The nature of the software issue, from pulling my hair out to make it work, is in how the spi bus gets used and initalized.

For me to make the system work, I had to put `#define USE_HSPI` into my User_Setup file.

No other mode worked for me. HSPI must be used from what I can gather, else you get the load prohibited error.

You can still use multiple other spi slaves with HSPI mode. 

If you want to use dma, you must deinitalize dma before using other spi devices.

I wrote a class that spits out spi register values if you need that to further work around.

vishnumaiea, 2025-05-09:
@OnkarRuikar Your solution of enabling the `USE_HSPI_PORT` flag fixed an infinite reboot issue on the **Waveshare ESP32-S3 2.8 inch LCD** using the ST7789 controller. Thanks.

Weasel1202, 2025-07-16:
it said "HSPI Does not have default pins on ESP32S3"
what can i do now?
pls help me :")))

tarunraj-excellentwebworld, 2025-08-13:
> I agree with that.
>
> The nature of the software issue, from pulling my hair out to make it work, is in how the spi bus gets used and initalized.
>
> For me to make the system work, I had to put `#define [...]

I was facing the same problem. Even I thought that it must be the problem of GPIOs that I have configured but it was not. 
I removed the commented **#define USE_HSPI_PORT** and it stopped throwing error. 
Thanks! 👍

FitzPatrick2002, 2026-01-18:
Yeah, I was having the same issue. 
The #define USE_HSPI_PORT did solve the problem.
Thank you~
. ^^ .

ikouchiha47, 2026-08-30:
Here is the solution, switch to Adafruit libraries:

```c
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
```