Bodmer/TFT_eSPI #3308: Can't get it to play with POE Ethernet

DrJaymz, 2024-05-03:
I am using a Olimex ESP32-POE-ISO board which has Ethernet.  The ethernet uses LAN8720 over the RMII interface - but I don't know that much about how it works.

I am using the following config for a TFT display:

 -DUSER_SETUP_LOADED=1
    -DILI9341_DRIVER=1       ; Switch to ILI9341
    -DTFT_WIDTH=240          ; Set width to 240 pixels
    -DTFT_HEIGHT=320         ; Set height to 320 pixels for ILI9341
    -DTFT_RST=4             
    -DTFT_CS=16              
    -DTFT_DC=2               
    -DTFT_MOSI=15
    -DTFT_SCLK=14            
    #-DUSE_HSPI_PORT=1
    -DSUPPORT_TRANSACTIONS=1

And the good news is that the Display works perfectly. The bad news is that it kills the Ethernet stone dead.  So when pinging it dies as soon as tft.init() is called.  I put a 5 second delay in my setup so that you can see the pings working until init is called.

If I have USE_HSPI_PORT then ethernet dies and displays the message Eth disconnected.  If let it use VSPI then I don't see that message but it still kills the ethernet but the lights stay on - so only half dead then.  I turned on transactions because it says you need that IF you are sharing the SPI bus.  Well, my understanding is that actually the LAN8720 interface actually doesn't use SPI - but clearly SPI contention of some form is the issue here and it makes sense if I remapped HSPI during the INIT that would break it.  But what is VSPI's excuse?

I could post the code but nothing really interesting, I have the above config and then I just call tft.ini() and write hello world and that works.  For ethernet, there's an eth.init() and thats about it.  I poked into the eth.c and eth.h and they do have configs for multiple implementations which DO use SPI but not the one I'm using.

https://github.com/OLIMEX/ESP32-POE-ISO/blob/master/HARDWARE/ESP32-PoE-ISO-Rev.K/ESP32-PoE-ISO_Rev_K.pdf <-- here is the schematic so you can see where my pins are and which ones it uses for ethernet.

kapyaar, 2024-05-03:
Are you using the ETH SPI from ESP32? If so, It is likely the SPI conflict. I have raised this question in the discussion. See 
https://github.com/Bodmer/TFT_eSPI/discussions/3298

DrJaymz, 2024-05-03:
Possibly
~platformio\packages\framework-arduinoespressif32\libraries\Ethernet\src\ETH.cpp

But SPI shouldn't be in play in my scenario.  I wondered if it was some shared interrupt or something.