Bodmer/TFT_eSPI #3310: Using with RP2040-PiZero and 1.3inch display HAT

hasaranga, 2024-05-05:
Hi,

I'm trying to use this library with RP2040-PiZero board and 1.3inch display HAT.
Board: https://www.waveshare.com/wiki/RP2040-PiZero
Display: https://www.waveshare.com/1.3inch-lcd-hat.htm

I'm using Colour_Test example. But I there is nothing on screen. Backlight is working.

Here is my User_Setup.h content:

`
#define ST7789_DRIVER

 #define TFT_WIDTH  240 
 #define TFT_HEIGHT 240

 #define TFT_BL   24 
 #define TFT_BACKLIGHT_ON HIGH

#define TFT_MOSI  10
#define TFT_SCLK  11
#define TFT_CS    8
#define TFT_DC    25
#define TFT_RST   27

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define TFT_SPI_PORT 1
#define SPI_FREQUENCY  27000000

#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
`

istvanzk, 2024-05-05:
Hi @hasaranga. I had similar problem but for a different screen (https://learn.sparkfun.com/tutorials/sparkfun-micromod-input-and-display-carrier-board-hookup-guide). 
My problem was that the SPI settings were incorrect for these pins
```
#define TFT_MISO  xx
#define TFT_MOSI  xx
#define TFT_SCLK  xx
```
Additionally, the SPI frequency needed to be lowered in my case.
I hope it helps.

hasaranga, 2024-05-05:
No miso pin in my display. I tried to change the spi speed and mosi pin with sclk but the result is same.

hasaranga, 2024-05-05:
I was able to use **Adafruit-ST7735-ST7789** library successfully by configuring SPI1 pins manually. 

See here: [https://github.com/adafruit/Adafruit-ST7735-Library/issues/198](url)