Bodmer/TFT_eSPI #3323: ESP32-3248S035C with GT911 touch chip not working

eagl1, 2024-05-13:
Hi,

I have this module which is using ST7796 display driver and GT911 capacitive touch driver.

It come with lvgl example from the manufacturer. So I tried to upload a code with the Arduino IDE.

1. I set everything in the `User_Setup_Select.h`
2. I'm using these pin settings which is exactly what is in the schematic:

```
// for ESP32-3248S035C with GT911 touch chip board
#define TFT_MISO 12
#define TFT_MOSI 13 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 14
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   2  // Data Command control pin
#define TFT_RST  -1  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   27  // LED back-light
```

The schematic:

[image]

I tried to upload one of TFT_eSPI examples, no errors, the code uploaded successfully but the display is blank and nothing is working. I uploaded a code to test the onboard LED and it works.

eagl1, 2024-05-13:
One thing I used now is that I uploaded a code from this website:
[https://nightly.openhasp.com/](url)

[image]

It uploaded, and the display lighted up but I think there is no graphics in that binary.

Juaxs2022, 2024-08-29:
Did you solve it?

eagl1, 2024-08-29:
Well, I got it to work with this library:

[https://www.arduino.cc/reference/en/libraries/gfx-library-for-arduino/](url)

But I still didn't get the touch to work.

closed 2024-09-20 by eagl1

matthiasbirkich, 2025-02-17:
Dear eagl1, 

here there is the Link for a zip-file including the complete documentation for the ESP32-3248S035C cheap yellow display clone. The library for the display is **lovyangfx-Library**.  There is one example with the touch functions: LVGL_Arduino-3.5CTP-gt911.ino in the demo folder available.

http://pan.jczn1688.com/directlink/1/ESP32%20module/3.5inch_ESP32-3248S035.zip?spm=a2g0o.detail.1000023.1.106bewy8ewy8cC&file=3.5inch_ESP32-3248S035.zip

One issue in the documentation is, that some getting started information maybe is for the cheap yellow display ESP32-2432S028R! 
You have to adapt that for the pins and other settings in the setup.h for ESP32-3248S035 by yourself…

I hope, that will help also other people, which are not able to run the ESP32-3248S035C display (like me)!
Please give response, if it functioned. Thanx!

Best regards
Matthias

gho1b, 2025-03-15:
> Hi,
>
> I have this module which is using ST7796 display driver and GT911 capacitive touch driver.
>
> [image]
>
> [image]
>
> It come with lvgl example from the manufacturer. So I tried to upload a code with [...]

I got same issue, could you please tell me ho to solve this issue? The device only show white screen

Amnessie, 2025-03-17:
same here cant figure it out beyond a white display

Amnessie, 2025-03-17:
// User_Setup.h – Configuration for esp32E 3.5

#define ILI9341_DRIVER    //  Correct display driver for your board

#define TFT_WIDTH  240    //  ILI9341 resolution: 240x320
#define TFT_HEIGHT 320    

//  SPI Pin connections for esp32 devboard 
#define TFT_MOSI  13       // SPI MOSI
#define TFT_MISO  12       // SPI MISO
#define TFT_SCLK  14       // SPI Clock
#define TFT_CS    15       // TFT Chip Select
#define TFT_DC    2       // TFT Data/Command
#define TFT_RST   -1       // TFT Reset (or -1 if connected to EN/RST)

//  Backlight control
#define TFT_BL    27        // Backlight pin
#define TFT_BACKLIGHT_ON HIGH 

//  Touchscreen (XPT2046) SPI
#define TOUCH_CS 21         //  XPT2046 Touch Controller Chip Select

//  SPI Speed 
#define SPI_FREQUENCY  27000000  //  2MHz 
#define SPI_READ_FREQUENCY  20000000 //  16MHz for reads
#define SPI_TOUCH_FREQUENCY  2500000 //  2.5MHz for touch

tonbor, 2025-03-21:
capacitive touch no (XPT2046) but gt911 i2c

customwebprogrammer, 2025-05-14:
Thank you

> Dear eagl1,
>
> here there is the Link for a zip-file including the complete documentation for the ESP32-3248S035C cheap yellow display clone. The library for the display is **lovyangfx-Library**. [...]

> Best regards Matthias

THANK YOU so much, Matthias! I found your post while hunting for config advice for this exact CYD. It was $27 on a site that rhymes with mamazon.  It worked like a champ the first time I compiled.  I did find that to put the board into "download" mode, I needed to hold down the button near the reset button, the "boot" button in the diagram posted above, until the software download started from the arduino interface.  My key to finding this post was searching on ESP32-3248035C - this was one of the top-ranked articles! Thanks!

eagl1, 2025-06-19:
I got the touch to work on this display:

[image]

But I had to disable the interrupts. And I actually cut the interrupt line coming from the gt911 chip.

If I want to use the interrupts again, the I have to figure a way to deal with the interrupt pin that is pulled up all the time which is the problem I faced at the start.

jkuusama, 2025-07-26:
This is the first Google result on "esp32-3248s035c touch", so I'l share my results from tonight. I have the exact same product, and here is demo code with the display and touch working:.

```
#include <Arduino.h>

#include <TFT_eSPI.h> 
#include <SPI.h>
#include <Wire.h>
#include <TAMC_GT911.h>

TFT_eSPI tft = TFT_eSPI(); 

// Touch pins
#define TOUCH_SDA   33
#define TOUCH_SCL   32
#define TOUCH_INT   21
#define TOUCH_RST   25
#define TOUCH_WIDTH  480
#define TOUCH_HEIGHT 320

TAMC_GT911 tp = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, TOUCH_WIDTH, TOUCH_HEIGHT);

void setup(void) 
{
    Serial.begin(115200);
    Serial.println("Start");    
    tp.begin();
    tp.setRotation(ROTATION_RIGHT);

    tft.init();
    tft.setRotation(1);

    tft.fillScreen(TFT_WHITE);
    tft.setTextSize(1);
    tft.setTextColor(TFT_BLACK);
    tft.drawString("Hello, World!", 10, 10, 4);
}

void loop() 
{
    tp.read();
    if (tp.isTouched)
    {
        for (int i=0; i<tp.touches; i++)
        {
            Serial.print("Touch ");Serial.print(i+1);Serial.print(": ");;
            Serial.print("  x: ");Serial.print(tp.points[i].x);
            Serial.print("  y: ");Serial.print(tp.points[i].y);
            Serial.print("  size: ");Serial.println(tp.points[i].size);
            Serial.println(' ');
        }
    }
}
```
Couple of notes: Setup from first post is correct. Interrupt pin to touch IC is correct, but the resistor (R25) on the board is not installed, you need to install it to use touch interrupt. Rotation for touch is not the default.

eagl1, 2025-07-27:
What a good idea jkuusama !

I was thinking of merging the `TAMC_GT911` library into `TFT_eSPI`, but actually there's no need for that, it was just an idea to merge this touch driver in the `TFT_eSPI` library for more development to support more touch drivers.

But we can use them separately ! Yep, that will work, I will test it now.

eagl1, 2025-07-27:
I tried to connect the board but the PC won't recognize it. I think it's related to USB drivers because I modified the drivers because I used other boards.

[image]

But the board is working and the TFT is working too holding the last code. The problem only with the USB driver.

eagl1, 2025-07-28:
Yep, it worked thank you so much, now it's perfect.

[image]

**An important note:**
You must enable `PSRAM` from `Tools` menu.

eagl1, 2025-07-29:
Another sketch to print the coordinates on the display:

[image]

```
//https://github.com/Bodmer/TFT_eSPI/issues/3323#issuecomment-2880633387

#include <Arduino.h>

#include <TFT_eSPI.h> 
#include <SPI.h>
#include <Wire.h>
#include <TAMC_GT911.h>

TFT_eSPI tft = TFT_eSPI(); 

// Touch pins for GT911 for the ESP32_3248S035
#define TOUCH_SDA   33
#define TOUCH_SCL   32
#define TOUCH_INT   21
#define TOUCH_RST   25
#define TOUCH_WIDTH  480
#define TOUCH_HEIGHT 320

TAMC_GT911 tp = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, TOUCH_WIDTH, TOUCH_HEIGHT);

void setup(void) 
{
    Serial.begin(115200);
    Serial.println("Start");    
    tp.begin();
    tp.setRotation(ROTATION_RIGHT);

    tft.init();
    tft.setRotation(1);

    tft.fillScreen(TFT_WHITE);
    tft.setTextFont(4);
    tft.setTextSize(1);
    tft.setTextColor(TFT_BLACK);
    tft.drawString(F("Hello, World!"), 10, 10);
}

bool printTouchLock = 0;
uint32_t printTouchTimer;

void loop() 
{
  tp.read();
  if (tp.isTouched)
  {
    if(!printTouchLock){
      tft.fillScreen(TFT_WHITE);
      //tft.fillRect(0, 100, 400, 50, TFT_WHITE);
      

      tft.drawString(F("Touch:"), 10, 100);
      tft.drawString(F("  x: "), 10, 120);
      tft.drawNumber((uint16_t)tp.points[0].x, 40, 120);
      tft.drawString(F("  y: "), 10, 140);
      tft.drawNumber((uint16_t)tp.points[0].y, 40, 140);

      for (int i=0; i<tp.touches; i++)
      {
        Serial.print("Touch ");  Serial.print(i+1);Serial.print(": ");;
        Serial.print("  x: ");   Serial.print(tp.points[i].x);
        Serial.print("  y: ");   Serial.print(tp.points[i].y);
        Serial.print("  size: ");Serial.println(tp.points[i].size);
        Serial.println(' ');
      }     

      printTouchLock = 1;
      printTouchTimer = millis();
    }

    if(printTouchLock && (millis() - printTouchTimer > 500)){
      printTouchLock = 0;
    }
  }
}
```