Bodmer/TFT_eSPI #3329: library not working with espressif32 Arduino core > 2.0.14

Jason2866, 2024-05-16:
Device crashes at boot.
Open issue in Arduino repo https://github.com/espressif/arduino-esp32/issues/9618#issuecomment-2113344119
and
https://github.com/Xinyuan-LilyGO/T-Dongle-S3/issues/26

thelastoutpostworkshop, 2024-05-18:
I confirm, esp32-s3 reset under 2.0.16, when installing espressif arduino core 2.0.14 works fine

jfseaman, 2024-05-23:
The issue persists under 2.0.17.

cmachsocket, 2024-05-26:
i met the problem too.
I use 2.0.17 and esp32s3,and the program will reboot when it run tft.init()

StuartsProjects, 2024-06-01:
Same here too;

[https://github.com/Bodmer/TFT_eSPI/issues/3346#issuecomment-2143324416](url)

A program to transmer images over LoRa and display on a TFT was working on core 2.0.11, but stopped when I updgraded the core. 

By going backwards with the cores I found the last one that worked was 2.0.14. For later core versions the program would crash on start. 

ESP32S3, Arduino IDE 1.8.19.

youngsu999, 2024-06-04:
Hi there,

I'm suffering for same issue.
- ESP32S3, Arduino 2.0.17
I found some issues on espressif32 Arduino fimeware.

the system reboots when call `TFT_WRITE_BITS` and i found out the macro uses the below defines

[image]

```
  #define _spi_cmd       (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT))
  #define _spi_user      (volatile uint32_t*)(SPI_USER_REG(SPI_PORT))
  #define _spi_mosi_dlen (volatile uint32_t*)(SPI_MOSI_DLEN_REG(SPI_PORT))
  #define _spi_w         (volatile uint32_t*)(SPI_W0_REG(SPI_PORT))
```
the macros such as `SPI_CMD_REG` uses `REG_SPI_BASE` and the SPI ports are define as 0/1 (HSPI/VSPI)

```
#define SPI_CMD_REG(i)          (REG_SPI_BASE(i) + 0x0)
```

however, in the defintion on REG_SPI_BASE, it always returns Null address (because of  `(i) >=2` )

```
#define REG_SPI_BASE(i) (((i) >= 2) ? (DR_REG_SPI2_BASE + (i - 2) * 0x1000) : (0)) // GPSPI2 and GPSPI3
```

And i fix it as ->  

```
#define REG_SPI_BASE(i) (((i) <= 2) ? (DR_REG_SPI2_BASE + (i - 2) * 0x1000) : (0)) // GPSPI2 and GPSPI3
```

Then it stops to reboot.

However, the macro was stuck in following lines.

[image]

Anyone can figure out why?

-----
1. espressif arduino firmware seems to have bugs.
2. I have no idea why the macro stuck on while in the macro.

Thanks.

juniorgarcia, 2024-06-06:
Hi guys. Very frustrating dealing with that, right? I use PlatformIO and what fixed the issue to me is changing my platformio.ini to:

```ini
; ...
[env:lilygo-t-display-s3]
platform = espressif32@6.5.0 ; This platform version works fine as it uses Arduino Core 2.0.14
board = lilygo-t-display-s3
framework = arduino
;...
```

Here is the release notes for espressif32 version 6.5.0 https://github.com/platformio/platform-espressif32/releases/tag/v6.5.0

After changing that everything works like a charm! :D

Jason2866, 2024-06-06:
@juniorgarcia Have your read the issue head line?

juniorgarcia, 2024-06-06:
> @juniorgarcia Have your read the issue head line?

I did. Just wanted to point out how to fix the problem using PlatformIO. No need for this patronizing questions.

gitboysimon, 2024-06-07:
I also have this issue, device re-starts continuously on tft.init()
Was previously using platform espressif32 6.5.0 and framework 2.0.14 on new pc I now have 6.7.0 and 2.0.16, I reverted to platform = espessif32@6.5.0 and and it's ok with 2.0.16

SergePD, 2024-06-09:
#define USE_HSPI_PORT
Yes, It works for me (esp32S3)

thelastoutpostworkshop, 2024-06-09:
#define USE_HSPI_PORT
Works for me also in the device setup header file.
thanks

PhilG1300, 2024-06-09:
For my ESP32S3 adding #define USE_HSPI_PORT in the variants/esp32s3/pins_arduino.h file in ESP32 version 3.0.1 works too. Had to make other changes for the backlight control as ledc... functions have changed in version 3.
Thanks for the suggestion as I thought I was stuck on version 2.0.14

jfseaman, 2024-06-09:
> #define USE_HSPI_PORT Yes, It works for me (esp32S3)
I am working from a Volos Projects GC9A01 example. I just like his clock face. His code uses a GFXglyph.

I tried defining HSPI and using the latest 3.01 from Espressif. The number of defines lost/forgotten renders code uncompilable. Basically every TFT_xxx define is lost. The GFXglyph typdef is lost.

I have common code that is running on ESP32S3DevKit1, Adafruit QT Py S3 and XAIO ESP32C3.

Board definitions after 2.0.14 just don't work. Compiler is broken so bad it is unusable.

bolkedebruin, 2024-06-14:
Seeing the issue here as well, but defining USE_HSPI_PORT is not working.

jack-changyan, 2024-06-14:
I also see this problem here. I also define USE_HSPI_PORT and it doesn't work.

JooJooBee666, 2024-06-21:
I have the same problem.  The #define USE_HSPI_PORT seems to work for me, or rolling espressif back to 6.6.0.

I'm using IL9341 with Touch Screen in latest TFT_eSPI on a ESP32-S3 N16R8 (enabling or disabling the PSRAM has no effect).  The screen has an SDCard  reader but I'm not using it. This happens on a base scetch with nothing else in it except for the TFT_eSPI import, the SPI.h import and tft.begin() in the setup so it's clearly some issue with a change made in the espressif framework.  Here's my backtrace:

```
/home/deadlink/Arduino/projects/PlatformIO/SHTC3Testing/.pio/libdeps/esp32-s3-n16r8/TFT_eSPI/TFT_eSPI.cpp:81
/home/deadlink/Arduino/projects/PlatformIO/SHTC3Testing/.pio/libdeps/esp32-s3-n16r8/TFT_eSPI/TFT_eSPI.cpp:692
/home/deadlink/Arduino/projects/PlatformIO/SHTC3Testing/.pio/libdeps/esp32-s3-n16r8/TFT_eSPI/TFT_eSPI.cpp:603
/home/deadlink/Arduino/projects/PlatformIO/SHTC3Testing/src/main.cpp:35 (discriminator 1)
/home/deadlink/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```
Basically pointing back to the writecommand method that @youngsu999 mentioned earlier.

jfseaman, 2024-07-04:
I managed to adapt code to the esp32 3.0.x Arduino IDE  board definitions. Many things have changed.

Setting  USE_HSPI_PORT works for all ESP32S3.

Does not work for ESP32C3.

jfseaman, 2024-07-10:
I replaced all my ESP32C3 with ESP32S3. 

Now able to use ESP32 board definitions 3.0.2 in Arduino-IDE with USE_HSPI_PORT.

A few dollars and hours and hours of diag for no gain other than current IDE.

xhy2008, 2024-07-20:
The #define USE_HSPI_PORT worked for me,but I'm worried that it will slow down the display.

confusedMech, 2024-07-22:
Same here, When I use #define USE_HSPI_PORT the ESP32-S3 stops crashing but then im unable to use an external SD card.  Going back to platformio Espressif version 6.6.0 allows everything to function properly.

oet-oet, 2024-08-18:
#define USE_HSPI_PORT
当前时间20240818，所有库都用当前最新版，使用这个宏定义，就好使了

Rollmops67, 2024-08-18:
Hello from France !
I was becoming crazy after updating the arduino core for ESP32 from 2.0.14 to 3.0.4 on the Arduino IDE (1.8.19)
After that one of my projects (using Bodmer's last TFT_eSPI library)  booted without end at restart, after compiling without errors.
The ESP32 is a LilyGo T7-S3.
I reverted than back the Arduino Core for Arduino version after version ( 3.0.4  > 3.0.3 ...... > 2.0.17 > 2.0.16 > 2.0.15 > 2.0.14)
and it worked again with 2.0.14.
So after doing a search on google I found this tread (and others)...
At least I now know where the problem is.
For me too [Kolodieiev](https://github.com/Kolodieiev)'s temporary solution works (thank you !), albeit it's just a workaround...

Programing is a neverending story...

Roland

khvolk, 2024-09-10:
I run into the same problem: Guru Meditation Error (What does this mean. Are the people kidding?):

XIAO ESP32-C3
st7789 240x240
Arduino IDE 2.3.2
ESP32 Core 3.0.4

Setting USE_HSPI_PORT was no solution for me. 

Downgrading to 2.0.14 worked for me. Does somebody working on solving this issue?

florinbaciuu, 2024-09-12:
Hello everyone. I tested this library (V2.5.43) on the following boards 
> Lilygo-T-Display-S3   ( parallel )
> Lilygo-T-HMI (S3)      ( parallel )
> Lilygo-T-QTPro-S3    ( `USE_FSPI_PORT `)
> Lilygo-T-Dongle-S3  ( `USE_FSPI_PORT `and `USE_HSPI_PORT`)
> ESP32S3dev-n16r8 with an ili9341 2.8 display  ( `USE_FSPI_PORT `and `USE_HSPI_PORT `)
> ESP32dev-n4 with an ili9341 2.8 display  ( without SPI define and with `USE_HSPI_PORT`)

with Arduino Core 3.0.4 and 3.0.3 in platformio and Arduino IDE and even in ESPIDF with arduino-as-component. 
The library works perfectly, I even used it together with the GUISlice library and everything was perfect except for the fact that I had to rewrite the code for touch(it has nothing to do with the problem), some having XPT2046(spi) (shared bus and separate bus) and others having cst816 (i2c) controller )... On some I had to put `USE_HSPI_PORT `(ESP32S3dev-n16r8 with ili9341), on others I had to put `USE_FSPI_PORT `(Lilygo-T-QTPro-S3) and on others it worked with every setup i tried. I can provide you the User_Config files for each board and I can even test on many esp32 variants. I own esp32, s2, s3, c3, c6 in various flavors and models.

jfseaman, 2024-09-12:
> Hello everyone. I tested this library (V2.5.43) on the following boards
>
>
> with Arduino Core 3.0.4 and 3.0.3 in platformio and Arduino IDE and even in ESPIDF with arduino-as-component. The library [...]

You have made it work for ESP32-C3 under core 3.0.4?

What config?

ChassaB, 2024-09-20:
> Hello everyone. I tested this library (V2.5.43) on the following boards
>
>
> with Arduino Core 3.0.4 and 3.0.3 in platformio and Arduino IDE and even in ESPIDF with arduino-as-component. The library [...]

That is great. I would love the User_Config flies for the Lily-T_Display - S3 and the ESP32S3dev-n16r8 with an ili9341 2.8 display

Mehdow, 2024-09-23:
> Hello everyone. I tested this library (V2.5.43) on the following boards
>
>
> with Arduino Core 3.0.4 and 3.0.3 in platformio and Arduino IDE and even in ESPIDF with arduino-as-component. The library [...]

Same here, I would love to have your platformio.ini and user_config files for T-Display S3, as 'm stuck in a "StoreProhibited" never-ending error regarding heap (funniest part is that the same code works perfectly with an arduino nano esp32....). Not sure if it's the same issue that everyone here has but I'm desperate to know how to proceed

florinbaciuu, 2024-09-24:
> > Hello everyone. I tested this library (V2.5.43) on the following boards
> > > Lilygo-T-Display-S3   ( parallel )
> > > Lilygo-T-HMI (S3)      ( parallel )
> > > Lilygo-T-QTPro-S3    ( `USE_FSPI_PORT `)
> > > Lilygo-T-Dongle-S3  ( `USE_FSPI_PORT `and `USE_HSPI_PORT`)
> > > ESP32S3dev-n16r8 with an ili9341 2.8 display  ( `USE_FSPI_PORT `and `USE_HSPI_PORT `)
> > > ESP32dev-n4 with an ili9341 2.8 display  ( without SPI define and with `USE_HSPI_PORT`)
> > 
> > 
> > with Arduino Core 3.0.4 and 3.0.3 in platformio and Arduino IDE and even in ESPIDF with arduino-as-component. The library works perfectly, I even used it together with the GUISlice library and everything was perfect except for the fact that I had to rewrite the code for touch(it has nothing to do with the problem), some having XPT2046(spi) (shared bus and separate bus) and others having cst816 (i2c) controller )... On some I had to put `USE_HSPI_PORT `(ESP32S3dev-n16r8 with ili9341), on others I had to put `USE_FSPI_PORT `(Lilygo-T-QTPro-S3) and on others it worked with every setup i tried. I can provide you the User_Config files for each board and I can even test on many esp32 variants. I own esp32, s2, s3, c3, c6 in various flavors and models.
> 
> You have made it work for ESP32-C3 under core 3.0.4 Arduino IDE?
> 
> What config?

Tonight I will try to attach a display or two to the c3 boards have and do some tests. I will come back with information later about the C3.

florinbaciuu, 2024-09-24:
> That is great. I would love the User_Config flies for the Lily-T_Display - S3 and the ESP32S3dev-n16r8 with an ili9341 2.8 display

Sure: 
-THIS IS THE CODE FOR LILYGOTDISPLAYS3:

```
#define USER_SETUP_ID 4000 // just put what you want 

#define ST7789_DRIVER // lilygo-t-display-s3
////#define ST7789_2_DRIVER
#define INIT_SEQUENCE_3 // Using this initialisation sequence improves the display image

#define CGRAM_OFFSET // lilygo-t-display-s3

/* Colour order Red-Green-Blue */
#define TFT_RGB_ORDER TFT_RGB // lilygo-t-display-s3
/* Colour order Blue-Green-Red */
////#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red

#define TFT_INVERSION_ON // lilygo-t-display-s3
////#define TFT_INVERSION_OFF

#define TFT_PARALLEL_8_BIT // lilygo-t-display-s3

#define TFT_WIDTH 170
#define TFT_HEIGHT 320

#define TFT_RST 5
#define TFT_CS 6 // Not connected
#define TFT_DC 7
#define TFT_WR 8
#define TFT_RD 9

#define TFT_D0 39
#define TFT_D1 40
#define TFT_D2 41
#define TFT_D3 42
#define TFT_D4 45
#define TFT_D5 46
#define TFT_D6 47
#define TFT_D7 48

#define TFT_BL 38
#define TFT_BACKLIGHT_ON HIGH // lilygo-t-display-s3

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT
```
Also for this board i use a init function for ST7789 display. Just call this function from setup after the `tft.init();` This is the code :
```
typedef struct
{
   uint8_t cmd;
   uint8_t data[14];
   uint8_t len;
} lcd_cmd_t;

lcd_cmd_t lcd_st7789v[] = {
    {0x11, {0}, 0 | 0x80},
    {0x3A, {0X05}, 1},
    {0xB2, {0X0B, 0X0B, 0X00, 0X33, 0X33}, 5},
    {0xB7, {0X75}, 1},
    {0xBB, {0X28}, 1},
    {0xC0, {0X2C}, 1},
[... 228 lines cut ...]

//#define TFT_SDA_READ

/*COLOR ORDER*/
#define TFT_RGB_ORDER (TFT_BGR)

/*TFT INVERSION*/
//#define TFT_INVERSION_ON

/*FONT DEFINE*/
#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
// #define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

/*SPI SETTINGS*/
#define SPI_FREQUENCY (40000000)
#define SPI_READ_FREQUENCY (6000000)
#define SPI_TOUCH_FREQUENCY (2500000)
```
For another boards i will test them in next days and i will post here the working setups.

florinbaciuu, 2024-09-24:
> Same here, I would love to have your platformio.ini and user_config files for T-Display S3, as 'm stuck in a "StoreProhibited" never-ending error regarding heap (funniest part is that the same code [...]

### My platformio.ini file is :
```
; PlatformIO Project Configuration File for ....lilygo-t-displays3

######################################################################################################

[platformio]
##-------------------------
workspace_dir = ./
core_dir = ${PROJECT_DIR}/core
platforms_dir = ${platformio.core_dir}/platforms
packages_dir = ${platformio.core_dir}/pkg
##-------------------------
src_dir = ${PROJECT_DIR}/src
libdeps_dir = ${PROJECT_DIR}/libdeps
lib_dir = ${PROJECT_DIR}/lib
data_dir = ${PROJECT_DIR}/data
include_dir = ${PROJECT_DIR}/include
boards_dir = ${PROJECT_DIR}/board
build_dir = ${PROJECT_DIR}/build
build_cache_dir = ${platformio.build_dir}/build_cache
test_dir = ${PROJECT_DIR}/test
monitor_dir = ${PROJECT_DIR}/monitor

default_envs = pioarduino_3_0_5
extra_configs = 

######################################################################################################

[env]
lib_ldf_mode = deep ;options : off, chain, deep, chain+, deep+
lib_archive = yes
lib_compat_mode = off
lib_extra_dirs = libraries 
				 components

######################################################################################################

[DEBUGx]
build_flags = 
	-DCORE_DEBUG_LEVEL=5
	-DCONFIG_ARDUHAL_LOG_COLORS=1

######################################################################################################

##      OPTIONS:  test - debug - release  ##
[BUILDx]
build_type = debug

######################################################################################################

[lilygotdisplays3] 
extends = env 
build_type = ${BUILDx.build_type}
board = lilygo-t-displays3
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
board_build.flash_size = 16MB
board_build.boot = qio         ;options: qio , dio
board_build.boot_freq= 80000000L
board_build.partitions = partition.csv
board_build.flash_mode = qio
board_build.f_flash = 80000000L
board_build.psram_type = opi
board_build.arduino.memory_type = qio_opi
board_build.filesystem = spiffs ;options: ffat , spiffs, littlefs

build.variants_dir = ${PROJECT_DIR}/variant
[... 120 lines cut ...]
#define SMOOTH_FONT

/************************************************** */
```
### **Optional my partition test:**
**partition.csv** is:
```
# ESP-IDF Partition Table,,,,,,
# Total 16384K (16 MB),,,,,,
#       Name, Type,  SubType, Offset,  Size, Flags,
# bootloader,     ,         , 0x1000,   32K,      , # located at the address 0x1000; Offset default 0x1000(4K);      
# partitions,     ,         , 0x8000,    4K,      , # Offset default 0x8000(32K);
         nvs, data,      nvs,       ,   16K,      , # Type:(0x02);
     otadata, data,      ota,       ,    8K,      , # Type:(0x00); Default: 8K(0x2000);
    phy_init, data,      phy,       ,    4K,      , # Type:(0x01);
     factory,  app,  factory,0x10000, 4096K,      , # Type:(0x00); offset default 0x10000(64K);  APP standard partition;
       ota_0,  app,    ota_0,       , 4096K,      , # Type:(0x10);
       ota_1,  app,    ota_1,       , 4096K,      , # Type:(0x11);
        ffat, data,      fat,       , 1024K,      , # Type:(0x81);
      spiffs, data,   spiffs,       , 1024K,      , # Type:(0x82);
#   littlefs, data, littlefs,       , 1024K,      , # Type:(0x83);
     eeproom, data,      nvs,       ,  892K,      , ### EEPROOM PARTITION
     nvs_key, data, nvs_keys,       ,    4k,      , # Type:(0x04); Default: 4kB(0x1000);
    coredump, data, coredump,       ,   64K,      , # Type:(0x03); Default: 64kB(0x10000);
```

ChassaB, 2024-09-25:
Thanks. That's great. I'll try it out in a few days

lapinos03, 2024-09-25:
On Arduino IDE 2.3.3, I succeeded to compile and run on a ESP32S3+ST7789 (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-2.8) the demo provided for this boad with the latest versions of espressif (3.0.5) and TFT_eSPI (2.5.43) by putting these lines:
```
// FSPI port (SPI2) used unless following defined
#define USE_HSPI_PORT
#define ESP32 1
#define CONFIG_IDF_TARGET_ESP32S3 1
```  

For this board, the complete **tft_setup.h** : 

```
//#define USER_SETUP_ID 305
//#define USER_SETUP_INFO "WAVESHARE_ESP32-S3-TOUCH-LCD-2.8"

#define USER_SETUP_LOADED

#define ST7789_DRIVER

#define TFT_WIDTH  240
#define TFT_HEIGHT 320

#define TFT_MISO -1
#define TFT_MOSI 45
#define TFT_SCLK 40
#define TFT_CS   42
#define TFT_DC   41
#define TFT_RST  39
#define TFT_BL   5
#define TFT_BACKLIGHT_ON HIGH

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6 
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_FONT8N
#define LOAD_GFXFF 

#define SMOOTH_FONT

// FSPI port (SPI2) used unless following defined
#define USE_HSPI_PORT
#define ESP32 1
#define CONFIG_IDF_TARGET_ESP32S3 1

#define SPI_FREQUENCY  80000000
//#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```
For LVGL, use version 8.4.0.

florinbaciuu, 2024-09-27:
> On Arduino IDE 2.3.3, I succeeded to compile and run on a ESP32S3+ST7789 (https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-2.8) the demo provided for this boad with the latest versions of espressif [...]

indeed these 2 lines :
```
#define ESP32
#define CONFIG_IDF_TARGET_ESP32S3
```
correctly select SPI. Unlike you, I put them in platformio.ini. 
```
-DESP32
-DESP32S3
-DCONFIG_IDF_TARGET_ESP32S
```
probably that's why I had no problem with this library. it compiles and runs without problems for me on any esp32 (s2, S3)

happychickencxk, 2024-10-20:
thank u very much!
I downgraded my board core version to **2.0.14**,and find it useless for me.
but **#define USE_HSPI_PORT** (at the same time)
in **User_Setup.h** save my life!
呦西,here we goooooooooo!

cyberreefguru, 2024-11-07:
I just spend about 6 hours bashing my head into the wall until I found this thread.  I'm using a Waveshare ESP32-S3 Zero, which has a  ESP32-S3FH4R2 on board.  I added:
>  -DESP32
>  -DESP32S3
>  -DUSE_HSPI_PORT
>  -DCONFIG_IDF_TARGET_ESP32S

to my platform.ini and everything started working. Prior to that, got an access fault:

```
19:35:06.980 > bootedGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
19:35:06.981 > 
19:35:06.981 > Core  1 register dump:
19:35:06.981 > PC      : 0x420426ce  PS      : 0x00060c30  A0      : 0x820427d4  A1      : 0x3fcebdd0  
19:35:06.992 > A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000004  
19:35:06.997 > A6      : 0x000000ff  A7      : 0x3fc99e20  A8      : 0x08000000  A9      : 0x3fcebda0  
19:35:07.003 > A10     : 0x3fc9b2dc  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000004  
19:35:07.014 > A14     : 0x00000000  A15     : 0x3fc94670  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
19:35:07.019 > EXCVADDR: 0x00000010  LBEG    : 0x4204691c  LEND    : 0x42046980  LCOUNT  : 0x00000003  
19:35:07.030 > 
19:35:07.030 > 
19:35:07.030 > Backtrace: 0x420426cb:0x3fcebdd0 0x420427d1:0x3fcebe00 0x420427dc:0x3fcebe20 0x4200ccca:0x3fcebe40 0x42002b3f:0x3fcebe80 0x42030b9a:0x3fcebef0
19:35:07.149 >   #0  0x420426cb in TFT_eSPI::begin_tft_write() at .pio/libdeps/seeed_xiao_esp32s3/TFT_eSPI/TFT_eSPI.cpp:81
19:35:07.149 >       (inlined by) TFT_eSPI::writecommand(unsigned char) at .pio/libdeps/seeed_xiao_esp32s3/TFT_eSPI/TFT_eSPI.cpp:989
19:35:07.149 >   #1  0x420427d1 in TFT_eSPI::init(unsigned char) at .pio/libdeps/seeed_xiao_esp32s3/TFT_eSPI/TFT_eSPI.cpp:699
19:35:07.149 >   #2  0x420427dc in TFT_eSPI::begin(unsigned char) at .pio/libdeps/seeed_xiao_esp32s3/TFT_eSPI/TFT_eSPI.cpp:603
19:35:07.149 >   #3  0x4200ccca in lv_tft_espi_create at .pio/libdeps/seeed_xiao_esp32s3/lvgl/src/drivers/display/tft_espi/lv_tft_espi.cpp:63 (discriminator 2)
19:35:07.149 >   #4  0x42002b3f in setup() at src/main.cpp:424
19:35:07.149 >   #5  0x42030b9a in loopTask(void*) at /Users/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

Hope this helps others.

-Tom

SinglWolf, 2024-11-25:
I have studied the definition of the `SPI` host number on the `ESP32-S3` chip.
The header file `TFT_eSPI_ESP32_S3.h` connected only when the `ESP32-S3` chip is selected. This is correct and logical.
[image]

Opening the file `TFT_eSPI_ESP32_S3.h` and insert `#define USE_HSPI_PORT` for the study.
[image]
Port number 3 has been activated. The comment states that this is for the `S2` chip.

Insert  `#define USE_FSPI_PORT` for the study.
[image]
Port number 2 has been activated. The comment again states that this is for the `S2` chip.

Now let's see how the default host is determined.
[image]
[image]
The default host has a zero number.
As far as I know, this host is used to control the internal `FLASH`, which causes the chip to freeze.
I hope that the author will pay attention to this.

razarahil, 2024-12-20:
Any update on this? None of the said changes works with my 2.8inch LCD Display Module with Touch panel, Embedded with ST7789T3 Display Driver and CST328 Capacitive Touch Controller.
I'm using the latest tft and esp version.

LoNE-W0LvES, 2024-12-26:
fixed i guess "\Documents\Arduino\libraries\TFT_eSPI\User_Setup_Select.h"

in here comment `#include <User_Setup.h>` this

and uncomment this line `#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>`

casascius, 2025-01-12:
Trying USE_HSPI_PORT wasn't working for me on this one.  I ended up getting this to work on Lilygo T-Deck Plus (ESP32S3) with ESP32 Arduino Core 3 by editing something in the TFT_eSPI library.

What I did: Changed TFT_eSPI_ESP32_S3.h so SPI_PORT was set to 2 instead of FSPI (only when Arduino core 3 was in use)... this did the trick and preserved compatibility with Arduino core 2.  Best I can tell, FSPI under these conditions was defined as 0.  

around line 85:

```
  #elif CONFIG_IDF_TARGET_ESP32S3
    // Hack for unaddressed TFT_eSPI compatibility issue for ESP32 Arduino core 3
    #if ESP_ARDUINO_VERSION_MAJOR >= 3
      #define SPI_PORT 2
    #else
      #define SPI_PORT FSPI	
   #endif
  #endif
```

linghuye, 2025-04-16:
I met this bug too, so I studied it for a while, and found that bug is due to the macro REG_SPI_BASE.
which is not defined before arduino v2.0.16 or platformio 6.7.0, 
and just in this version, it is defined as 
```
#define REG_SPI_BASE(i)     (((i)==2) ? (DR_REG_SPI2_BASE) : (0)) 
```
in framework-arduinoespressif32\tools\sdk\esp32c3\include\soc\esp32c3\include\soc\soc.h
and 
```
#define REG_SPI_BASE(i)         (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))    // GPSPI2 and GPSPI3
```
in framework-arduinoespressif32\tools\sdk\esp32c3\include\soc\esp32s3\include\soc\soc.h
</br>
This macro conflicts with the bug fix code in TFT_eSPI's source code TFT_eSPI_ESP32_C3.h/TFT_eSPI_ESP32_S3.h:
```
#ifndef REG_SPI_BASE
#define REG_SPI_BASE(i) DR_REG_SPI2_BASE
#endif
```
and TFT_eSPI_ESP32_S3.h
```
#ifndef REG_SPI_BASE
#define REG_SPI_BASE(i) (((i)>1) ? (DR_REG_SPI3_BASE) : (DR_REG_SPI2_BASE))
#endif
```
then it crashes at boot.
So I fixed this bug by changing TFT_eSPI's source code
1,Delete all SPI_PORT define macro then
`#define SPI_PORT 2` 
in TFT_eSPI_ESP32_C3.h,
```
#ifdef USE_HSPI_PORT
    #define SPI_PORT 3     //HSPI is port 3 on ESP32 S3
#else
    #define SPI_PORT 2
#endif
```
in TFT_eSPI_ESP32_S3.h.
Then TFT espi works fine for me now.
In short, you change the SPI_PORT to fit the new macro to make it work for your board.

Nefojxj, 2025-04-25:
> Lilygo-T-QTPro-S3 ( `USE_FSPI_PORT `)

Hello! May I ask what the User_Config file is for Lilygo-T-QTPro-S3? 
My board can be written, but it couldn't display properly. Thank you so much for your help :)

electronicsguy, 2025-06-14:
Yes the `USE_HSPI_PORT` port define definitely works for ESP32S3! Funny thing is: Do  NOT use:

```#define CONFIG_IDF_TARGET_ESP32```

It will fail if the above #define is present, even if you define `USE_HSPI_PORT`.

I've attached the custom user setup file that works for me with:
Dev board: Waveshare-ESP32-S3-AMOLED-1.91
Display: ST7735 based 1.44" 128x128 (greentab)
Arduino version: 2.3.6
latest tft_espi version

(rename this file to .h)
[Setup_ESP32S3_ST7735_128x128.txt](https://github.com/user-attachments/files/20737801/Setup_ESP32S3_ST7735_128x128.txt)