Bodmer/TFT_eSPI #3332: ESP32 Core  1 panic'ed (StoreProhibited). Exception was unhandled when tft.init() is executed.

taylorjwnc20, 2024-05-21:
The issue that I am encountering is that a "Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled." just prior to executing the tft.init() function. This issue appears to be only graphics bases since I can execute this sketch with only Serial.println("..."); calls.     I can use the Adafruit_GFX.h with the Adafruit_ST7789.h driver to do a test on a GFX application.

During an update something changed and I am at a loss to determine the root cause.   Wiring is correct ( I have checked it many times).

1. IDE - Arduino version 2.3.2   CLI version 0.35.3
2. TFT_eSPI - version 2.5.43
3. Board Version - esp32  version 3.0.0.rc3 (I have regressed it back to ESP version 2.0.16 with no change in functionality, still reboots).
4. PROCESSOR - ESP32-S3
5. TFT Driver - ST7789 (TFT_eSPI sourced)
6. TFT Interface - SPI

See the following attached files for additional information you require:

[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/15383585/User_Setup.zip)

Sketch:
[WIFI_Channel_List.zip](https://github.com/Bodmer/TFT_eSPI/files/15383589/WIFI_Channel_List.zip)

TFT ; ST7789 IC 1.9 Inch ips Full View TFT Display LCD Color Module SPI Serial Port HD 170x320 MSP430 and C51 Programs
I have only one of these displays....  

[image]

DEBUG Output:
[Debug Output.zip](https://github.com/Bodmer/TFT_eSPI/files/15383726/Debug.Output.zip)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the setup 

[image]

Thanks, John Taylor
I can be reached at taylorjw.nc@att.net

taylorjwnc20, 2024-05-21:
Corrected Debug Output in zip file.  (elements dropped during the copy)

Regressed to 2.0.16 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

-----------------------------------
Hello! ESP32-S3-WROOM SPI PINS
-----------------------------------

MOSI: 11
MISO: 13
SCK: 12
SS: 10

-----------------------------------
Start Channel Scanning
-----------------------------------

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42003e9d  PS      : 0x00060830  A0      : 0x82003f6c  A1      : 0x3fcebda0  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x0000000b  
A6      : 0x000000ff  A7      : 0x00000000  A8      : 0x08000000  A9      : 0x3fcebd70  
A10     : 0x3fc996ac  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000040  
A14     : 0x00000000  A15     : 0x3fc940c4  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x420083f0  LEND    : 0x42008453  LCOUNT  : 0x00000000  

Backtrace: 0x42003e9a:0x3fcebda0 0x42003f69:0x3fcebdd0 0x42002b98:0x3fcebdf0 0x420091ca:0x3fcebe20

elgerg, 2024-05-21:
Looks like this is a duplicate of:
https://github.com/Bodmer/TFT_eSPI/issues/3329
?

APF-MRoger, 2024-05-21:
@elgerg Yup, I would tend to agree...

taylorjwnc, 2024-05-21:
Yes, I agree with the assessment....   I did not read far enough....   my oversight!

ats3788, 2024-05-24:
I have the same Issue with ESP32-S3 and the ILI9341 
Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202934a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   112][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
[   121][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[   129][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42002c3f  PS      : 0x00060830  A0      : 0x82002d14  A1      : 0x3fcebbd0  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000023
A6      : 0x000000ff  A7      : 0x00000000  A8      : 0x08000000  A9      : 0x3fcebba0
A10     : 0x3fc9732c  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000040  
A14     : 0x00000000  A15     : 0x3fc93748  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000010  LBEG    : 0x42018dd4  LEND    : 0x42018e38  LCOUNT  : 0x00000002  

Backtrace: 0x42002c3c:0x3fcebbd0 0x42002d11:0x3fcebc00 0x42001ace:0x3fcebc20 0x4200635a:0x3fcebc50

ELF file SHA256: 9ad77e336525fede

Rebooting...
␀�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202934a
SPIWP:0xee

taylorjwnc20, 2024-05-25:
I have since regressed my versions to TFT_eSPI to 2.5.0 an ESP32 to 2.015 and everything is working well.  My projects are working as expected .  This is true for my ESP32-S3 and ESP32-C3 designs.    My ESP32-S3 code test configuration looks now displays the information very nicely.  At this point in time, I am ok with the current versions, but still want to work with the current releases.

[image]

ats3788, 2024-05-25:
with this platformIO 
[env:esp32-s3-devkitc-1]
platform = espressif32 @ 6.5.0
board = esp32-s3-devkitc-1

; change microcontroller
board_build.mcu = esp32s3

framework = arduino
monitor_speed = 115200
monitor_port = COM10
upload_port = COM10
build_flags = -DCORE_DEBUG_LEVEL=5

lib_deps = Wire
           https://github.com/Bodmer/TFT_eSPI.git
           bitbank2/PNGdec@^1.0.2
    
it works now with the ESP32-S3

LiuGuoGY, 2024-05-27:
I have the same problem .

[image]

```c

    end_tft_write();
  } // end of: if just _booted

Serial.printf("!!!!here!!!\n");

  // Toggle RST low to reset
#ifdef TFT_RST
  #if !defined(RP2040_PIO_INTERFACE)
    // Set to output once again in case MISO is used for TFT_RST
    if (TFT_RST >= 0) {
      Serial.printf("!!!!5555here!!!\n");
      pinMode(TFT_RST, OUTPUT);
    }
  #endif
  if (TFT_RST >= 0) {
    Serial.printf("!!!!6666here!!!\n");
    writecommand(0x00); // Put SPI bus in known state for TFT with CS tied low
    Serial.printf("!!!!7777here!!!\n");
    digitalWrite(TFT_RST, HIGH);
    Serial.printf("!!!!8888here!!!\n");
    delay(5);
    digitalWrite(TFT_RST, LOW);
    delay(20);
    digitalWrite(TFT_RST, HIGH);
  }
  else writecommand(TFT_SWRST); // Software reset
#else
  writecommand(TFT_SWRST); // Software reset
#endif
```

log is bellows:

```
lv init over

!!!!here!!!
!!!!5555here!!!
!!!!6666here!!!
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200ddce  RA      : 0x4200ddbe  SP      : 0x3fcae920  GP      : 0x3fc92400
TP      : 0x3fc7a558  T0      : 0x00000003  T1      : 0xffc3ffff  T2      : 0x00000020
S0/FP   : 0x60004000  S1      : 0x3fc9dac4  A0      : 0x00000001  A1      : 0xffffffff
A2      : 0x20000000  A3      : 0xfdffffff  A4      : 0x08000000  A5      : 0x00000010
A6      : 0xffffffff  A7      : 0x04c4b400  S2      : 0x00000000  S3      : 0x3fcae978
S4      : 0x3fcae96c  S5      : 0x3fca453a  S6      : 0x3c14c000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x0196e6aa  T4      : 0x0000000f  T5      : 0x003c0000  T6      : 0x00000003
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010
MHARTID : 0x00000000

Stack memory:
3fcae920: 0x3fca4000 0x3fca4000 0x019bfcc0 0xfc000301 0x00000001 0x3fc9f000 0x3fc9dac4 0x4200e320
3fcae940: 0x00332220 0x3a227668 0x0000000d 0x4204fa0a 0x3fca4000 0x3fc9e000 0x3fc9f000 0x42003b44
3fcae960: 0x00000000 0x00000000 0x00000000 0x41444345 0x39434233 0x43354637 0x00000000 0x00000000
3fcae980: 0x00000000 0x00000000 0x03004000 0xfbcacc51 0x00000000 0x00000000 0x00000000 0x00000000
3fcae9a0: 0x00000000 0x3fca4000 0x4204f000 0x42053c40 0x00000000 0x00000000 0x00000000 0x4038d966
3fcae9c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcae9e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154
3fcaea00: 0x3fcae8d0 0x892c9f6b 0x3fca09bc 0x3fca09bc 0x3fcaea00 0x3fca09b4 0x00000018 0xdb8fc590
3fcaea20: 0x0a6e05b9 0x3fcaea00 0x00000000 0x00000001 0x3fcac9f0 0x706f6f6c 0x6b736154 0x0e8b7900
3fcaea40: 0x00fb7f40 0x00000000 0x3fcae9e0 0x00000001 0x00000002 0x00000000 0x00000000 0x00000000
3fcaea60: 0x3fca62c4 0x3fca632c 0x3fca6394 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fcaea80: 0x3fcafb20 0x420fc1ee 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeaa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeae0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeb00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeb20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaeb40: 0x00000000 0x00000000 0x00000000 0x00000000 0x0e000000 0xbaad5678 0x00000060 0xabba1234
3fcaeb60: 0x00000054 0x00000000 0x3fcaeb64 0x00000000 0x00000000 0x00000000 0x3fcaeb7c 0xffffffff
3fcaeb80: 0x3fcaeb7c 0x3fcaeb7c 0x00000000 0x3fcaeb90 0xffffffff 0x3fcaeb90 0x3fcaeb90 0x00000001
3fcaeba0: 0x00000001 0x00000000 0x5e00ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678 0x00000100
3fcaebc0: 0xabba1234 0x000000f4 0x00000000 0x00000014 0x3fcaf258 0x00000000 0x00000000 0x00000000
3fcaebe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaec00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaec20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaec40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcaec60: 0x00000000 0x00000000 0x00000101 0x00000000 0x00000000 0x0000000a 0x3fcaf220 0x00000000
3fcaec80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcaeccc
3fcaeca0: 0x3fcaee20 0x00000000 0x3fcaf090 0x3fcaf02c 0x3fcaf1bc 0x3fcaf158 0x3fcaf0f4 0xbaad5678
3fcaecc0: 0x00000150 0xabba1234 0x00000144 0x3fcaed20 0x3fcaed20 0x3fcaee10 0x3fcaee04 0x00000000
3fcaece0: 0x3fcaece4 0xffffffff 0x3fcaece4 0x3fcaece4 0x00000000 0x3fcaecf8 0xffffffff 0x3fcaecf8
3fcaed00: 0x3fcaecf8 0x00000000 0x00000014 0x0000000c 0xdd00ffff 0x00000000 0xb33fffff 0x00000000

ELF file SHA256: d4ac4283e18a9e36

E (493) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

platform.ini as bellows:
```
build_flags =
	-Wall
	-Os

	-D FIRMWARE_VER=${common.firmware_version}
    -D LV_LVGL_H_INCLUDE_SIMPLE
	'-D LV_CONF_PATH=${PROJECT_DIR}/lib/lvgl_conf/lv_conf.h'

	; TFT_eSPI
	-D USER_SETUP_LOADED
	-D DISABLE_ALL_LIBRARY_WARNINGS
	-D ST7789_2_DRIVER
	-D TFT_RGB_ORDER=TFT_RGB
	-D TFT_WIDTH=240
	-D TFT_HEIGHT=240
	-D TFT_MOSI=7
	-D TFT_SCLK=6
	-D TFT_CS=10
	-D TFT_DC=3
	-D TFT_RST=4
	-D TFT_BL=1
	-D SPI_FREQUENCY=27000000
	-D SPI_READ_FREQUENCY=20000000
	-D SPI_TOUCH_FREQUENCY=2500000
```

ChrisZang0815, 2024-05-29:
same problem here.
I'm using a Seeduino XIAO ESP32-C3 (but basically the same happens on a XIAO ESP32S3)
I am using version 2.5.43 of TFT_eSPI and am on version 3.0.0 for esp32

Here's my sketch:
#include <TFT_eSPI.h> 
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
 
void setup() 
{
  Serial.begin(115200);
  Serial.println("*** Program: " __FILE__ " \n*** compiled on: " __DATE__ " at: " __TIME__ );

  SPI.begin();   

  tft.begin();
  //tft.setRotation(2);
  //tft.fillScreen(TFT_BLACK);
  //tft.setTextColor(TFT_BLACK);  
}  
 
void loop() 
{
  Serial.println ("loop: "+String(millis()));
  delay (1000);
}

as soon as I uncomment the line tft.begin(); I get:

*** Program: C:\Users\chris\Dropbox\Arduino\Arduino Project Code\CAN\SportChrono FOLDER\XIAO_TFT\XIAO_TFT.ino 
*** compiled on: May 29 2024 at: 08:59:54
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42001bbc  RA      : 0x42001bac  SP      : 0x3fc94740  GP      : 0x3fc8be00  
TP      : 0x3fc85628  T0      : 0x00051616  T1      : 0xffc3ffff  T2      : 0x2a0a206f  
S0/FP   : 0x60004000  S1      : 0x3fc8c860  A0      : 0x00000000  A1      : 0x00002001  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x08000000  A5      : 0x00000010  
A6      : 0x00000002  A7      : 0x00000020  S2      : 0x00000001  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x01927358  T4      : 0x0196e6aa  T5      : 0x00000041  T6      : 0x0000000f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010  
MHARTID : 0x00000000

ChrisZang0815, 2024-05-30:
OK, quick update: 
I first downgraded my versions to TFT_eSPI to 2.5.0 and ESP32 to 2.015 
AFter that it worked on XIAO ESP32S3 but on XIAO ESP32C3 I still have the crash
Then I upgraded TFT_eSPI back to 2.5.43 and it still works on the ESP32S3.
So that seems to indicate that the problem lies with esp32 and not with TFT_eSPI.
However I would think that Bodmer has way more pull with the powers to be that write esp32 than we, so I hope that he can do some "magic" there.

ChrisZang0815, 2024-05-30:
one more update: It also works with esp 2.0.17 (which is the final version of 2.0.x)
meaning they broke it between 2.0.17 and 3.0.0

ChrisZang0815, 2024-06-01:
final (?) update:
"last known good" version for esp32:
* ESP32S3: 2.0.17
* ESP32C3: 2.0.14

SergePD, 2024-06-09:
For me, last working core is: 2.0.14

Sofronio, 2024-06-24:
2.0.14 works for me. Thanks

mulcmu, 2024-06-29:
After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup.  This was with a custom board 
* ST7789 LCD 2inch
* ESP32-S3-WROOM-1U-N16R8 module
* TFT_eSPI 2.5.43
* ESP core version 3.0.2

```
//Configuration for tft_espi user_setup.h
// #define ST7789_DRIVER  
// #define TFT_RGB_ORDER TFT_BGR
// #define TFT_WIDTH  240
// #define TFT_HEIGHT 320 
// #define TFT_MOSI            10
// #define TFT_SCLK            11
// #define TFT_CS              12
// #define TFT_DC              13
// #define TFT_RST             9
// #define TFT_INVERSION_ON

// #define USE_HSPI_PORT
```

I don't know how or why...I'm pretty sure I reverted all the other hacks that were tested in [TFT_eSPI_ESP32_S3.h](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32_S3.h)

ChrisZang0815, 2024-06-29:
I just tried with esp32 3.0.2. Result
ESP32-S3: works, no longer getting panic
ESP32-C3: still the same problem

andremalenfant, 2024-07-31:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]

That fixed it for me! Thanks!

gfranju, 2024-08-01:
It works !!!!!!!
I removed ESP32 then installed version 2.0.14 and version 2.5.0 TFT eSPI.
I added the line :
#define USE_HSPI_PORT
For pin numbers :
#define TFT_CS 10 (I don't have pin 34 on the ESP32S3 weact)
#define TFT_MOSI 35 
#define TFT_SCLK 36 
#define TFT_DC 7 
#define TFT_RST 6 

Thank you very much...
I couldn't have done it without your help.

Yutu0k, 2024-08-07:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]
yep! it confuses me for nearly half of the day trying to solve the issue.

Yutu0k, 2024-08-07:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]

It also works for me! 
The interesting thing is... The same code worked two days ago. After that I upload another firmware to test other codes. But when I switched back to the original code, it doesn't work. I guess the issue has something to do with esp32 spi interface.

Thank you so much!

ChrisZang0815, 2024-08-07:
how does that work when every line is commented out?

mulcmu, 2024-08-08:
> how does that work when every line is commented out?

I copy the board settings from the library headers into the project .ino file in case my configuration gets wiped out in a library update.  I copied the consolidated settings from my .ino that happened to be commented out.  The difference that avoids the panic is `#define USE_HSPI_PORT`, the rest of the values should be configured for the screen being used.

elaw01887, 2024-08-27:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]

Just came to say THANK YOU for this!  I spent hours yesterday fighting with this issue... which is now fixed. Hooray!

zachtos, 2024-09-04:
I also had the problem and the #define USE_HSPI_PORT was the root cause.  Not sure why suddenly it's required?  perhaps a library investigation should happen?

MAKERSUN99, 2024-11-05:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]

It works!

Addition of #define USE_HSPI_PORT in user setup.

ILI9341 LCD 2.4inch
ESP32-S3-WROOM-1U-N16R8 module
TFT_eSPI 2.5.43
ESP core version 3.0.7

Thank You !!!

zalexzperez, 2024-11-16:
> After much trial and error, I was able to avoid the panic reboot from `tft.init()` with addition of `#define USE_HSPI_PORT` in user setup. This was with a custom board
>
> * ST7789 LCD 2inch
> * [...]

Adding DUSE_HSPI_PORT=1 build flag to my platformio.ini avoids the panic reboot in my ESP32-S3.
However, my display (ST7789_DRIVER) doesn't work anymore: it shows garbage:
[image]

Guess I'll just keep using platform = espressif32 @ 6.5.0

joellecam, 2024-11-24:
Hello,

I have the same problem . I put some serial print in the code and I found where the guru panic appears.

I use VScode and Platform IO. The problem don't exist if I use a ESP32 wroom but it appears with ESP32-S3 type

the problem is in the function TFT_eSPI::begin_tft_write(void)  at the line SET_BUS_WRITE_MODE; 

my SPI pinout
```
#define SCK               12 
#define MISO            13 
#define MOSI            11
#define TFT_SS          38 
```

```
inline void TFT_eSPI::begin_tft_write(void){
  if (locked) {
    locked = false; // Flag to show SPI access now unlocked
#if defined (SPI_HAS_TRANSACTION) && defined (SUPPORT_TRANSACTIONS) && !defined(TFT_PARALLEL_8_BIT) && !defined(RP2040_PIO_INTERFACE)
    Serial.println ( "spi.beginTransaction; start ;");
    spi.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));
    Serial.println ( "spi.beginTransaction; end ;");
#endif
    CS_L;
    Serial.println ( "CS_L; end ;");
    SET_BUS_WRITE_MODE;  // Some processors (e.g. ESP32) allow recycling the tx buffer when rx is not used
    Serial.println ( "SET_BUS_WRITE_MODE; end ;");
  }
}
```

Log in serial monitor

```
tft.init()
writecommand(0x00) start ;
begin_tft_write() start ;
spi.beginTransaction; start ;
spi.beginTransaction; end ;
CS_L; end ;
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
```

Have you some news since September?

Regards, 
Joël  French

tkggithub, 2024-12-24:
> Hello,
>
> I have the same problem . I put some serial print in the code and I found where the guru panic appears.
>
> I use VScode and Platform IO. The problem don't exist if I use a ESP32 wroom but it [...]

I have recently received a ESP32C3 super mini board and using it with ESP core version 3.0.5 in Arduino IDE 2.3.4.
I have faced several issues with the USB serial mostly not working than working but at one particular time when the planets aligned I was able to debug the TFT Clock Digital example and this example generated a fault on SET_BUS_WRITE_MODE.
Upon inspection it looks like the **_spi_user** variable used by the SET_BUS_WRITE_MODE macro is not properly set.
In this 3.0.5 core version for ESP32C3 the REG_SPI_BASE define is found in the core headers and it is declared as:
`#define REG_SPI_BASE(i)                         (((i)==2) ? (DR_REG_SPI2_BASE) : (0))   // only one GPSPI`
as opposed to the ESP v2 Arduino cores where it is not defined in the actual core and a custom define is inside the TFT_eSPI library:
```
// Fix IDF problems with ESP32C3
#if CONFIG_IDF_TARGET_ESP32C3
  // Fix ESP32C3 IDF bug for missing definition (VSPI/FSPI only tested at the moment)
  #ifndef REG_SPI_BASE
    #define REG_SPI_BASE(i) DR_REG_SPI2_BASE
  #endif

  // Fix ESP32C3 IDF bug for name change
  #ifndef SPI_MOSI_DLEN_REG
    #define SPI_MOSI_DLEN_REG(x) SPI_MS_DLEN_REG(x)
  #endif
#endif
```
The  plot thickens as REG_SPI_BASE at least for ESP32C3 is used in the library based on the SPI_PORT define which uses SPI2_HOST:
`#define SPI_PORT SPI2_HOST`
You probably almost guessed it that SPI2_HOST has a value of 1 instead of the expected value of 2 in the core REG_SPI_BASE define so the actual address of the  _spi_user variable gets rebased using 0x0 as a base address instead of the expected 0x60024000 address for SPI2.
So this is why the fault happens as the write goes to address 0x10 instead of 0x60024000+0x10.
That might seem like an easy fix but it is a bit more complex to do it properly.
**As a quick test**, I have changed the REG_SPI_BASE definition from using (i)==2 to (i)==1 and that kind of works but in a way for possibly more wrong reasons. 
The esp32c3 TRM states:
"_27 SPI Controller (SPI)
27.1 Overview
The Serial Peripheral Interface (SPI) is a synchronous serial interface useful for communication with external
peripherals. The ESP32-C3 chip integrates three SPI controllers:
• SPI0,
• SPI1,
• General Purpose SPI2 (GP-SPI2).
SPI0 and SPI1 controllers are primarily reserved for internal use. This chapter mainly focuses on the GP-SPI2
controller._" 
So basically the user should use the GP-SPI2 interface.
Well, in the same core headers for ESP32C3 we have:
```
/**
 * @brief Enum with the three SPI peripherals that are software-accessible in it
 */
typedef enum {
//SPI1 can be used as GPSPI only on ESP32
    SPI1_HOST=0,    ///< SPI1
    SPI2_HOST=1,    ///< SPI2
#if SOC_SPI_PERIPH_NUM > 2
    SPI3_HOST=2,    ///< SPI3
#endif
    SPI_HOST_MAX,   ///< invalid host value
} spi_host_device_t;

```
together with:
```
/*-------------------------- SPI CAPS ----------------------------------------*/
#define SOC_SPI_PERIPH_NUM          2
```
makes it impossible to get to SPI3_HOST that would have a value of 2 which would be what the REG_SPI_BASE from the core expects...
So what would be a proper fix for ESP32C3?

The SPI peripherals bases in the ESP32C3 core according to the TRM are mapped to:
'
SPI Controller 1 0x6000_2000
SPI Controller 0 0x6000_3000
SPI Controller 2 0x6002_4000 
'
and this is kind of mirrored to the core defines:
```
#define DR_REG_SPI1_BASE                        0x60002000
#define DR_REG_SPI0_BASE                        0x60003000
#define DR_REG_SPI2_BASE                        0x60024000
```

Should the SPI interface used be SPI3_HOST which would have an id of 2  to match the REG_SPI_BASE expectations (this would also require the SOC_SPI_PERIPH_NUM changed) or SPI2_HOST with an id of 1 and change the REG_SPI_BASE macro in the core headers.
Or create an even more "interesting" fix of undefining the core define and redefining it. This might work but if some complied code in the core libraries already used the wrong define then it creates a potential situation where the original define was already used.
Thoughts?
On a related note there are several links where the confusion about which SPI is which goes further:
https://esp32.com/viewtopic.php?t=22717
"The ESP32C3 only has SPI0 (unusable by software as it is controlled by the flash cache) SPI1 (which is kinda hard to use as it's also connected to the flash) and SPI2; the hardware does not contain SPI3."
and also related to the SOC_SPI_PERIPH_NUM
https://github.com/espressif/esp-idf/issues/10349

Personally I would say that indeed there is no SPI3 interface.  If we start counting at SPI0, we have SPI0, SPI1 and SPI2.
But then having SPI1_HOST=0, and SPI2_HOST=1 and no SPI0_HOST might have been what generated the confusion of using the i==2 in the REG_SPI_BASE define expecting that a value of 2 might correspond to interface SPI2. Maybe a better test would have been (i)==SPI2_HOST?
What about if the user gives a value other than SPI2_HOST?
In this case the macro will rebase using 0x0 which would end up in the same fault case as we are now.
Would it be better to use the define that has no check on the interface and always assume the interface is SPI2_HOST?

Thoughts?

IvarDahl, 2025-03-02:
> Having a similar issue on a completely different hardware setup (Waveshare round LCD with integrated ESP32S3). Using platformio, this happened once the platform 6.7.0 was updated for espressif32. For now to workaround I am locking the platform version to 6.6.0 in my platformio.ini using "platform = espressif32@6.6.0"

Confirming that this worked for me, running TFT ILI9341 on ESP32-C3.
Thanks for the solution.

dcvxvf, 2025-04-27:
I using vscode and platformio, How can I add "USE_HSPI_PORT" to my .ini ?

laziz24, 2025-07-20:
just uncomment // #define USE_HSPI_PORT in User_setup.h (remove //)
 and you will get it work

niu541412, 2025-08-01:
For esp32 c3 user, see my reply https://github.com/Bodmer/TFT_eSPI/issues/3743#issuecomment-3145633679 .

XOasisX, 2025-10-21:
Muchas gracias, llevo dos días solucionando eso, porfavor verifiquen que la linea #define USE_HSPI_PORT este habilitada en el User_Setup.h de la libreira, Dios mio que error tan molesto.

sophatvathana, 2025-10-23:
> Confirming that this worked for me, running TFT ILI9341 on ESP32-C3. Thanks for the solution.

Confirming that this worked for me, running TFT ILI9341 on ESP32-WROOM-32U.