Bodmer/TFT_eSPI #3338: Issue with 240*320 display

Rolletten, 2024-05-23:
Hello, 
I wanted to test out a new screen i just got and I'm having issues, the image looks like it's rotated 90degrees and there is random noise on the side, happens with all example scetches from the library.
Arduino ide
library: V 2.5.43
board package: esp32 by espressif V 2.0.16
ESP-WROOM-32
ST7789 - SPI
2'' 240*320

[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/15421311/User_Setup.zip)
[image]

nicholasbinder, 2024-06-29:
Did you set the Display Size in the User Setup.h?

Loetnase, 2024-08-06:
Did you set the Rotation 1?
tft.setRotation( 1 ); /* Landscape orientation */

lbernstone, 2024-08-28:
You may need to switch the width and height settings (ie 240x320 instead of 320x240).