Bodmer/TFT_eSPI #3346: compile error in arduino ide

MartinNohr, 2024-05-31:
I just loaded the latest arduino ide and I got this error:
In file included from C:\Users\marti\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:
C:\Users\marti\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'void TFT_eSPI::begin_SDA_Read()':
C:\Users\marti\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:72:3: error: 'gpio_set_direction' was not declared in this scope
   72 |   gpio_set_direction((gpio_num_t)TFT_MOSI, GPIO_MODE_INPUT);

It compiled last time I tried it. Any ideas anybody? Thanks.

MartinNohr, 2024-05-31:
I figured it out. Espressif released a new version 3.0 of their board manager. I switched back to 2.0.17 and it compiles again.

StuartsProjects, 2024-06-01:
> I figured it out. Espressif released a new version 3.0 of their board manager. I switched back to 2.0.17 and it compiles again.

I noticed that too, when I updated to the latest 3.0 ESP32 core.  

However for a working example that I had (writtten using the 2.0.11 core) displaying on TFT an JPG image in memory on an ESP32S3, I had to go back to version 2.0.14 of the ESP32 core to get it to work, the program would just crash on startup otherwise.

JiriBilek, 2024-06-01:
I met this problem today. My quick solution (until there would be a new version of the library) for ESP32 core was to add
```
#include "driver/gpio.h"
#include <rom/ets_sys.h>
```
to `Processors\TFT_eSPI_ESP32.h`

agusmal, 2024-06-03:
> I figured it out. Espressif released a new version 3.0 of their board manager. I switched back to 2.0.17 and it compiles again.

Hi,, did you success? 
I got this error=
exit status 1
Compilation error: exit status 1

JiriBilek, 2024-06-03:
Yes, I it works for me. Recently, I changed it a little but it shouldn't make any difference.

```
// For core 3.0.0 and higher
#if ESP_ARDUINO_VERSION_MAJOR >= 3
	#include "driver/gpio.h"
	#include <rom/ets_sys.h>
#endif
```

The `ets_sys.h` is included because of the delayus() function I am using with my touchpad. You won't need it.
I am using the latest version of this library (2.5.43).

stevempotter, 2024-06-13:
Thanks, @JiriBilek , you fixed the problem!
```
// For core 3.0.0 and higher
#if ESP_ARDUINO_VERSION_MAJOR >= 3
	#include "driver/gpio.h"
	#include <rom/ets_sys.h>
#endif
```

104TMR, 2024-06-20:
I've also just updated Arduino IDE 2.3.2 with ESP32 Core 3.x, and now find that compiling and running even the simple example tft_espi "Graph_demo_1" program on an EP32-S3 goes into an endless runtime "Guru Meditation Error" when the sketch hits the tft.begin() line.
I've tried inserting 
        #include "driver/gpio.h"
	#include <rom/ets_sys.h>
into the /Processors/TFT_eSPI_ESP32_S3.h file, but it doesn't seem to help, and the runtime exception still occurs.
Has anyone any further thoughts on what other remediations/workarounds may be possible for this problem?

104TMR, 2024-06-20:
Actually, I've been able to get the Arduino debugger working with the ESP32-S3 JTAG/USB port, and have put a break point at the tft.begin() line.
Stepping into it, it seems to show that it might be getting tripped up at line 81 ("SET_BUS_WRITE_MODE;") in the file TFT_eSPI.cpp.
Does anyone have an idea if this command might be conflicting with anything in the ESP32 Core 3.x upgrade?
Any thoughts appreciated.

stevempotter, 2024-06-20:
Yes, something Espressif did broke TFT_eSPI. It is about bidirectional SPI lines.
After lots of looking at a black display on my LilyGO T-Display,
 I found here in one of the Issues a workaround:
Edit Processors/TFT_eSPI_ESP32.h to include
#include "driver/gpio.h"

JiriBilek, 2024-06-20:
Just an idea: `SET_BUS_WRITE_MODE` is defined as `*_spi_user = SPI_USR_MOSI | SPI_CK_OUT_EDGE`.
And _spi_user is defined as a volatile variable for esp32 https://github.com/Bodmer/TFT_eSPI/blob/fae22f785f8b4f7970195487826a348d08d082ae/Processors/TFT_eSPI_ESP32.c#L56 and as define for esp32-s3 https://github.com/Bodmer/TFT_eSPI/blob/fae22f785f8b4f7970195487826a348d08d082ae/Processors/TFT_eSPI_ESP32_S3.h#L117 Honestly, I don't understand, why. 
I'd try to copy the code from esp32 and see what happens.

104TMR, 2024-06-20:
Hi Steve,
Yes, I’m rolling back to ESP32 Core 2.x.
I can’t get tft_espi to work with 3.x.

The #include "driver/gpio.h" didn’t work for me.

104TMR, 2024-06-21:
Hmmm...I think I'm now in over my head.
Rolled back ESP32 Core 3.x to 2.x, and thought everything should compile and run the way it used to.
Wan hope...:-(
My code is going into the same Guru Meditation runtime loop that it did under Core 3.x.
I've managed to find/use the ESP Exception Decoder for Arduino IDE 2.x (by dankeboy36), and it reports on the following runtime exceptions:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x42013153  PS      : 0x00060a30  A0      : 0x82013314  A1      : 0x3fcebe90  
A2      : 0x3fc97280  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000010  
A6      : 0x3fc9a990  A7      : 0xff000000  A8      : 0x08000000  A9      : 0x3fcebe60  
A10     : 0x3fc9a990  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000040  
A14     : 0x00000000  A15     : 0x3fc927f0  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x42019474  LEND    : 0x420194d7  LCOUNT  : 0x00000000  

Backtrace: 0x42013150:0x3fcebe90 0x42013311:0x3fcebec0 0x42013570:0x3fcebef0 0x42004e8e:0x3fcebf10 0x4201a3ae:0x3fcebfd0

<Decoded results:>
PC: 0x42013153:  is in TFT_eSPI::writecommand(unsigned char) (c:\Users\<Username>\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
EXCVADDR: 0x00000010

Decoding stack results
0x42013150:  is in TFT_eSPI::writecommand(unsigned char) (c:\Users\<UserName>\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
0x42013311:  is in TFT_eSPI::init(unsigned char) (c:\Users\<UserName>\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:692).
0x42013570:  is in TFT_eSPI::begin(unsigned char) (c:\Users\<UserName>\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:603).
0x42004e8e: setup() at D:\Transfer\Data\Users\House\TemperatureLogger\ESP32S3_6Chan_TCLogger_WebServer\ESP32S3_6Chan_TCLogger_WebServer.ino:397
0x4201a3ae: loopTask(void*) at C:\Users\<UserName>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\cores\esp32\main.cpp:42

I've looked into each of these lines in the respective files, but can't make head-nor-tail of what might be going on the commands/functions to cause the exception.

If any of this makes any sense to anyone, I'd really appreciate some enlightening interpretations.
PS.  I tried your suggestion, JiriBilek, of inserting the line fromESP32.c  into the ESP32_S3.h, but when I compiled, it threw back a bunch of errors, all saying that a variable was being defined in multiple places.

JiriBilek, 2024-06-21:
> I tried your suggestion, JiriBilek, of inserting the line fromESP32.c into the ESP32_S3.h, but when I compiled, it threw back a bunch of errors, all saying that a variable was being defined in multiple places.

Suppose you copied all 4 lines into the .c file and deleted the 4 defines from .h?

104TMR, 2024-06-21:
No, I just copied the 'volatile' from ESP32.c, commented out the '#define' in ESP32_S3.h, and pasted the 'volatile' line below it.
Also, I found that while "volatile uint32_t* _spi_user      = (volatile uint32_t*)(SPI_USER_REG(SPI_PORT));" resides at line 56 in my ESP32.c, the "#define _spi_user      (volatile uint32_t*)(SPI_USER_REG(SPI_PORT))" line occurs at line 123 (not 117) in my ESP32_S3.h file.  Seems like some differences between our ESP32_S3.h files(?).
Also, your original post suggested copying from the ESP32(.c?) file, whereas your last implies copying into the .c file.
Am I missing something?

104TMR, 2024-06-22:
Hi JiriBilek,
I've decided not to pursue altering the .h and .c files, as I have (i) withdrawn from trying to get ESP32 board package 3.x to work, and (ii) I have found that the problems with tft_espi are resolved if I roll back to core 2.0.14.  This is discussed at https://github.com/Bodmer/TFT_eSPI/issues/3284.

4aryash, 2024-06-22:
> I figured it out. Espressif released a new version 3.0 of their board manager. I switched back to 2.0.17 and it compiles again.

Switching back is probably the easiest solution. Thanks for this!

JiriBilek, 2024-07-01:
I had the same problem with C3. Fixed here: https://github.com/Bodmer/TFT_eSPI/issues/3384

mrmonteith, 2024-09-16:
> I met this problem today. My quick solution (until there would be a new version of the library) for ESP32 core was to add
> 
> ```
> #include "driver/gpio.h"
> #include <rom/ets_sys.h>
> ```
> 
> to `Processors\TFT_eSPI_ESP32.h`

How do you go about rolling it back to a previous version.   Under Board or Libraries?  What to look for?