Bodmer/TFT_eSPI #3351: Crashes on esp32c3 on tft.init() while using library in esp-idf with arduino component

NaitikSanas, 2024-06-04:
I am using the latest version of this library with arduino core that is from idf-release/v4.4 branch. the esp-idf is 4.4.7

I have followed the instructions from the https://github.com/Bodmer/TFT_eSPI/pull/1770#issuecomment-1096478997 but i can't understand what is causing crash here.

I also tried the esp-idf 5.1 and arduino core from  idf-release/v5.1 branch and i still received same errors. 

it would be great if anyone can help me figure out what is causing this behaviour. 
thanks. 

Here is attached log : 
```
I (42) boot: ESP-IDF v4.4.7 2nd stage bootloader
I (42) boot: compile time 16:49:18
I (42) boot: chip revision: v0.4
I (44) boot.esp32c3: SPI Speed      : 80MHz
I (49) boot.esp32c3: SPI Mode       : DIO
I (54) boot.esp32c3: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0f088h ( 61576) map
I (120) esp_image: segment 1: paddr=0001f0b0 vaddr=3fc8c600 size=00dcch (  3532) load
I (121) esp_image: segment 2: paddr=0001fe84 vaddr=40380000 size=00194h (   404) load
I (127) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2883ch (165948) map
I (162) esp_image: segment 4: paddr=00048864 vaddr=40380194 size=0c304h ( 49924) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (186) cpu_start: Unicore app
I (187) cpu_start: Pro cpu up.
I (195) cpu_start: Pro cpu start user code
I (195) cpu_start: cpu freq: 160000000
I (195) cpu_start: Application information:
I (198) cpu_start: Project name:     eTFT_ESP_IDF
I (203) cpu_start: App version:      1
I (208) cpu_start: Compile time:     Jun  4 2024 16:49:05
I (214) cpu_start: ELF file SHA256:  54ecfb76e76cd222...
I (220) cpu_start: ESP-IDF:          v4.4.7
I (225) cpu_start: Min chip rev:     v0.3
I (229) cpu_start: Max chip rev:     v1.99 
I (234) cpu_start: Chip rev:         v0.4
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC8E9A0 len 0004DD70 (311 KiB): DRAM
I (252) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (259) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (266) spi_flash: detected chip: winbond
I (270) spi_flash: flash io: dio
I (275) sleep: Configure to isolate all GPIO pins in sleep state
I (281) sleep: Enable automatic switching of GPIO sleep configuration
I (288) cpu_start: Starting scheduler.
I (300) gpio: GPIO[2]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (305) gpio: GPIO[7]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (314) gpio: GPIO[6]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (323) gpio: GPIO[2]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (332) gpio: GPIO[7]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (342) gpio: GPIO[6]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200c52e  RA      : 0x4200c518  SP      : 0x3fc977b0  GP      : 0x3fc8ce00
0x4200c52e: TFT_eSPI::begin_tft_write() at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/TFT_eSPI-2.5.34/TFT_eSPI.cpp:81
 (inlined by) TFT_eSPI::writecommand(unsigned char) at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/TFT_eSPI-2.5.34/TFT_eSPI.cpp:982

0x4200c518: TFT_eSPI::begin_tft_write() at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/TFT_eSPI-2.5.34/TFT_eSPI.cpp:80
 (inlined by) TFT_eSPI::writecommand(unsigned char) at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/TFT_eSPI-2.5.34/TFT_eSPI.cpp:982

TP      : 0x3fc8905c  T0      : 0x00000000  T1      : 0x01927358  T2      : 0x00000000
S0/FP   : 0x3fc8e16c  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x00002001
A2      : 0x00000003  A3      : 0x00000001  A4      : 0x08000000  A5      : 0x00000010
A6      : 0x0196e6aa  A7      : 0x00002001  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000010
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fc977b0: 0x00000003 0x00000000 0x019bfcc0 0x00000301 0x00000000 0x00000000 0x3fc8e16c 0x4200c818
0x4200c818: TFT_eSPI::init(unsigned char) at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/TFT_eSPI-2.5.34/TFT_eSPI.cpp:693

3fc977d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x42006bb2
0x42006bb2: setup() at C:/Users/Naitik/Documents/eTFT_ESP_IDF/main/eTFT_ESP_IDF.cpp:33

3fc977f0: 0x00000000 0x00000000 0x00000000 0x42007d50 0x00000000 0x00000000 0x00000000 0x40388222
0x42007d50: loopTask(void*) at C:/Users/Naitik/Documents/eTFT_ESP_IDF/components/arduino/cores/esp32/main.cpp:42

0x40388222: prvTaskExitError at C:/Users/Naitik/esp/v4.4.7/esp-idf/components/freertos/port/riscv/port.c:140

3fc97810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97830: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc97150 0x3fc8e9b4 0x3fc8dc1c
3fc97850: 0x3fc8dc1c 0x3fc97844 0x3fc8dc14 0x00000018 0xa5a5a5a5 0xa5a5a5a5 0x3fc97844 0x00000000
3fc97870: 0x00000001 0x3fc92a20 0x706f6f6c 0x6b736154 0xa5a5a500 0x00a5a5a5 0x00000000 0x3fc97830
3fc97890: 0x00000001 0x00000001 0x00000000 0x00000000 0x00000000 0x3fc8f368 0x3fc8f3d0 0x3fc8f438
3fc978b0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4201ba70 0x00000000
0x4201ba70: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fc978d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc978f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97910: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97990: 0x00000000 0x3f000000 0x00000054 0x00000000 0x3fc9799c 0x00000000 0x00000000 0x00000000
3fc979b0: 0x3fc979b4 0xffffffff 0x3fc979b4 0x3fc979b4 0x00000000 0x3fc979c8 0xffffffff 0x3fc979c8
3fc979d0: 0x3fc979c8 0x00000001 0x00000001 0x00000000 0xa500ffff 0x00000000 0xb33fffff 0x00000000
3fc979f0: 0x00000054 0x00000000 0x3fc979f4 0x00000000 0x00000000 0x00000000 0x3fc97a0c 0xffffffff
3fc97a10: 0x3fc97a0c 0x3fc97a0c 0x00000000 0x3fc97a20 0xffffffff 0x3fc97a20 0x3fc97a20 0x00000001
3fc97a30: 0x00000001 0x00000000 0xa500ffff 0x00000000 0xb33fffff 0x00000000 0x00044cc1 0x3fc8e9b4
3fc97a50: 0x3fc8e9b4 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97a70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97a90: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97ab0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97ad0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97af0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97b10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97b30: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97b50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97b70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc97b90: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 54ecfb76e76cd222

Rebooting...
```

JimDrewGH, 2024-06-28:
I have the same issue.  This was reported awhile ago.  Only core v2.0.14 and earlier work.  v2.0.15 of the Arduino core results in this problem.  (https://github.com/Bodmer/TFT_eSPI/issues/3284).   I would love to move to the latest 3.x but I need the TFT support.  There is also another thread on this issue:  https://github.com/espressif/arduino-esp32/issues/9618

@Bodmer is there a hack that we can make to fix this temporarily until a real fix is made?

digibake, 2024-06-28:
I think this might be the same issue that I'm facing now using platformio.
My project worked fine using `platform = espressif32@6.6.0` however yesterday I updated to the latest version 6.7.0 and now my project crashes when calling `tft.init()`

Here's a link to what's changed in the platformio/platform-espressif32 repository just in case it helps identify what's changed:

https://github.com/platformio/platform-espressif32/releases

JimDrewGH, 2024-06-28:
Yes, it is known that moving from v6.60->v6.70 under PlatformIO causes the same issue as moving from v2.0.14 to any higher version under the Arduino core.

It would be great if there was something we could do to at least temporarily fix this issue as it affects ALL of the ESP32 family *except* the original ESP32 - so ESP32-S2, S3, C3, H6, etc.

The-Kinky-Coder, 2024-07-20:
Quick Fix: Add `#define USE_HSPI_PORT` to your setup file
(for those using platformio, like myself, add `-D USE_HSPI_PORT to your build flags either in platformio.ini or the board json.)

Not dug into the issue too much, but I'm guessing (more stabbing in the dark actually because I really not too deep into this at all) that somewhere in the transition from 2.0.14 to anything higher (6.6.0 to 6.7.0 for use platformio peps) something they changed something in SPI due to PSRAM issues which is messing with the SPI driver in TFT_eSPI.

@Bodmer did add the ability to use the second HSPI peripheral instead of the default VSPI peripheral many moons ago here -> #286, so I defined `USE_HSPI_PORT` and the issue went away.

i-am-shodan, 2024-07-30:
Also seeing this issue

gpena208777, 2024-09-02:
Still seeing this major issue.

jonhp, 2024-09-26:
I am stuck too. I have esp32 by Espressif v2.0.15 in the board manager. I am still getting an exception. Now simply dead-in-the-water. Need tft-espi to work for this project. 

`ESP Exception Decoder
Sketch: HVAC-HRV-Conditions FQBN: esp32:esp32:esp32s3

Backtrace: 0x42004ac9:0x3fcebbc0 0x42004b99:0x3fcebbf0 0x42003bb7:0x3fcebc10 0x42008a76:0x3fcebc50

Decoding stack results
0x42004ac9:  is in TFT_eSPI::writecommand(unsigned char) (z:\Source\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:81).
0x42004b99:  is in TFT_eSPI::init(unsigned char) (z:\Source\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:692).
0x42003bb7: setup() at Z:\Source\Arduino\HVAC-HRV-Conditions\HVAC-HRV-Conditions.ino:178
0x42008a76: loopTask(void*) at C:\Users\jon\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\cores\esp32\main.cpp:42
`

i-am-shodan, 2024-09-26:
I found LovyanGFX works for me