Bodmer/TFT_eSPI #3353: ESP32 Dev Module 'TFT_WIDTH' was not declared in this scope; did you mean 'TFT_WHITE'?

IOM2021, 2024-06-05:
First thanks for all the work you do, much appreciated.

I am using a ESP32 dev module with ILI9468 with FT6236 Capactive Touch screen

I have upgraded my library to 2.5.43 and I am suddenly getting an error message.

I am using the FT6236.h library for touch

Arduino IDE version 2,3,0
ESP32 version 3.0.0
ESP32 Wroom Dev Module
SPI interface

The issue only occured once I upgraded the library, before that I have had no issues.

I am confused because if I run Free_Font_Demo.ino it runs OK.
If I run my test sketch included below it runs OK

For my main project sketch it wouldn't compile at first time because I got an error with the line tft.loadFont(Calibri30), (a custom font I had made and used previously) as it said loadFont didn't exist. I assumed this was a library change so modded all those lines to: tft.setFreeFont(FSS24); as I couldn't work out how to get my font to work.

Once I had made that change I then got the TFT_WIDTH error. I can't understand how this could suddenly be not declared

User_setup.h included below is the same for all sketches.

I have been searching online for hours and got no closer so thought I shouold contact you directly.

Simon

Error Message reads
--------------------------------------------------------------------------------------------------------------
In file included from K:\MyArduinoSketches\NOWRail\nowRailv0_7handControllerV1.1\nowRailv0_7handControllerV1.1.ino:10:
k:\MyArduinoSketches\libraries\TFT_eSPI/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  973 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
k:\MyArduinoSketches\libraries\TFT_eSPI/TFT_eSPI.h:432:25: error: 'TFT_WIDTH' was not declared in this scope; did you mean 'TFT_WHITE'?
  432 |   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
      |                         ^~~~~~~~~
      |                         TFT_WHITE
k:\MyArduinoSketches\libraries\TFT_eSPI/TFT_eSPI.h:432:49: error: 'TFT_HEIGHT' was not declared in this scope
  432 |   TFT_eSPI(int16_t _W = TFT_WIDTH, int16_t _H = TFT_HEIGHT);
      |                                                 ^~~~~~~~~~

exit status 1

Compilation error: exit status 1

---------------------------------------------------------------------------------------------------------

----------------------------------test sketch------------------------------------------------
```
#include "TFT_eSPI.h"
#include "FT6236.h"
TFT_eSPI tft = TFT_eSPI();  // Invoke custom library with default width and height
FT6236 ts = FT6236();

void setup() {
  Serial.begin(115200);
  Serial.println("9468FT6236ESP32V1");
  tft.begin();
  tft.init();
  tft.setRotation(2);

  tft.fillScreen(TFT_RED);
  tft.drawRect(0,0,319,476,TFT_WHITE);//screen seems to only display 476 wide, rest is hidden under black surround
  tft.drawRect(10,10,10,10,TFT_WHITE);//just to let me know top corner
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(40,40);
  tft.println("Some Text");

  if (!ts.begin(40)) {
    Serial.println("Unable to start the capacitive touchscreen.");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  touchProcess();
}

void touchProcess() {
  int q;
  int xTouch;
  int yTouch;
  //deal with screen presses
  //if (currentMillis - screenMillis >= screenTimer) {
    if (ts.touched()) {
      //screenMillis = currentMillis;  //reset debounce only if screen pressed
      // Retrieve a point
      TS_Point p = ts.getPoint();

      xTouch = p.x;
      xTouch = map(xTouch, 316, 10, 0, 319);
      if (xTouch < 0) {
        xTouch = 0;
      }
      if (xTouch > 319) {
        xTouch = 319;
      }
      Serial.println("x:" + String(xTouch));

      yTouch = p.y;
      yTouch = map(yTouch, 463, 2, 0, 479);
      if (yTouch < 0) {
        yTouch = 0;
      }
      if (yTouch > 479) {
        yTouch = 479;
      }
      Serial.println("y:" + String(yTouch));
      
      
    }
  //}
}

```
---------------------------------------------- end test sketch --------------------------------------------------

----------------------------------------------User_Setup.h--------------------------------------------------------------
#define USER_SETUP_INFO "User_Setup"

#define ILI9486_DRIVER

#define TFT_INVERSION_OFF

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_TOUCH_FREQUENCY  2500000
--------------------------------------------end User_Setup.h--------------------------------------------------------

IOM2021, 2024-06-10:
Solution found, the issue was another library with a user_setup.h causing a conflict

closed 2024-06-10 by IOM2021

suited-1, 2025-03-16:
what other library? im having the same issue but the only other lib i have imported is SPI