Bodmer/TFT_eSPI #3355: Arduino-ESP32 version 3.0.0 does not complile with LilyGo T-display S3 and TFT_eSPI

Berg0162, 2024-06-06:
Board: LilyGo T-display S3 --> Board manager setting: "ESP32S3 Dev Module" with the right Tools settings
Arduino IDE version: 2.3.2; CLI Version: 0.35.3
TFT_eSPI version: 2.5.43; previous version: 2.5.34

Arduino-ESP32 version: 2.0.17
--> All my code compiles without problems and works fine with TFT_eSPI (versions 2.5.43 and 2.5.34) !!

**After installing: Arduino-ESP32 version 3.0.0;** 

### Compilation error: exit status 1
```
In file included from d:\Users\31651\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:
d:\Users\31651\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c: In member function 'uint8_t TFT_eSPI::readByte()':
d:\Users\31651\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c:96:8: error: 'gpio_get_level' was not declared in this scope; did you mean 'gpio_ll_get_level'?
   96 |   b  = gpio_get_level((gpio_num_t)TFT_D0); // Read three times to allow for bus access time
      |        ^~~~~~~~~~~~~~
      |        gpio_ll_get_level

exit status 1
```
After re-installing Arduino-ESP32 version: **2.0.17** ---> **NO Compilation error**
Compliation Error is NOT dependent on the TFT_eSPI version. All my code works fine with TFT_eSPI and the LilyGo board of version 2.0.17 and previous

Our friends with Espressif have made changes (once again) that break up the TFT_eSPI library for this board (and others?).

johgoy, 2024-06-07:
See this link for a workaround: https://github.com/Bodmer/TFT_eSPI/issues/3346

Berg0162, 2024-06-08:
Thanks for this suggestion! 
I have changed the file: **Processors/TFT_eSPI_ESP32.h** and added **#include "driver/gpio.h"**
```
// Include processor specific header
#include "soc/spi_reg.h"
#include "driver/spi_master.h"
#include "driver/gpio.h"
#include "hal/gpio_ll.h"
```
Updated to version arduino-ESP32 version 3.0.1 ....

The above mentioned compile error complaining about **gpio_get_level** is persistent in the new situation:
**Compilation error: exit status 1**

This is NOT solving the problem!

jhbmueller, 2024-06-09:
Try add it in TFT_eSPI_ESP32**_S3**.h.
Worked for me.

johgoy, 2024-06-10:
"Try add it in TFT_eSPI_ESP32_S3.h.
Worked for me."

Worked for me too.

AlbCod, 2024-06-13:
> "Try add it in TFT_eSPI_ESP32_S3.h. Worked for me."
> 
> Worked for me too.

Yes, I can compile but the screen remains black. It worked before

Berg0162, 2024-06-15:
I have added `#include "driver/gpio.h"` to the file: Processors/TFT_eSPI_ESP32_S3.h

```C++
// Processor ID reported by getSetup()
#define PROCESSOR_ID 0x32

// Include processor specific header
#include "soc/spi_reg.h"
#include "driver/spi_master.h"
#include "hal/gpio_ll.h"
#include "driver/gpio.h"
```
I have tested this with arduino-ESP32 version 2.0.17 and the latest version: 3.0.1
Indeed, for both versions: **NO Compile error anymore** with my code -> problem solved!

Please include this in the next TFT_eSPI version!

steffenkicherer, 2025-01-28:
"Try add it in TFT_eSPI_ESP32_S3.h.
Worked for me."

Worked for me too. Thank you very much!!!!

closed 2025-03-03 by Berg0162